JSBSim-Team/jsbsim #310: Possible out of bounds access on array in debug mode

sthagen, 2020-08-04:
In `src/models/flight_control/FGFilter.cpp` and when `(debug_lvl & 1) && from == 0` the array `C` with 7 slots will be accessed at index `7` and evaluate that value in a boolean context during the last iteration of the loop. cppcheck 1.82 warns on that one ... but maybe this is not a problem?
```cpp
void FGFilter::Debug(int from)
{
  if (debug_lvl <= 0) return;

  if (debug_lvl & 1) { // Standard console startup message output
    if (from == 0) { // Constructor
      cout << "      INPUT: " << InputNodes[0]->GetName() << endl;

      for (int i=1; i <= 7; i++) {
        if (!C[i]) break;  // here access of C[7] in final iteration - test random value or seg fault?
// ...
```

seanmcleod, 2020-08-04:
Yep it looks like the for loop will run off the end of the array.

https://github.com/JSBSim-Team/jsbsim/blob/a9fb5200c176953003253344f3ae5274897d4f6b/src/models/flight_control/FGFilter.h#L231

Please post a link to the relevant code as opposed to cutting and pasting, makes it easier and quicker for others to quickly jump to the code and browse around the relevant snippet. 

https://github.com/JSBSim-Team/jsbsim/blob/a9fb5200c176953003253344f3ae5274897d4f6b/src/models/flight_control/FGFilter.cpp#L207-L226

seanmcleod, 2020-08-04:
@sthagen do you want to submit a pull request, or do you just want to report the issue and have one of the JSBSim developers fix it?

sthagen, 2020-08-05:
I am happy to provide a pull request. Just wanted to be sure upfront that the project considers this an issue and indeed is open for contribution. Thanks for the fast and encouraging response (also on issue #309).

PS: From years at stackoverflow, I thought links would not be welcome, but with a versioning system under the hood the linkage I would prefer myself 😉 - thank you @seanmcleod for the hint.

closed 2020-08-05 by bcoconni

seanmcleod, 2020-08-05:
@sthagen yes we are definitely open for contributions.

Yep the github links are nice and smart ;-)

PR #311 (2020-08-05): Fixed one off error in array iteration
Closes #310 

Since I maintain an active fork (via github web ui) there maybe quite some merge commits in my history, but this is a single character change.

The deployment stages of travis failed in my fork - is this OK? Hope the checks in this upstream repo succeed though.