JSBSim-Team/jsbsim #316: Plans for Ubuntu Focal release

Jaeyoung-Lim, 2020-08-29:
I was wondering if there were any plans for a release on Ubuntu Focal, since this is the latest LTS of Ubuntu.

I have tried installing the Ubuntu 18.04 rolling release deb package from this repo, but it doesn't seem to work. 

The way currently I installed jsbsim on 20.04 was by installing flightgear. However, it would be nice if we could have a jsbsim deb package as a rolling release. 

Related: https://github.com/PX4/containers/pull/276

bcoconni, 2020-08-31:
Generally speaking, we are constrained by the availability of the different versions of Ubuntu LTS on GitHub servers. Fortunately, it seems that [Ubuntu 20.04 LTS is now available on GitHub hosted runners](https://docs.github.com/en/actions/reference/software-installed-on-github-hosted-runners) so we need to update our GitHub workflow to build packages for 20.04.

I have added this issue to our project "Migrating our CI workflow to GitHub".

Jaeyoung-Lim, 2020-08-31:
@bcoconni Thanks!

bcoconni, 2020-08-31:
> If it is a matter of just extending this workflow to a matrix build, I think I can also try to extend it. Is there anything more that needs to be done?

Yes, just that workflow. Nothing else.

closed 2020-09-05 by bcoconni