JSBSim-Team/jsbsim #320: Starting flightgear / JSBSim together floods the terminal with "send: Connection refused"

Jaeyoung-Lim, 2020-09-10:
I have an integration of jsbsim and on the side flightgear to visualize the vehicle.

My goal is to automate starting both processes from a single make target.

I am running flightgear as a detached process with udp connection, e.g.
```
fgfs --fdm=null \
			--native-fdm=socket,in,60,,5550,udp \
			--aircraft=$JSBSIM_AIRCRAFT_MODEL &
```

and on the jsbsim integration side, I set the output directives through `SetOutputDirectives` which look like the following. There is currently only one way of information flow since flightgear is just used for visualization. 
```
<?xml version="1.0" encoding="utf-8"?>
<output name="localhost" type="FLIGHTGEAR" protocol="UDP" port="5550" rate="60">
</output>
```
However, when I start both processes, jsbsim floods the terminal with `send: Connection refused` until flightgear has been fully loaded. I understand this is happening since flightgear has not started to listen to the UDP port while the jsbsim application already started. Currently I am dealing this "inconvenience" with adding a sleep, but was wondering if there was a better way to handle this.

Would there be a way to detect whether flightgear has been fully brought up on the JSBSim side?
- I have looked into the code without any success on finding any method to check if flightgear has been fully loaded.
- I tried using TCP as the output format, but didn't manage to get flightgear and jsbsim communicating correctly.

seanmcleod, 2020-09-10:
Well in a sense FlightGear accepting UDP packets is an indication that it is fully loaded.

However it's not 100% clear what your requirements are. Do you not want JSBSim to start executing the simulation until FlightGear is up and running or do you simply want to get rid of the `send: Connection refused` warning messages in which case you may miss the first couple of seconds of the simulation in FlightGear.

Jaeyoung-Lim, 2020-09-10:
> However it's not 100% clear what your requirements are. Do you not want JSBSim to start executing the simulation until FlightGear is up and running or do you simply want to get rid of the send: [...]

@seanmcleod I would like to get rid of `send: Connection refused` messages, It is okay if flightgear misses some packages.
Is there anyway to configure the verbosity of these warnings?

seanmcleod, 2020-09-10:
@Jaeyoung-Lim in general JSBSim does have support for debug levels of output, for example:

https://github.com/JSBSim-Team/jsbsim/blob/0279e4b8e683c019b80ad128db8aebd718072759/src/input_output/FGfdmSocket.cpp#L406-L428

However in the case of socket errors when sending the debug level isn't used to determine whether to output a debug/error message. In this case `perror()` is called which will write to `stderr`, which is what you're seeing.

https://github.com/JSBSim-Team/jsbsim/blob/0279e4b8e683c019b80ad128db8aebd718072759/src/input_output/FGfdmSocket.cpp#L357-L362

seanmcleod, 2020-09-10:
Actually if you look through `FGfdmSocket.cpp` the `debug_lvl` is never used to suppress error messages, only to allow or suppress informational messages, e.g.

https://github.com/JSBSim-Team/jsbsim/blob/0279e4b8e683c019b80ad128db8aebd718072759/src/input_output/FGfdmSocket.cpp#L119-L124

Whereas errors are always output, e.g.

https://github.com/JSBSim-Team/jsbsim/blob/0279e4b8e683c019b80ad128db8aebd718072759/src/input_output/FGfdmSocket.cpp#L209-L211

Jaeyoung-Lim, 2020-09-10:
@seanmcleod Thanks for looking into this. Would it make sense to suppress errors? (not sure if this is a good idea)

seanmcleod, 2020-09-10:
@Jaeyoung-Lim as a quick work-around you could redirect all output to `stderr` to a file that gets overwritten on each run. So on the assumption that 99% of your runs work as expected you'll see no error messages in the console. If something does go wrong on a particular run you could then view your `stderr.txt` file.

Jaeyoung-Lim, 2020-09-11:
@seanmcleod Thanks, I was looking for more of a way to detect if flightgear have opened the UDP port without flooding the messages, but redirecting stderr works for now

closed 2020-09-11 by Jaeyoung-Lim