JSBSim-Team/jsbsim #321: Reset JSBSim Python to new initial conditions

sdeal110, 2020-09-11:
Hello, I have been using JSBSim for a while now, but I am testing it with an RRT planning algorithm that requires me to reset the sim at different previous states along the trajectory. Despite calling numerous of the supplied reset functions, it always seems to start with the same initial conditions as the very first point, except location. (I have gotten it to start at different locations, but always heading in the same direction etc as the start). Below is an example of the code I have been calling to reset the sim. Anyone know why this isn't working for me?

     def reset(self, state, lla):
            self.fdm.reset_to_initial_conditions(0)
            # self.fdm.load_ic("new_ic.xml", useStoredPath=False) # I tried loading IC's from an xml file to see if that would work
            
            self.fdm.set_property_value("ic/lat-geod-deg",self.lla[0]) # the llo seem to set but none of the other parameters
            self.fdm.set_property_value("ic/long-gc-deg",self.lla[1])
            self.fdm.set_property_value("ic/h-sl-meters",state[2])
            self.fdm.set_property_value("ic/u-fps",state[3]/0.3048)
            self.fdm.set_property_value("ic/v-fps",state[4]/0.3048)
            self.fdm.set_property_value("ic/w-fps",state[5]/0.3048)
            self.fdm.set_property_value("ic/p-rad_sec",state[6])
            self.fdm.set_property_value("ic/q-rad_sec",state[7])
            self.fdm.set_property_value("ic/r-rad_sec",state[8])
            self.fdm.set_property_value("ic/phi-rad",state[9])
            self.fdm.set_property_value("ic/theta-rad",state[10])
            self.fdm.set_property_value("ic/psi-rad",state[11])
            self.fdm.set_property_value('propulsion/set-running', -1)
            
            self.fdm.run_ic()

seanmcleod, 2020-09-11:
@sdeal110 what happens if you try:

```python
 def reset(self, state, lla):    
        self.fdm.set_property_value("ic/lat-geod-deg",self.lla[0]) # the llo seem to set but none of the other parameters
        self.fdm.set_property_value("ic/long-gc-deg",self.lla[1])
        self.fdm.set_property_value("ic/h-sl-meters",state[2])
        self.fdm.set_property_value("ic/u-fps",state[3]/0.3048)
        self.fdm.set_property_value("ic/v-fps",state[4]/0.3048)
        self.fdm.set_property_value("ic/w-fps",state[5]/0.3048)
        self.fdm.set_property_value("ic/p-rad_sec",state[6])
        self.fdm.set_property_value("ic/q-rad_sec",state[7])
        self.fdm.set_property_value("ic/r-rad_sec",state[8])
        self.fdm.set_property_value("ic/phi-rad",state[9])
        self.fdm.set_property_value("ic/theta-rad",state[10])
        self.fdm.set_property_value("ic/psi-rad",state[11])
        self.fdm.set_property_value('propulsion/set-running', -1)
        
        self.fdm.reset_to_initial_conditions(0)
```

`reset_to_initial_conditions(0)` has code to call `run_ic()` etc.

sdeal110, 2020-09-11:
Still no good. I think your right, but whenever it restarts at a new spot it just uses the same old ic's still. See the results in the picture below.

[image]

bcoconni, 2020-09-12:
Not sure if that is the cause of you problem, but the properties `ic/h-sl-meters` and `ic/psi-rad` do not exist, you should use `ic/h-sl-ft` and `ic/psi-true-rad` instead.
You can get the list of valid properties for initial conditions with the command
```python
self.fdm.query_property_catalog('ic/')
```

sdeal110, 2020-09-14:
You were correct, that seemed to be my problem.

here is the list of f16 IC's for reference:

'ic/lat-gc-deg (RW)'
'ic/long-gc-deg (RW)'
'ic/h-sl-ft (RW)'
'ic/h-agl-ft (RW)'
'ic/terrain-elevation-ft (RW)'
'ic/vg-fps (RW)'
'ic/vt-fps (RW)'
'ic/vw-bx-fps (R)'
'ic/vw-by-fps (R)'
'ic/vw-bz-fps (R)'
'ic/vw-north-fps (R)'
'ic/vw-east-fps (R)'
'ic/vw-down-fps (R)'
'ic/vw-mag-fps (R)'
'ic/vw-dir-deg (RW)'
'ic/roc-fps (RW)'
'ic/u-fps (RW)'
'ic/v-fps (RW)'
'ic/w-fps (RW)'
'ic/vn-fps (RW)'
'ic/ve-fps (RW)'
'ic/vd-fps (RW)'
'ic/gamma-rad (RW)'
'ic/alpha-rad (RW)'
'ic/theta-rad (RW)'
'ic/beta-rad (RW)'
'ic/phi-rad (RW)'
'ic/psi-true-rad (RW)'
'ic/lat-gc-rad (RW)'
'ic/long-gc-rad (RW)'
'ic/p-rad_sec (RW)'
'ic/q-rad_sec (RW)'
'ic/r-rad_sec (RW)'
'ic/lat-geod-rad (RW)'
'ic/lat-geod-deg (RW)'
'ic/geod-alt-ft (R)'
'ic/targetNlf (RW)'

seanmcleod, 2020-09-14:
@sdeal110 is that the complete list?

I would've expected the list to include all the `ic/` properties created from `FGInitialCondition::bind()`

https://github.com/JSBSim-Team/jsbsim/blob/687eb3facca9d99fb92175982ed244256caa0ca4/src/initialization/FGInitialCondition.cpp#L1446

But your list appears to be a subset.

bcoconni, 2020-09-14:
@seanmcleod `FGFDMExec::QueryPropertyCatalog`is supposed to be exhaustive. Exactly which properties did find were missing ?

sdeal110, 2020-09-14:
That was the list I got when I entered the following python commands:

```
import jsbsim
fdm = jsbsim.FGFDMExec('.', None)
fdm.load_model('f16')
fdm.query_property_catalog('ic/')
```

Now its reseting like I wanted it to so I appreciate the help. I am not sure what IC's are missing, but the ones I needed are there even if the units need conversion

seanmcleod, 2020-09-14:
Take the 1st 5 that appear in `FGFDMExec::QueryPropertyCatalog` 

https://github.com/JSBSim-Team/jsbsim/blob/687eb3facca9d99fb92175982ed244256caa0ca4/src/initialization/FGInitialCondition.cpp#L1446

```sh
ic/vc-kts
ic/ve-kts
ic/vg-kts
ic/vt-kts
ic/mach
```

They don't appear in @sdeal110's list, in general @sdeal110's list is much shorter.

On closer comparison it looks like maybe the first 20 or so scrolled out of view since all the missing ones appear before the 1st one that @sdeal110 lists, i.e. `ic/lat-gc-deg`.

seanmcleod, 2020-09-14:
Sure enough they do all appear, it does look like, I'm guessing, the 1st 12 properties that were missing in @sdeal110's list were missing because they scrolled out of view or were missed during a cut and paste.

They stood out because they were the 1st set of properties I looked for based on what I saw in the source code.

```python
import jsbsim
fdm = jsbsim.FGFDMExec('.', None)
fdm.load_model('f16')

for ic in fdm.query_property_catalog('ic/'):
   print(ic)

ic/vc-kts (RW)
ic/ve-kts (RW)
ic/vg-kts (RW)
ic/vt-kts (RW)
ic/mach (RW)
ic/roc-fpm (RW)
ic/gamma-deg (RW)
ic/alpha-deg (RW)
ic/beta-deg (RW)
ic/theta-deg (RW)
ic/phi-deg (RW)
ic/psi-true-deg (RW)
ic/lat-gc-deg (RW)
ic/long-gc-deg (RW)
ic/h-sl-ft (RW)
ic/h-agl-ft (RW)
ic/terrain-elevation-ft (RW)
ic/vg-fps (RW)
ic/vt-fps (RW)
ic/vw-bx-fps (R)
ic/vw-bz-fps (R)
ic/vw-north-fps (R)
ic/vw-east-fps (R)
ic/vw-down-fps (R)
ic/vw-mag-fps (R)
ic/vw-dir-deg (RW)
ic/roc-fps (RW)
ic/u-fps (RW)
ic/v-fps (RW)
ic/w-fps (RW)
ic/vn-fps (RW)
ic/ve-fps (RW)
ic/vd-fps (RW)
ic/gamma-rad (RW)
ic/alpha-rad (RW)
ic/theta-rad (RW)
ic/beta-rad (RW)
ic/phi-rad (RW)
ic/psi-true-rad (RW)
ic/lat-gc-rad (RW)
ic/long-gc-rad (RW)
ic/p-rad_sec (RW)
ic/q-rad_sec (RW)
ic/r-rad_sec (RW)
ic/lat-geod-rad (RW)
ic/lat-geod-deg (RW)
ic/geod-alt-ft (R)
ic/targetNlf (RW)

```

sdeal110, 2020-09-14:
Totally possible I messed up with the copying and pasting. Python wasn't printing each one out on a new line so I might have missed something pushing each entry to a new line for the thread post

sdeal110, 2020-09-14:
I just ran the commands again too and ya I missed the first several properties in the copy past... fortunately I didn't need them or I might have still been scratching my head!

bcoconni, 2020-09-14:
> Now its resetting like I wanted it to so I appreciate the help.
> [...] I just ran the commands again too and ya I missed the first several properties in the copy past...

So I guess we shall consider this issue closed ?

closed 2020-09-15 by sdeal110