JSBSim-Team/jsbsim #325: Flightgear visualization not working on jsbsim

Jaeyoung-Lim, 2020-09-28:
When running JSBSim on Ubuntu Focal, the flightgear visualization is not working,

I noticed the error
```
   11.00 [ALRT]:io         Error: version mismatch in FGNetNativeGUI2Props()
   11.00 [ALRT]:io              read 402653184 need 8
   11.00 [ALRT]:io              Need to upgrade net_fdm.hxx and recompile.
```

Where can I find the compatible versions of flightgear and jsbsim?

bcoconni, 2020-09-28:
That's weird: [a modification has been introduced recently in `net_fdm.hxx`](https://sourceforge.net/p/flightgear/flightgear/ci/932938cbeebff84eef24e97b0e8a7c789c0a4f71/) in FlightGear but this should be not available before the next release.

In order to reproduce this error, we would need to know which versions of FlightGear and JSBSim you are using ?

Jaeyoung-Lim, 2020-09-29:
The version is as the following
- FlightGear version: 2020.3.0
- JSBSim: JSBSim-devel_1.1.0.dev1-89.focal.amd64.deb

bcoconni, 2020-09-29:
Ok so you are using the bleeding edge code from FlightGear. in that case, I would say that the easiest workaround in the short term is to use FlightGear 2020.2 instead.

Meanwhile I will investigate how to make JSBSim compatible with 2020.3

bcoconni, 2020-10-02:
Just a heads up: I confirm that JSBSim is not compatible with FlightGear 2020.3 since the FDM protocol has been modified as I mentioned earlier. JSBSim network interface does indeed work with FlightGear 2020.2 or earlier so you can use one of these releases as a fallback for the time being.

[I have reopened a ticket in FlightGear](https://sourceforge.net/p/flightgear/codetickets/2313) to raise our issue to their knowledge. Given that FlightGear 2020.2 will a be a LTS release and that the protocol has been changed in FlightGear 2020.3, I guess that our only viable option is to be compatible with both versions of the protocol.

Jaeyoung-Lim, 2020-10-03:
@bcoconni Thanks!

bcoconni, 2020-10-03:
@Jaeyoung-Lim I've just pushed a revision to handle both versions of FlightGear's FDM network protocol.
You need to specify the version:
* `24` for FlightGear 2020.2 and earlier
* `25` for FlightGear 2020.3 and later

The version should be specified in the file `data_output/flightgear.xml` in the attribute `version`:
https://github.com/JSBSim-Team/jsbsim/blob/b42f4238880ae1729acd55894381f74a0fe1ca13/data_output/flightgear.xml#L18-L19

The binary packages are available in our release section so could you please test and let us know if that works ?
Thanks.

Jaeyoung-Lim, 2020-10-03:
@bcoconni Thanks for the quick response,

I tried with 
```
<output name="localhost" type="FLIGHTGEAR" protocol="UDP" port="5550" rate="60" version="25">
</output>
```
but this still didn't work for me. (The vehicle doesn't move when the simulation is running.

This is the JSBSim FDM model that is being used, so I think this should be the latest release you are referring to
```
     JSBSim Flight Dynamics Model v1.1.0.dev1 [GitHub build 97/commit b42f4238880ae1729acd55894381f74a0fe1ca13] Oct  3 2020 16:56:25
            [JSBSim-ML v2.0]
```

bcoconni, 2020-10-05:
OK that's unfortunate indeed. Could you please provide the exact sequence of commands you are using ? I'll see if I can reproduce the problem locally. If you are using some Python code, please provide as short as possible a piece of code that triggers the problem.

Also have you checked using FlightGear 2020.2 or earlier ? If so, did that fix the problem ?

bcoconni, 2020-10-19:
@Jaeyoung-Lim Have you had the opportunity to investigate about the questions I asked ?

Jaeyoung-Lim, 2020-10-20:
@bcoconni Sorry to come back so late. Unfortunately, trying Flightgear 2020.2 did not fix the issue

The sequence I am using is as the following, and is currently part of the PX4 autopilot project [startup script](https://github.com/PX4/Firmware/blob/d9b3db1082611f7a7c8fab47f7d57575336dd9de/Tools/sitl_run.sh#L154-L169)

I start flightgear with 
```
		fgfs --fdm=null \
			--native-fdm=socket,in,60,,5550,udp \
			--aircraft=$JSBSIM_AIRCRAFT_MODEL \
			--airport=${world} \
			--disable-hud \
			--disable-ai-models &> /dev/null &
```
and run a jsbsim bridge that we created
```
"${build_path}/build_jsbsim_bridge/jsbsim_bridge" ${model} -s "${src_path}/Tools/jsbsim_bridge/scene/${world}.xml" 2> /dev/null &
```

The bridge can be found in [px4-jsbsim-bridge](https://github.com/Auterion/px4-jsbsim-bridge) where the output format is selected [here](https://github.com/Auterion/px4-jsbsim-bridge/blob/aa8997c309d1966d1f6b5d82e51cec4ed8ca66e6/src/jsbsim_bridge.cpp#L121) and the output directives are defined in [flightgear.xml](https://github.com/Auterion/px4-jsbsim-bridge/blob/master/data_out/flightgear.xml): 
```
    _fdmexec->SetOutputDirectives(SGPath("data_out/flightgear.xml"));
```

bcoconni, 2020-10-20:
@Jaeyoung-Lim Thanks. I will investigate that.

bcoconni, 2020-10-31:
@Jaeyoung-Lim, some changes have been introduced to the socket code by the commit 4fdf246. As per @cgdae feedback, we have moved from non portable socket code (using `ioctl(... FIONBIO ...)`) to POSIX functions calls `fnctl(... O_NONBLOCK ...)` as per the explanation in [this StackOverflow message](https://stackoverflow.com/questions/1150635/unix-nonblocking-i-o-o-nonblock-vs-fionbio#1151077).

Could you please test the code from `HEAD` of the `master` branch and let us know if that fixes the problem on Ubuntu 20.04 ? The compiled binaries are available in the [Rolling release](https://github.com/JSBSim-Team/jsbsim/releases/tag/Linux) as usual.

**N.B.** If you plan to conduct your tests with FlightGear 2020.3, do not forget to set the `version` parameter to `25` in `data_output/flightgear.xml` as we discussed above.

Jaeyoung-Lim, 2020-11-02:
@bcoconni Thanks!

I am wondering if there is something wrong with my system. I will try to test with a fresh system and report back

bcoconni, 2020-12-11:
@Jaeyoung-Lim Any progress on that topic ?

Jaeyoung-Lim, 2020-12-17:
@bcoconni Hi, I still can't get it to work even with a fresh system...

bcoconni, 2020-12-18:
OK. I suggest to leave PX4 aside and to go back to basics.
Can you run successfully the [basic example from our README](https://github.com/JSBSim-Team/jsbsim#how-can-i-interface-jsbsim-with-flightgear-) ?

layssi, 2024-09-07:
I can confirm flightgear does not run in ubuntu

bcoconni, 2024-12-21:
> I can confirm flightgear does not run in ubuntu

Not sure about what you mean @layssi ? If it is FlightGear itself that is not working, I'd suggest that you discuss that with the FlightGear team. The repo JSBSim-Team/jsbsim is dedicated to the FDM JSBSim which is used by FlightGear but also by other simulations.

bcoconni, 2024-12-21:
Good news ! The FlightGear team has finally dropped the change [FlightGear ticket 2313](https://sourceforge.net/p/flightgear/codetickets/2313/) that was breaking the socket connection between JSBSim and FlightGear. It was also breaking the connection between MatLab and Flightear by the way. So the protocol 25 is no longer needed. I'll remove it from our code base.

bcoconni, 2025-02-28:
The tentative fix for FlightGear new protocol has been removed from JSBSim. We are now back to the FlightGear network protocol v24 which is the legacy protocol that has been successfully use for years.

The branch `master` should now be compatible again with FlightGear under Ubuntu. Sorry for the very long time before providing a fix.

Jaeyoung-Lim, 2025-03-18:
@bcoconni Would it be possible to get a release with the fixes included?

bcoconni, 2025-03-22:
> @bcoconni Would it be possible to get a release with the fixes included? Would be great to have a release that works with this again.

Version 1.2.2 has just been released which includes the fixes.

> Unfortunately building jsbsim from source did not seem to fix the problem.
> 
> I am a little confused though, I am running FG 2020.3, which is the version disturbuted on the ppa., and build jsbsim fron source. So this problem should not exist in this case, right?
> 
> Or do I need to build flight gear from source as well?

The code has been restored to the version that used to work.

At least, following this change, I was able to drive FlightGear from JSBSim over a network connection using the process described in [FAQ - How can I interface JSBSim with FlightGear ?](https://github.com/JSBSim-Team/jsbsim/wiki/Frequently-Asked-Questions#how-can-i-interface-jsbsim-with-flightgear-) I have run the test with the version 2020.4 of FlightGear but it should work with 2020.3 as well.