JSBSim-Team/jsbsim #329: Reset without Trimming

dershow, 2020-10-02:
I’m running JSBSim, using the python API, and it is working.  But, I would like to be able to reset to prior conditions without rerunning the trim each time.  So, what I would like to do is to run trim once, and store the relevant variables.  Then, later reset and just reload them, instead of running trim again.  That would also allow me to continue running a simulation that had been run before.  I would also like to be able to just continue a simulation with different conditions. ie trim, then run for 60 seconds and store the state.  Then continue for another 30 seconds.  Then reset to the 60 second point and run it again from there.  
It is almost working, but I am not getting everything to work.  For example when I’m using the c172x and if I run reset01.xml then run do simple trim and run the sim, it holds altitude very well (as expected).  While if I then reset and set the variables, the aircraft starts to descend.    I think that I am just missing some pitch or engine related variable, or not setting it correctly.  Here are some specifics:
 
```
#reset and run trim:
fdm.reset_to_initial_conditions(True)
fdm['simulation/do_simple_trim'] = 1
 
#Store the variables I will need later:
alpha=fdm['aero/alpha-deg']
roll=fdm['attitude/phi-deg']
 
trim=fdm['fcs/pitch-trim-cmd-norm']
throttle=fdm['fcs/throttle-pos-norm']
ailerons=fdm['fcs/aileron-cmd-norm']
rudder=fdm['fcs/rudder-pos-norm']
elev=fdm['fcs/elevator-pos-deg']
 
#do some running:
t = fdm['simulation/sim-time-sec']
while t<60.:
result=fdm.run()
t = fdm['simulation/sim-time-sec']
 
```
 
Then, do this, and I hope to get identical results to above
 
```
#Load a few initial conditions
fdm['ic/alpha-deg']=alpha
fdm['ic/phi-deg']=roll
#reset to apply those
fdm.reset_to_initial_conditions(True)
 
fdm['fcs/pitch-trim-cmd-norm']=trim
fdm['fcs/throttle-cmd-norm']=throttle
fdm['fcs/aileron-cmd-norm']=ailerons
fdm['fcs/rudder-pos-norm']=rudder
fdm['propulsion/set-running']=-1
 
t = fdm['simulation/sim-time-sec']
while t<60.:
result=fdm.run()
t = fdm['simulation/sim-time-sec']

``` 
 
But, I’m getting different results.  As I said, I think that I must have missed some variable, or not set it correctly (command versus position?)
So, I guess the overall question is what variables does do_simple_trim actually set?  And, can I just store and reload those, to repeat or continue a simulation? 
Any guidance about what might be missing from the above?
 
Thanks,
 
-- Adam

bcoconni, 2020-10-02:
It seems that you are not providing the complete position and orientation parameters before calling the function `fdm.reset_to_initial_conditions()`. The properties you need to specify are:
* The orientation: `ic/theta-rad` (pitch), `ic/phi-rad` (roll) and `ic/psi-true-rad` (heading)
* The velocity: `ic/u-fps`, `ic/v-fps` and `ic/w-fps` which are the velocity components in the body frame, respectively along the X, Y and Z axes.
* The angular rate: `ic/p-rad_sec`, `ic/q-rad_sec` and `ic/r-rad_sec` which are the rotational velocity components in the body frame, respectively around the X, Y and Z axes.
* The position (at least the altitude since the aircraft performance depends on the air density) `ic/h-sl-ft`.

Please note that there is some interdependence between all the parameters that describe aircraft attitude. For instance once you have specified `ic/u-fps`, `ic/w-fps` and `ic/phi-rad` then the angle of attack `ic/alpha-rad` is also set.

Also note that the property `ic/alpha-deg` is not an initial conditions parameter. For the complete list of initial conditions properties, you can read the discussion in issue #321 where you can find more details on that subject.

seanmcleod, 2020-10-02:
So in general to re-use a previously trimmed state you're going to have to replicate the following:

- Velocities (U,V,W)
- Attitude (pitch, roll, heading)
- Position, particularly altitude
- Control positions 
- Flaps, gear etc. configuration (could be seen as a subset of control positions)
- Total mass
- Mass distribution

So if the sim has been running for a while does `fdm.reset_to_initial_conditions()` also reset the mass? I haven't double-checked, it may well.

Also if you're manually changing things like throttle position (also potentially applies to other controls) remember there could be lags, delays modelled which you have to take into account to ensure that the command and position are in sync. I'm pretty sure the trim code has specific code to disable things like actuator delays etc.

bcoconni, 2020-10-02:
> So if the sim has been running for a while does `fdm.reset_to_initial_conditions()` also reset the mass? I haven't double-checked, it may well.

Yes, good point.
> Also if you're manually changing things like throttle position (also potentially applies to other controls) remember there could be lags, delays modeled which you have to take into account to ensure that the command and position are in sync. I'm pretty sure the trim code has specific code to disable things like actuator delays etc.

The method `fdm.reset_to_initial_conditions()` also takes care of that since it calls `FGFDMExec::ResetToInitialConditions`
https://github.com/JSBSim-Team/jsbsim/blob/8585b486d0af0d9b174c9ab82e847eb6be759a0c/src/FGFDMExec.cpp#L602-L616
which calls `FGFDMExec::InitializeModels` which resets the models by calling their method `InitModel`:
https://github.com/JSBSim-Team/jsbsim/blob/8585b486d0af0d9b174c9ab82e847eb6be759a0c/src/FGFDMExec.cpp#L291-L300

bcoconni, 2020-10-19:
@dershow Has your problem been fixed ? Or do you need further help ?

bcoconni, 2020-11-02:
Issue closed (no response from @dershow).

closed 2020-11-02 by bcoconni