JSBSim-Team/jsbsim #330: Error running script

StepLogic, 2020-10-02:
Hi  I just got JSBSim  installed and I get the error below no matter which script reference
I am runnig python 3.6 and I used this [script](https://github.com/c172p-team/c172p/blob/master/c172p.xmll)
File: Path "C:/Users/Gyaase Elvis/Desktop/Python Projects/tools/sim/scripts/c.xml" is not a script file
Attempt to divide by zero in method     FGColumnVector3::operator/(const double scalar),     object 0 , 0 , 0

  A proper axis type has NOT been selected. Check your aerodynamics definition.

bcoconni, 2020-10-02:
I might be wrong but it seems you did not specified correctly the script file name since the error message mentions a file name `c.xml` while the script you are referring to is named `c172p.xml`.

StepLogic, 2020-10-02:
I renamed the file c.xml

bcoconni, 2020-10-02:
OK. So you need to specify the "root directory" to JSBSim that is the directory which contains the folders `aircraft`, `engine` and `systems`.

If you are using the command line utility `JSBSim` then you must specify the argument `--root`:
```bash
> JSBSim --root=path/to/root/directory /path/to/scripts/c.xml
```
If you are using JSBSim in a Python program then you must pass the the root directory path to `FGFDMExec`:
```python
import jsbsim

fdm = jsbsim.FGFDMExec('path/to/root/directory')
fdm.load_script('path/to/scripts/c.xml')
```
Let us know if that helps.

bcoconni, 2020-10-02:
Oh and by the way, the contents of the "root directory" is not part of the Python module. You need to [download the zip file of the source code](https://github.com/JSBSim-Team/jsbsim/archive/Windows-MSVC.zip) and unzip it its content to a folder that will become your "root directory".

bcoconni, 2020-10-03:
I just realized that the file you mentioned **is not** a script but an aircraft definition file. Scripts are files like the ones located in the folder [`scripts`](https://github.com/JSBSim-Team/jsbsim/tree/master/scripts) of JSBSim source files that you can download as I explained above.

StepLogic, 2020-10-04:
thanks let me give it a go

StepLogic, 2020-10-04:
Thanks it worked

closed 2020-10-04 by StepLogic