JSBSim-Team/jsbsim #332: Magnetometer Startup Sequence Causes ~5 Seconds of Erroneous Data

mvacanti, 2020-10-18:
**Issue Description**
Using an airframe with a [magnetometer system](https://github.com/JSBSim-Team/jsbsim/blob/master/src/models/flight_control/FGMagnetometer.cpp) the startup sequence does not correctly apply the magnetic variation until `counter` reaches 1,000 - effectively delaying the correct publication of magnetic values for several seconds after startup. 

https://github.com/JSBSim-Team/jsbsim/blob/3743a5c3d849e16cb7d23bdfdf7322608f25246a/src/models/flight_control/FGMagnetometer.cpp#L103-L117

**Observed Issue (output csv)**

Time | /fdm/jsbsim/px4/mag-x | /fdm/jsbsim/px4/mag-y | /fdm/jsbsim/px4/mag-z
-- | -- | -- | --
0 | 0 | -nan | 1.27066231393875E+236
0.0083333 | 0 | -nan | 1.27062309229309E+236
0.0166666 | 0 | -nan | 1.27062309229387E+236
0.0249999 | 0 | -nan | 1.2706230898948E+236
0.0333332 | 0 | -nan | 1.27062308521534E+236
0.0416665 | 0 | -nan | 1.27062307843842E+236
0.0499998 | 0 | -nan | 1.27062306968209E+236
0.0583331 | 0 | -nan | 1.27062305912773E+236
0.0666664 | 0 | -nan | 1.27062304689168E+236
0.0749997 | 0 | -nan | 1.27062303315325E+236
0.083333 | 0 | -nan | 1.27062301802605E+236

**To Duplicate Observed Issue**
Using existing release version of JSBSim and forked versions of [737.xml](https://github.com/mvacanti/jsbsim/blob/pr-update-mag-startup/aircraft/737/737.xml), [B737_Runway](https://github.com/mvacanti/jsbsim/blob/pr-update-mag-startup/scripts/B737_Runway.xml) and [px4_default_mag_sensor.xml](https://github.com/mvacanti/jsbsim/blob/pr-update-mag-startup/systems/px4_default_mag_sensor.xml) system file:  `JSBSim --script=scripts/B737_Runway --realtime`

**Proposed Solution**
https://github.com/mvacanti/jsbsim/blob/468dec7ff7a0b697c1ef5d576961ea0fabb313a3/src/models/flight_control/FGMagnetometer.cpp#L99-L112

```c++
void FGMagnetometer::updateInertialMag(void)
{
  counter++;
  if (counter > INERTIAL_UPDATE_RATE || fcs->GetExec()->GetSimTime() < 1 )//dont need to update every iteration
  {
    counter = 0;
    usedLat = (Propagate->GetGeodLatitudeRad());//radians, N and E lat and long are positive, S and W negative
    usedLon = (Propagate->GetLongitude());//radians
    usedAlt = (Propagate->GetGeodeticAltitude()*fttom*0.001);//km

    //this should be done whenever the position changes significantly (in nTesla)
    calc_magvar( usedLat, usedLon, usedAlt, date, field );
  }
}
```

**Outcome of Proposed Solution (output csv)**

Time | /fdm/jsbsim/px4/mag-x | /fdm/jsbsim/px4/mag-y | /fdm/jsbsim/px4/mag-z
-- | -- | -- | --
0 | -13575.5159865581 | 19990.4290375256 | 50878.2586714236
0.0083333 | -13972.978773585 | 19990.4302273556 | 50770.5431394632
0.0166666 | -13972.9787702055 | 19990.4302273829 | 50770.5431403886
0.0249999 | -13972.9909709449 | 19990.4302274048 | 50770.5397825087
0.0333332 | -13973.0148882115 | 19990.4146933469 | 50770.5393164848
0.0416665 | -13973.0494504092 | 19990.4014946501 | 50770.5350011517
0.0499998 | -13973.0941734815 | 19990.3752352782 | 50770.5330319266
0.0583331 | -13973.1480113585 | 19990.3514606423 | 50770.5275756488
0.0666664 | -13973.2104770532 | 19990.3159477989 | 50770.5243666195
0.0749997 | -13973.2805485248 | 19990.2826480493 | 50770.5181927539
0.083333 | -13973.3577403789 | 19990.2386259766 | 50770.5142809315

seanmcleod, 2020-10-18:
It's not clear whether `calc_magvar()` is very expensive and that's the reason for only calling it every 1000 simulation frames or whether the original code author thought the result returned wouldn't be different every simulation frame and so even if it is an inexpensive function it's a waste to call it for every simulation frame.

Basing it on a fixed number of simulation frames can be a poor proxy for significant movement, the aircraft could be stationary, be flying at very different speeds plus the simulation could be running at 120Hz or 250Hz etc. A better check would be something based on say the latitude or longitude having changed by some delta.

If we stick with an update rate based on a fixed number of simulation frames then instead of some magic 1s check what about simply using the following?

```c++
if (counter++ % INERTIAL_UPDATE_RATE == 0)
```

mvacanti, 2020-10-18:
>`if (counter++ % INERTIAL_UPDATE_RATE == 0)` 

Thanks, much better! 

Agreed that it would be more appropriate if there was a light-weight way to update magvar based on location rather than sim frames. Would need to consider some type of variability particularly near the poles to avoid step changes. I will take the above solution in the near term and consider solutions for a more contextually aware option.

seanmcleod, 2020-10-19:
@mvacanti do you want to submit a pull request with the simple `if (counter++ % INERTIAL_UPDATE_RATE == 0)` change or do you want me to submit it?

bcoconni, 2020-10-19:
@seanmcleod Now that we have an official release 1.1.0, the commit should also be "cherrypicked" to the branch `release/v1.1` so that to be included in the future bug fix release 1.1.1

bcoconni, 2020-10-20:
Thanks @mvacanti and @seanmcleod. I guess this issue can be closed now ?

P.S. I have cherry-picked the commit to the branch `release/v1.1` so it will be part of the bug fix release `v1.1.1`

mvacanti, 2020-10-20:
Closed per #334

closed 2020-10-20 by mvacanti