JSBSim-Team/jsbsim #333: Propeller, engine improvements to support quadcopters etc.

seanmcleod, 2020-10-19:
I've been looking to model small quadcopters along the lines of the [DJI Phantom](https://www.dji.com/phantom) plus hybrid models like the [Wingcopter](https://wingcopter.com/) and the [Alti](https://www.altiuas.com/).

I did a bit of JSBSim work for [VTOL Technologies](http://www.vtol-technologies.com/) a couple of years ago and Marta Marimon mentioned that:

> We used the external forces and moments to model the rotors. 
> 
>  Basically, our main limitation was the thrust-vectoring. As far as I remember, at that point in time, it was not possible to change the direction of the thrust vector (engine tilt) using the built-in model. This was our main reason for making use of the external forces and moments. 
>
>The second reason was that we had our own tool to compute the rotors thrust. Due to thrust-vectoring and the A/C flight envelope, the rotor(s) thrust was not only function of RPM/Power, but also the vehicle TAS and the tilt angle of the rotor(s). Therefore, the built-in model was not compatible since was not initially conceived for this type of propulsion.
>
> As mentioned in the video you shared with me, the limitation of the built-in thrust/torque model was not a limitation at the end. Thanks to the flexibility of the external forces and moments capability and the way the configuration files are created, enabling different means to input data (equations, 3D tables, etc.), our ‘problem’ could be solved. Such flexibility is one of the advantages of JSBSim I believe. However, in the future it could be interesting to expand the built-in thrust/torque model and add a capability for drone propulsion, as more and more companies within the drone development use JSBSim.
>
> Another thing I vaguely remember was that we had to slightly modify the workflow of the execution of the different JSBSim modules, since, for us, the external forces and moments had the dependency of AOA, BETA… 

I came across a presentation by Matt Vacanti (@mvacanti) and Jesse Hoskins - [Open Source Workflow for Advanced Vehicle Dynamics Simulation](https://px4developersummitvirtual2020.sched.com/event/cjNx/open-source-workflow-for-advanced-vehicle-dynamics-simulation) which included the following slide:

[image]

So I asked Matt Vacanti to elaborate on what the core assumptions are that break down when dealing with small UAS models in JSBSim.

> A key point of clarification is that the issue is most pronounced when using the propeller / engine combination in a vertical lift scenario. 
>
> I ran into several problems and will dig up my old models to grab the gritty details but the problem is linked primarily to two issues:
>
> 1. [JSBSim overrides](https://github.com/JSBSim-Team/jsbsim/blob/b42f4238880ae1729acd55894381f74a0fe1ca13/src/models/propulsion/FGPropeller.cpp#L77-L78) the user entered moment of inertia for the propeller if it is less than 0.001 SLUG*FT2 so that it is always 0.001. This can be orders of magnitude too large for small propellers.
>
> 2. When using a jsbsim electric engine model, power is assumed to be linear to throttle input and [torque is derived](https://github.com/JSBSim-Team/jsbsim/blob/4fe1aa72cd4234824f686b9495194f2ef5556774/src/models/propulsion/FGPropeller.cpp#L200-L202).
>
>The combination of the two factors results in thrust and torque generation that significantly lags input commands (and also real world performance). This makes a small multicopter vehicle (vertical lift) completely unstable.
>
>I settled on the lookup table solution as I felt the likely use case for someone interested matching real-world performance would be to [measure actual system](https://www.rcbenchmark.com/) performance and apply coefficients as appropriate. There is a great wealth of small propeller data here: https://www.apcprop.com/technical-information/ which is where I extracted data for some of the models. The models I made publicly available are intentionally simple to make it easier for those picking it up to understand what is happening.
>
>If I were to create a new function for small vertical lift motors / props I think it would need to have several considerations:
>1. Battery Voltage / Current Capacity (and therefore a battery model)
>2. User defined Motor Power and Torque Curves

Taking an initial glance at the source code for `FGPropeller` I noticed in addition to the issue with propeller inertia being limited to a minimum of 0.001 SLUG*FT2 that Matt mentioned:

https://github.com/JSBSim-Team/jsbsim/blob/4fe1aa72cd4234824f686b9495194f2ef5556774/src/models/propulsion/FGPropeller.cpp#L78

I also noticed the following:

https://github.com/JSBSim-Team/jsbsim/blob/4fe1aa72cd4234824f686b9495194f2ef5556774/src/models/propulsion/FGPropeller.cpp#L53-L56

So the x-axis assumption is going to be an issue:

https://github.com/JSBSim-Team/jsbsim/blob/4fe1aa72cd4234824f686b9495194f2ef5556774/src/models/propulsion/FGPropeller.cpp#L270

https://github.com/JSBSim-Team/jsbsim/blob/4fe1aa72cd4234824f686b9495194f2ef5556774/src/models/propulsion/FGPropeller.cpp#L357

https://github.com/JSBSim-Team/jsbsim/blob/4fe1aa72cd4234824f686b9495194f2ef5556774/src/models/propulsion/FGPropeller.cpp#L211

bcoconni, 2020-10-20:
There are quite a number of interesting topics that you are bringing up here 😄

## Direction of the propeller axis
Regarding the propeller axis that is limited to the X-axis, you have to keep in mind that `FGPropeller` inherits from `FGThruster`.

`FGThruster` allows to set an arbitrary frame transformation to propellers and nozzles using the element `<orient>` (lines 95-98 below). And in addition the X direction can also be modified later in the simulation with the properties `propulsion/engine[*]/pitch-angle-rad` and `propulsion/engine[*]/yaw-angle-rad` (lines 99-102):
https://github.com/JSBSim-Team/jsbsim/blob/966098628f1b721ed1fd9926bdbd6e09b3f20b76/src/models/propulsion/FGThruster.cpp#L95-L102
So yes, computations are limited to the X-direction in the `FGPropeller` class but there is a frame transformation prior to that. And this frame transformation can be rigged at any time during the simulation by the user. It is for that reason that the air velocity vector is transformed from the body frame to the propeller frame at the very beginning of the method `FGPropeller::Calculate` (line 208):
https://github.com/JSBSim-Team/jsbsim/blob/966098628f1b721ed1fd9926bdbd6e09b3f20b76/src/models/propulsion/FGPropeller.cpp#L206-L208
As far as I know, the `<orient>` feature is mainly used to model the fact that propeller engines are generally tilted by a few degrees with respect to the aircraft X axis in some aircraft such as for the P-51D. I guess these values are chosen by the aircraft manufacturer to account for installation effects (aerodynamic interaction between the propeller air flow and the fuselage and wings as well as compensation of the aircraft incidence in cruise conditions).
https://github.com/JSBSim-Team/jsbsim/blob/966098628f1b721ed1fd9926bdbd6e09b3f20b76/aircraft/p51d/p51d.xml#L413-L426
The properties `propulsion/engine[*]/pitch-angle-rad` and `propulsion/engine[*]/yaw-angle-rad` are typically used by rockets model in closed loop control of the nozzle orientation
https://github.com/JSBSim-Team/jsbsim/blob/966098628f1b721ed1fd9926bdbd6e09b3f20b76/aircraft/J246/Systems/J246FirstStageEffectors.xml#L32-L44
It must also be noted that the modifications of the propeller thrust direction, if any, are assumed to take place at a very slow pace. Otherwise, the rotational velocities should be taken into account to compute the gyroscopic moment ; something that neither `FGThruster` nor `FGPropeller` are capable of at the moment.

## Propeller inertia lower limit
It is very likely that the lower limit 0.001 has been implemented to avoid null inertia which would obviously result in NaNs. In all likelihood, this value has been chosen arbitrarily and can be modified to a lower value as long as it is not zero.

Caution should be taken however about the size of the time step when very small inertias and masses are used. Historically, JSBSim has been designed for general aviation, civil and military aircraft and rockets. For such applications, the default time step of 120 Hz is generally a good choice. However for quadcopters, the time step might need to be chosen in the 1000 Hz so that the simulation produces plausible results.

## Electric motors
Indeed, JSBSim model of electric motor is extremely simplistic. In addition to the limitations that @mvacanti mentioned, there are also the following ones (and most likely some others):
* There is an upper limit to the current that can be extracted from the batteries and that should place an upper limit on the power and torque of the engine
* Even though an electric engine as a very good efficiency, it is not 100% but rather in the 90% (the rest is mostly dissipated in heat by the Joule effect).

Pull requests are welcome ! 😄

## Power transmission between propeller and engine
A topic that you did not mention but which is nonetheless a limitation of JSBSim propellers: the power transmission is one way, meaning that the engine drives the propeller but nothing can be transmitted from the propeller to the engine. This prevents windmilling and restart to be modelled for propeller engines and it also prevents [autorotation](https://en.wikipedia.org/wiki/Autorotation) to be modelled for copters.

bcoconni, 2020-10-20:
Regarding the implications of the last topic I mentioned above, @jonsberndt wrote an interesting use case about [an engine out failure on a twin engine aircraft C310](https://sourceforge.net/p/jsbsim/bugs/29).

seanmcleod, 2020-10-21:
> Pull requests are welcome

Will do, just getting more familiar with the code first.

> However for quadcopters, the time step might need to be chosen in the 1000 Hz so that the simulation produces plausible results.

I had noticed that AirSim uses 1KHz for their physics engine and the PX4 SITL - JSBSim bridge defaults to 250Hz.

seanmcleod, 2020-10-23:
I've updated the allowable minimum propeller inertia to 1e-6 slug*ft2 from the current 1e-3 slug*ft2.

The original value of 0.001 seemed potentially a bit arbitrary since both the minimum propeller diameter in feet was also set to 0.001 and so is the minimum gear ratio. 

In Aircraft Control and Simulation (Stevens & Lewis) 3rd edition they have a small (2.8lb) quadrotor model and list the propeller's inertia as 3e-5 slug*ft2.

Also came across an inertia value of 4.46e-5 for DJI Z-blade 9450 propellers.

So I've set the minimum allowable inertia value 1 order of magnitude smaller at 1e-6.

bcoconni, 2020-10-23:
Pushed. Thanks !

seanmcleod, 2020-11-29:
@mvacanti has been busy implementing a new electric engine model and testing it with small UAS models, take a look at his detailed report - 

https://github.com/mvacanti/jsbsim/blob/pr-bldc-validation/SUAS_BLDC.md

@rega0051 have you tested out your F450 model after I made the change to allowable minimum propeller inertia, see - https://github.com/JSBSim-Team/jsbsim/issues/253#issuecomment-714731927

I'll be interested to compare the difference in performance of the F450 when used with @mvacanti's changes.

rega0051, 2020-11-30:
@mvacanti   I did a quick check with my F450 model and your BLDC_validation build. It seems to work fine. I had temporarily put an actuator model in for the ESC to slow down the response to work with reduced prop inertia (v1.1+).

I had to stare at the write-up and code for a good long while before I could follow along. I think the 'torqueconstant' parameter has a misleading name, and cause me a lot of confusion. I'd suggest not even having 'torqueconstant' as an input, the value in the examples is really just a combination of several conversion factors (peak-peak Volts -> RMS, RPM -> rad/s, Netwons -> Lbs, meters -> feet). Ideally, with no losses, k_V and k_T would be the same. Instead, I think it would be good to have an efficiency as an input (that's the only reason I can think of that I would change the 'torqueconstant' value as it was defined anyway). Then internally compute k_T, defined in a more conventional way:

  RPM = Thruster->GetEngineRPM();
  Jxx = ((FGPropeller*)Thruster)->GetIxx();
  Torque = ((FGPropeller*)Thruster)->GetTorque();

  TorqueConstant = Efficiency * (3.0/2.0) * 1.0/sqrt(3.0) * (1.0 / (VelocityConstant * rpmtoradpsec)) * newtontopound * metertofeet;

  V = MaxVolts * in.ThrottlePos[EngineNumber];
  CommandedRPM = V * VelocityConstant;
  DeltaRPM = round(CommandedRPM - RPM);
  
  TorqueRequired = abs(Torque);
  CurrentRequired = TorqueRequired / TorqueConstant;
  TorqueAvailable = (TorqueConstant * MaxCurrent) - TorqueRequired;
  DeltaTorque = Jxx * (DeltaRPM * rpmtoradpsec) / max(0.00001, in.TotalDeltaT);

rega0051, 2020-11-30:
@seanmcleod the changes in the BLDC model include introduction of a new lever arm in the prop model. I thought the prop forces and moments were applied to the body correctly given the prop location and orientation. I thought the only thing that may be considered missing was if the rotor moved wrt the body frame.

seanmcleod, 2020-11-30:
> changes in the BLDC model include introduction of a new lever arm in the prop model

Yep I noticed that. Busy having a look at the references he mentioned in the report and the current JSBSim code.

seanmcleod, 2020-11-30:
@mvacanti, @rega0051, @bcoconni  I haven't worked on any of the JSBSim propeller code in the past, so I'm not that familiar with it, so there may be something I'm missing, but here is my review of the code to try and understand why @mvacanti felt the need to include a new lever arm into the calculations.

> Current JSBSim application of the propeller torque on the airframe does not take into consideration the location of the prop (lever arm) as described by BOUABDALLAH, MURRIERI, and SIEGWART here....

I assume the reference is to this?

[image]

Where the torque around the x and y axis is simply the difference in thrust times the moment arm l.

And the torque around the z axis is the sum of the propeller torques.

Now the JSBSim code calculates the thrust of the propeller:

```c++
Thrust = ThrustCoeff*RPS*RPS*D4*rho;
vFn(eX) = Thrust;
``` 

Plus the torque:

```c++
PowerRequired = cPReq*local_RPS*local_RPS*local_RPS*D5*in.Density;
vTorque(eX) = -Sense*PowerRequired / (local_RPS*2.0*M_PI);
```

Plus the angular  momentum for the gyroscopic effect:

```c++
FGColumnVector3 vH(Ixx*omega*Sense*Sense_multiplier, 0.0, 0.0);
```

Then the resultant moment combination of the gyroscopic effect and the propeller torque is calculated:

```c++
// Transform Torque and momentum first, as PQR is used in this
// equation and cannot be transformed itself.
vMn = in.PQRi*(Transform()*vH) + Transform()*vTorque;
```

Lastly the thruster's location relative to the cg will be used to generate an additional moment based on it's thrust. FGPropeller inherits from FGThruster which inherits from FGForce.

```c++
FGColumnVector3& FGForce::GetBodyForces(void)
{
  vFb = Transform()*vFn;

  // Find the distance from this vector's acting location to the cg; this
  // needs to be done like this to convert from structural to body coords.
  // CG and RP values are in inches

  vDXYZ = fdmex->GetMassBalance()->StructuralToBody(vActingXYZn);

  vM = vMn + vDXYZ*vFb;

  return vFb;
}
```

Now @mvacanti's proposed changes includes a new lever arm which is used in 2 places, multiplying the propeller torque by it and the angular momentum by it.

```c++
vTorque(eX) = (-Sense*PowerRequired / (local_RPS*2.0*M_PI)) * TorqueLeverArm;

FGColumnVector3 vH(Ixx*omega*Sense*Sense_multiplier*TorqueLeverArm, 0.0, 0.0);
```

@mvacanti I don't really follow the logic and reasoning for this new lever arm.

bcoconni, 2020-11-30:
@seanmcleod I have not yet read the document that @mvacanti wrote but I plan to.

I am in complete agreement with your analysis of the computation of the force and torques generated by a propeller. The lever arm between the propeller and the CG is indeed accounted for by JSBSim via the `FGForce`class.

mvacanti, 2020-12-01:
@seanmcleod @bcoconni @rega0051 Thank you for the feedback and support. I have validated the F450 with the same BLDC model and published the configuration [here](https://github.com/mvacanti/OpenFlightSim/tree/pr-bldc-F450/Simulation/aircraft/F450).

The results from the JSBSim PX4 Bridge Simulation for the F450 using the same environment etc as the earlier write-up are similarly successful as the hexacopter: 

[image]

@rega0051 please take a look at the modifications I made to the F450 model and let me know how that compares against the version you flew. 

@seanmcleod I agree that the lever arm should not be required. However, I have not been able to prove that the FGForce class is actually doing what we expect. I will devise a test to validate. Will be back shortly with more assessment.

mvacanti, 2020-12-03:
I have produced a validation that I believe demonstrates the FGForce class is not behaving as expected (also updated in the earlier documentation):

Configuration | Setting
------        | :------:
Environment   | [px4_jsbsim_docker](https://github.com/mvacanti/jsbsim/blob/pr-bldc-validation/px4_jsbsim_docker/Dockerfile)
PX4           | https://github.com/PX4/PX4-Autopilot.git
-- Branch     | master 
-- Revision   | 5d7ea62190bf6b64263fb53d3b1515bb0757a44b
JSBSim        | https://github.com/mvacanti/jsbsim.git
-- Branch     | pr-bldc-validation 
-- Revision   | fb7de0a093f0c9da084bd97229d15e013ca0a6bd
JSBSim Bridge | https://github.com/mvacanti/px4-jsbsim-bridge.git
-- Branch     | pr-jsbsim-bldc 
-- Revision   | e89153756262de9edc31040b843c4e859d89b301
-- Aircraft   | [hexarotor_x](https://github.com/mvacanti/px4-jsbsim-bridge/blob/c68c9ec704c5d8d232160828e39d766028227337/models/hexarotor_x/hexarotor_x.xml)
-- Engine     | [DJI-3510-380](https://github.com/mvacanti/px4-jsbsim-bridge/blob/c68c9ec704c5d8d232160828e39d766028227337/models/hexarotor_x/Engines/DJI-3510-380.xml)
-- Propeller  | [APC_13x8E_8K](https://github.com/mvacanti/px4-jsbsim-bridge/blob/c68c9ec704c5d8d232160828e39d766028227337/models/hexarotor_x/Engines/APC_13x8E_8K.xml)
-- Torque     | [Baseline](https://github.com/mvacanti/px4-jsbsim-bridge/blob/c68c9ec704c5d8d232160828e39d766028227337/models/hexarotor_x/Engines/APC_13x8E_8K.xml#L8-L9)
Flight Plan   | [yaw_test.plan](https://github.com/mvacanti/jsbsim/blob/pr-bldc-validation/doc/suas_bldc/yaw_test.plan)

The results of completing the flight plan above (climb to altitude then yaw to a heading) produces a .csv file that 
includes the following data:
- JSBSim standard Moments
- JSBSim standard Forces
- Torque produced by the propeller [here.](https://github.com/mvacanti/jsbsim/blob/ec9b7da0e4aa0d425e366b51f851df14a69461ed/src/models/propulsion/FGPropeller.h#L263)

Given BOUABDALLAH, MURRIERI, and SIEGWART (p. 175 (5)) we would expect that the Total N moment calculated by JSBSim
(absent other external forces) to equal the sum all propeller torque multiplied by the distance of the propeller to the 
center of gravity (C.G.). In the case of the hexarotor_x this distance is 2.53 Ft or 
[0.772M.](https://github.com/mvacanti/px4-jsbsim-bridge/blob/e89153756262de9edc31040b843c4e859d89b301/models/hexarotor_x/hexarotor_x.xml#L113)

[Plotting](https://github.com/mvacanti/jsbsim/blob/pr-bldc-validation/doc/suas_bldc/torque_validation_plots.py) the .csv file generate by JSBSim for this flight we observe that the Total N moment is equal to the sum of the 
total propeller torque. This indicates that the distance of the propeller to the C.G. is not being applied.
[image]

seanmcleod, 2020-12-03:
> expect that the Total N moment calculated by JSBSim (absent other external forces) to equal the sum all propeller torque multiplied by the distance of the propeller to the center of gravity (C.G.).

Why do you expect there to be a lever arm for this? The total N moment is the last entry in the following vector and there is no moment arm for it, unlike for the 1st two entries.

[image]

jonsberndt, 2020-12-03:
I agree with Sean, here. Torque is torque, and there is no moment arm to be applied. Unless I am missing something.

 

Jon

 

 <https://user-images.githubusercontent.com/824856/101018662-7fb55200-3574-11eb-8e8a-31458a5fbeb6.png>

bcoconni, 2020-12-04:
@mvacanti
>  we would expect that the Total N moment calculated by JSBSim (absent other external forces) to equal the sum all propeller torque multiplied by the distance of the propeller to the center of gravity (C.G.).

My teachers in physics always told me **"Check your units !"**. So if I read correctly your statement above, then it cannot be correct: a [torque](https://en.wikipedia.org/wiki/Torque) is expressed in *N.m* (i.e. a force multiplied by a distance), a distance is expressed in **meters** and a moment in **N.m** so there is no way you can sum torques multiplied by a distance (units: **N.m^2**) and get a moment (units: **N.m**)

mvacanti, 2020-12-04:
@seanmcleod @jonsberndt  @bcoconni Thank you, all fair feedback and inputs. 

Here is where I am really stuck:

The performance of the vehicle in yaw without this "multiple" of torque is unstable. Applying the multiple I have described instantly makes the vehicle perform "realistically" hence how I went down this rabbit hole. This could be caused by:

1. Control Laws
2. Calculation of Propeller Torque

What bothers me is that the appearance of performance is not off by a little, it is off by more than >2x in the case of the hexarotor example. 

I will keep digging.

seanmcleod, 2020-12-04:
@mvacanti if you're confident the Ct is correct or close enough what about the moment of inertia of the drone, maybe it's off by a factor of 2 or so?

rega0051, 2020-12-04:
My guess is Propeller Torque. The sim assumes that all the mechanical power produces Thrust, via the CP and CT values. The leftover power is all assumed to generate torque. Effectively, CQ = CP/(2*pi). This a fairly good assumption for a prop aircraft operating near its design condition. The static condition is only momentarily encountered; in the multicopter case the prop is nearly always operating near the static condition.

The traditional definition of prop efficiency would be: eff = CT * J / CP. We really want: eff = (CT*J + CQ*n) / CP. With this definition power loss would only be associated with acoustics, heat, turbulence, etc.

I pulled the APC dataset for the 9X4.5E at n= 2000 RPM. 
V = 0 mph
J = 0
Efficiency = 0 (ill-defined, clearly thrust is generated from the power)
Ct = 0.1157
Cp = 0.469
Pwr = 13.2 in lbf/s (sorry, units are horrible; and it looks like Pwr is computed from CP with a significant loss in precision)
Torque = 0.056 in-lbf
Thrust = 0.097 lbf

Looking at the power usage in this condition:
PwrTorque = Torque * n = 0.056 in-lbf * 133 (rad / s) = 7.5 (in lbf / s)
PwrThrust = Thrust * Vinduced = 0.097 lbf * 33.2 (in / s) = 3.2 (in lbf / s)

So, about 20% of the power is lost to the ether. There is a lot of ugly rounding and loss of precision due to the database though. Unfortunately, they don't provide the torque coefficient directly. I believe the code, as it exists now, would have put 100% of the power to torque generation in the form of prop acceleration; but if 13.2 of power were available we should really be steady-state (APC data is steady-state).

I think the approach should be to modify FGPropeller.c to use a C_Torque table (if provided) to compute the power required for the steady state thrust and torque, then compute the remaining power available for acceleration. Backward compatibility can be maintained by retaining the existing algorithm if a C_Torque table is not provided.

bcoconni, 2020-12-04:
@rega0051
> The traditional definition of prop efficiency would be: eff = CT * J / CP. We really want: eff = (CT * J + CQ * n) / CP. With this definition power loss would only be associated with acoustics, heat, turbulence, etc.

I may be wrong but CP and CQ are really 2 sides of the same coin which means that the relationship
<a href="https://www.codecogs.com/eqnedit.php?latex=C_P=2\pi&space;C_Q" target="_blank"><img src="https://latex.codecogs.com/gif.latex?C_P=2\pi&space;C_Q" title="C_P=2\pi C_Q" /></a>
is always true. Whether or not we assume some losses in the propeller and/or the motor, it is always true.

Furthermore, the relationship you suggest for the efficiency does not stand because you did not check your units :smile:
<a href="https://www.codecogs.com/eqnedit.php?latex=\eta=\frac{C_T&space;J&plus;C_Qn}{C_P}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\eta=\frac{C_T&space;J&plus;C_Qn}{C_P}" title="\eta=\frac{C_T J+C_Qn}{C_P}" /></a>
CT, CP, CQ and J are dimensionless while n has the dimension of rotations per second or *s^-1* so you cannot add CT\*J to CQ\*n.
> PwrThrust = Thrust * Vinduced = 0.097 lbf * 33.2 (in / s) = 3.2 (in lbf / s)

Where do you get this relationship from ? Froude's momentum theory states that an hovering rotor produces no thrust power even though the velocity through the propeller (V induced) is non zero.
> I think the approach should be to modify FGPropeller.c to use a C_Torque table (if provided) to compute the power required for the steady state thrust and torque, then compute the remaining power available for acceleration.

As I mentioned above I think this table would be redundant with `C_POWER` as `C_TORQUE` could be trivially obtained by dividing the table `C_POWER` by <a href="https://www.codecogs.com/eqnedit.php?latex=2\pi" target="_blank"><img src="https://latex.codecogs.com/gif.latex?2\pi" title="2\pi" /></a>.

rega0051, 2020-12-05:
My efficiency equation was intended as illustration, as you point out the units don't work. Sloppy, sorry. I wasn't advocating to redefine prop efficiency. The point was to highlight that efficiency only includes Thrust, but for multi-copters the torque component is essential for control.

The approach in FGPropeller may no longer represent a set of good assumptions. We're using data that wasn't generated based on conservation-based approaches. In particular, in the static case it seems to be significantly different (this isn't surprising to me at all). Which is where multi-copters operate, which may help explain why mvacanti is having stability issues. If that dataset is broadly representative it could result in a significant difference in control sensitivity, particularly in yaw.

I used: Thrust * Vinduced to estimate a power value for the axial flow, I think that's reasonable. I can only imagine that the statement about Froude's momentum conservation in hover is a consequence of defining advance ratio with the free-stream velocity rather than using the total velocity at the rotor disk. It's far more convenient to use the free-stream velocity, but it does make the static condition ill-defined. Clearly power is used to produce static thrust!

seanmcleod, 2020-12-05:
@mvacanti in your original email to me you mentioned making use of [measured data](https://www.rcbenchmark.com/) from rcbenchmark using their propeller test stands. In your report you compared the JSBSim calculated data to data from http://www.flybrushless.com/ and https://www.ecalc.ch/motorcalc.php however I noticed that they don't report torque data.

So we should be able to compare the JSBSim calculated toque, assuming a matching propeller etc., with some ground truth torque data from rcbenchmak, e.g. https://database.rcbenchmark.com/tests/qzy/multistar-elite-2306-2150kv-gemfan-6030r

bcoconni, 2020-12-05:
I have just re-read Stevens & Lewis' *"Aircraft Control and Simulation (3rd edition)"* especially the chapter 8.2 *"Propeller/Rotor Forces and Moments"*. If I am not mistaken, it seems that JSBSim has at least 2 things wrong regarding propellers generated forces and moments.
## Torque applied on the aircraft by the propeller
On figure 8.2-1 Stevens & Lewis show how the various torques add or subtract to influence the propeller and aircraft motions.
[image]
It should be noticed in particular that it is the torque generated by the motor that is affecting the aircraft motion (see the red arrow below):
[image]
However, in JSBSim, the torque is computed from the propeller aerodynamics.

First, the propeller power is computed from the `C_POWER` table:
https://github.com/JSBSim-Team/jsbsim/blob/05bd64aaaaaf32ab816bec0fa4b4df4a0e5c97c6/src/models/propulsion/FGPropeller.cpp#L294
then the torque is obtained by dividing by propeller angular rate
https://github.com/JSBSim-Team/jsbsim/blob/05bd64aaaaaf32ab816bec0fa4b4df4a0e5c97c6/src/models/propulsion/FGPropeller.cpp#L351-L352
and finally the torque is added to the aircraft moments (after being transformed from the propeller frame to the body frame)
https://github.com/JSBSim-Team/jsbsim/blob/05bd64aaaaaf32ab816bec0fa4b4df4a0e5c97c6/src/models/propulsion/FGPropeller.cpp#L280-L282

So what is implemented in JSBSim is the red arrow below:
[image]
As long as the propeller RPM is constant this does not make any difference. However, when the propeller accelerates or decelerates it does make a difference since we are underestimating the torque applied on the aircraft when accelerating (resp. overestimating it when decelerating).

According to that, I think that the torque `vTorque` should be computed from the engine power `EnginePower`:
```diff
--- a/src/models/propulsion/FGPropeller.cpp
+++ b/src/models/propulsion/FGPropeller.cpp
@@ -277,6 +277,8 @@ double FGPropeller::Calculate(double EnginePower)
 
   if (RPM < 0.0) RPM = 0.0; // Engine won't turn backwards
 
+  vTorque(eX) = -Sense*EnginePower / (RPS*2.0*M_PI);
+
   // Transform Torque and momentum first, as PQR is used in this
   // equation and cannot be transformed itself.
   vMn = in.PQRi*(Transform()*vH) + Transform()*vTorque;
@@ -349,7 +351,6 @@ double FGPropeller::GetPowerRequired(void)
   double local_RPS = RPS < 0.01 ? 0.01 : RPS; 
 
   PowerRequired = cPReq*local_RPS*local_RPS*local_RPS*D5*in.Density;
-  vTorque(eX) = -Sense*PowerRequired / (local_RPS*2.0*M_PI);
 
   return PowerRequired;
 }
```
## Air velocity in the propeller frame
Another strong assumption that JSBSim is making is that the aerodynamic velocity is uniform over the aircraft. In particular it means that the local air velocity at the propeller is not influenced by the position of the propeller. Strictly speaking this not true since air velocity is affected by the angular rates Pi, Qi, Ri of the aircraft relative to the inertial frame:

<a href="https://www.codecogs.com/eqnedit.php?latex=v_{rel}^P=v_{rel}^b&plus;\omega_{b/i}^b\times&space;p^b_{P/CM}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?v_{rel}^P=v_{rel}^b&plus;\omega_{b/i}^b\times&space;p^b_{P/CM}" title="v_{rel}^P=v_{rel}^b+\omega_{b/i}^b\times p^b_{P/CM}" /></a>
where
* <a href="https://www.codecogs.com/eqnedit.php?latex=v_{rel}^P" target="_blank"><img src="https://latex.codecogs.com/gif.latex?v_{rel}^P" title="v_{rel}^P" /></a> is the propeller air velocity
* <a href="https://www.codecogs.com/eqnedit.php?latex=v_{rel}^b" target="_blank"><img src="https://latex.codecogs.com/gif.latex?v_{rel}^b" title="v_{rel}^b" /></a> is the air velocity at the aircraft CG
* <a href="https://www.codecogs.com/eqnedit.php?latex=\omega_{b/i}^b" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\omega_{b/i}^b" title="\omega_{b/i}^b" /></a> is the aircraft angular rates (coordinates Pi, Qi, Ri) with respect to the inertial frame
* <a href="https://www.codecogs.com/eqnedit.php?latex=p^b_{P/CM}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?p^b_{P/CM}" title="p^b_{P/CM}" /></a> is the position of the propeller with respect to the aircraft CG expressed in the body frame.

However, in JSBSim, the air velocity is obtained from `FGAuxiliary`:
https://github.com/JSBSim-Team/jsbsim/blob/05bd64aaaaaf32ab816bec0fa4b4df4a0e5c97c6/src/FGFDMExec.cpp#L513-L514
then transferred to the thruster by `FGEngine` (the member `FGEngine::in` is a simple reference to `FGPropulsion::in`)
https://github.com/JSBSim-Team/jsbsim/blob/05bd64aaaaaf32ab816bec0fa4b4df4a0e5c97c6/src/models/propulsion/FGEngine.cpp#L149-L155
and finally used by `FGPropeller` after a simple transformation from the body frame to the propeller frame:
https://github.com/JSBSim-Team/jsbsim/blob/05bd64aaaaaf32ab816bec0fa4b4df4a0e5c97c6/src/models/propulsion/FGPropeller.cpp#L201-L206
It is therefore obvious that the term <a href="https://www.codecogs.com/eqnedit.php?latex=\omega_{b/i}^b\times&space;p^b_{P/CM}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\omega_{b/i}^b\times&space;p^b_{P/CM}" title="\omega_{b/i}^b\times p^b_{P/CM}" /></a> is ignored.

Fortunately that should not make much difference for an aircraft for which the angular rates Pi, Qi and Ri are small. However this might not be true for a small quadcopter that is capable of much more higher angular rates due to its small inertia.

[... cut ...]
 double FGPropeller::Calculate(double EnginePower)
 {
-  FGColumnVector3 localAeroVel = Transform().Transposed() * in.AeroUVW;
+  FGColumnVector3 vDXYZ = fdmex->GetMassBalance()->StructuralToBody(vActingXYZn);
+  FGColumnVector3 localAeroVel = Transform().Transposed() * (in.AeroUVW + in.PQRi*vDXYZ);
   double omega, PowerAvailable;
 
   double Vel = localAeroVel(eU);
```
Strictly speaking the turbulence `in.AeroPQR` should be subtracted from `in.PQRi` but we will leave that apart for a start.

Let me know what you all think.

bcoconni, 2020-12-05:
@rega0051
> The approach in FGPropeller may no longer represent a set of good assumptions. We're using data that wasn't generated based on conservation-based approaches. In particular, in the static case it [...]

Which assumptions are you referring to ? Could you be more specific ? In particular, I don't understand what you mean by "We're using data that wasn't generated based on conservation-based approaches." What do you call conservation-based approaches ?
> I used: Thrust * Vinduced to estimate a power value for the axial flow, I think that's reasonable. I can only imagine that the statement about Froude's momentum conservation in hover is a [...]

Forget what I said: it was rubbish. Froude's momentum theory indeed finds that a propeller produces/consumes power even while hovering. Actually, in most cases the power produced by a propeller is determined by the effect it has on the aircraft velocity (Thrust*Velocity) but it can also be determined by the effect it has on air: it pushes air down to produce thrust.

[image]

The thrust is equal to the difference of the momentum of air between the stream tube inlet and the stream tube outlet:<a href="https://www.codecogs.com/eqnedit.php?latex=T=\dot{m}(V_T^P&plus;2v_i-V_T^P)=\dot{m}2v_i" target="_blank"><img src="https://latex.codecogs.com/gif.latex?T=\dot{m}(V_T^P&plus;2v_i-V_T^P)=\dot{m}2v_i" title="T=\dot{m}(V_T^P+2v_i-V_T^P)=\dot{m}2v_i" /></a>
The work on the air flow upstream of the propeller during a time interval dt is:
<a href="https://www.codecogs.com/eqnedit.php?latex=W_1=\dot{m}V_T^P\times&space;(V_T^Pdt)=\dot{m}{V_T^P}^2dt" target="_blank"><img src="https://latex.codecogs.com/gif.latex?W_1=\dot{m}V_T^P\times&space;(V_T^Pdt)=\dot{m}{V_T^P}^2dt" title="W_1=\dot{m}V_T^P\times (V_T^Pdt)=\dot{m}{V_T^P}^2dt" /></a>
The work on the air flow downstream of the propeller during a time interval dt is:
<a href="https://www.codecogs.com/eqnedit.php?latex=W_2=\dot{m}(V_T^P&plus;2v_i)\times&space;(V_T^P&plus;2v_i)dt=\dot{m}(V_T^P&plus;2v_i)^2dt" target="_blank"><img src="https://latex.codecogs.com/gif.latex?W_2=\dot{m}(V_T^P&plus;2v_i)\times&space;(V_T^P&plus;2v_i)dt=\dot{m}(V_T^P&plus;2v_i)^2dt" title="W_2=\dot{m}(V_T^P+2v_i)\times (V_T^P+2v_i)dt=\dot{m}(V_T^P+2v_i)^2dt" /></a>

So the power produced by the propeller is
<a href="https://www.codecogs.com/eqnedit.php?latex=P=\frac{W_2-W_1}{dt}=\dot{m}\left[(V_T^P&plus;2v_i)^2-{V_T^P}^2\right]=\dot{m}2v_i(V_T^P&plus;2v_i)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?P=\frac{W_2-W_1}{dt}=\dot{m}\left[(V_T^P&plus;2v_i)^2-{V_T^P}^2\right]=\dot{m}2v_i(V_T^P&plus;2v_i)" title="P=\frac{W_2-W_1}{dt}=\dot{m}\left[(V_T^P+2v_i)^2-{V_T^P}^2\right]=\dot{m}2v_i(V_T^P+2v_i)" /></a>
or
<a href="https://www.codecogs.com/eqnedit.php?latex=P=T(V_T^P&plus;2v_i)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?P=T(V_T^P&plus;2v_i)" title="P=T(V_T^P+2v_i)" /></a>

For a hovering propeller we have <a href="https://www.codecogs.com/eqnedit.php?latex=V_T^P=0" target="_blank"><img src="https://latex.codecogs.com/gif.latex?V_T^P=0" title="V_T^P=0" /></a> and thus <a href="https://www.codecogs.com/eqnedit.php?latex=P=T2v_i" target="_blank"><img src="https://latex.codecogs.com/gif.latex?P=T2v_i" title="P=T2v_i" /></a>

seanmcleod, 2020-12-06:
> Let me know what you all think.

I read chapter 8 of Stevens & Lewis and agree with your observations in terms of the toque applied to the aircraft and the air velocity in the propeller frame.

In terms @mvacanti's issue with too small an N moment during the yaw the combination of underestimating the torque on the aircraft during RPM acceleration and overestimating when the RPM is decelerating means they'll more than likely roughly cancel out since half the propellers will be accelerating and half will be decelerating to produce the yawing moment, i.e. we won't see a ~2x increase in the N moment.

bcoconni, 2020-12-06:
> In terms @mvacanti's issue with too small an N moment during the yaw the combination of underestimating the torque on the aircraft during RPM acceleration and overestimating when the RPM is [...]

On the contrary, since the yawing moment is obtained by say decreasing power of clockwise rotating propellers and increasing power of counterclockwise propellers then the variation of each rotor will add up (since we are subtracting values of opposite sign). This may or may not fix the problem reported by @mvacanti but the effect will definitely not cancel out.

bcoconni, 2020-12-06:
> I read chapter 8 of Stevens & Lewis and agree with your observations in terms of the toque applied to the aircraft and the air velocity in the propeller frame.

Great ! These modifications are now implemented in the `master` branch (commit 22f8330).

seanmcleod, 2020-12-06:
> On the contrary, since the yawing moment is....

You're right, I wasn't thinking straight in terms of the signs 😉

mvacanti, 2020-12-08:
@seanmcleod and @bcoconni Thanks for the great discourse and updates. I will pull the commit and report back on the performance. I have begun investigating how it may be possible to use [open source real world vehicle log data](https://review.px4.io/browse) as a reference for validation purposes as well.

seanmcleod, 2020-12-08:
@mvacanti when I took at https://www.rcbenchmark.com/ I couldn't find any tests using APC propellers in terms of trying to get some ground truth data for torque.

bcoconni, 2020-12-11:
I completely forgot to mention that JSBSim computation of the sign of the gyroscopic effect has been erroneous until April 2015 and has been fixed in the commits 6b7a2103bc8b47c62c8bab17fbe7 and efef744d4c7ef8c946f185add3

Since this bug has existed for a long time, it has been decided to fix the problem by adding a `version` attribute to the propeller section of the aircraft XML definition. When the `version` attribute is specified to a value strictly higher than 1.0, JSBSim computes the correct sign of the gyroscopic effect. Otherwise JSBSim uses the historical *erroneous* sign for the gyroscopic moment.
```xml
<propeller name="xyz" version="1.01">
  <!-- Propeller definition goes here -->
</propeller>
```
So it is strongly recommended to specify the `version` attribute to the propeller definition so that the correct sign of the gyroscopic effect is used.

rega0051, 2021-02-04:
I made a PR in my fork: https://github.com/rega0051/jsbsim/pull/1

I'm more accustomed  to a different FDM system, so I hadn't fully realized how the motor and prop models were coupled in JSBSim, I had a weird mixed interpretation a few months ago. I can't even decipher some of the things I said. So, I started a fresh look.

I started with what @mvacanti had provided for the BLDC, and reduced it down a lot.  The characteristic that I desire is that of a first-order system, so rather that use a feed forward like command and torque limits, I instead put the first-order lag on the values from the Prop model; so the RPM measured is filtered. Since I really only care about the response of the system (not interested in modeling the motor controller itself) it seems reasonable to just treat the closed-loop response of the system as first-order with a user supplied time constant, tau.

With these goals in mind it seemed reasonable to change FGElectric. I added two XML parameters; maxrpm and tau. If maxrpm > 0, then the input command is treated as a proportion of maxrpm and prop RPM is the feedback. If maxrpm == 0, then max power is the scaling factor, and prop power required is the feedback. The other parameter, tau, sets the time constant of the first-order filter on feedback; whether its prop RPM or power required. Backward compatibility is preserved if neither maxrpm or tau are set.

I tweaked my F450 model to use the changes. The flight control has the same inner-loop PID gains as PX4, so it should be fairly representative of the close-loop system (minus transport delays). I set the maxrpm to 16000, which is kV * max voltage (960 RPM/Volt * 16.8 Volt). I set tau to 0.100 sec, which is roughly what I've seen in bench tests (it could be anywhere from 0.050 sec to 0.150 sec, want to test specifically for bandwidth after pandemic).

F450: https://github.com/UASLab/OpenFlightSim/tree/master/Simulation/aircraft/F450
Motor Model: https://github.com/UASLab/OpenFlightSim/blob/master/Simulation/engine/DJI_E305.xml
Test Script: https://github.com/UASLab/OpenFlightSim/blob/master/Simulation/scripts/Test_F450_Launch.xml

I still don't have a nice PX4 setup yet so if @mvacanti could check it out I'd be appreciative. I also want to mimic that "hold-down" bench check.

bcoconni, 2021-02-05:
@rega0051
> The characteristic that I desire is that of a first-order system, so rather that use a feed forward like command and torque limits, I instead put the first-order lag on the values from the Prop [...]

I am not very fond of these kind of C++ hardcoded phenomenological models. We already have this kind of model for `FGTurbine` and what we have ended up with are lots of PR from people asking for being able to tweak the model to their need (PR #121, #139, #188, #382, #389 - and that's only for the last 2 years) and there are plans for more (see discussion #389).
> With these goals in mind it seemed reasonable to change FGElectric.

I don't think so. `FGElectric` is a trivial model: its power output is proportional to the throttle position. I am pretty sure you can use XML code to assign a first order lag of the throttle command and drive the engine power through the throttle position.

Most people underestimate the actual power of `FGElectric` because it's so trivial. But it's the only engine model in JSBSim that gives you the full control over your engine. You can theoretically code in XML any kind of motor/engine with `FGElectric` !

rega0051, 2021-02-05:
@bcoconni Phenomenological models are completely appropriate for flight dynamics models. That's what the whole sim is! The details to model a real motor requires parameters that are not readily knowable. For electric motor (actuators or for propulsion) we won't know the internal control system the controllers are using, nor should we need a slew of parametric values (motor velocity constant, inductance, resistance, controller capacitance, resistance, frame rate, gains, etc.). Knowing the response function should be adequate for a FDM.

The filter in feedback is not the same as a filter on the input. I previously had a FGActuator upstream of the motor command, and it does at least prevent the control system from using the infinite bandwidth of the motor. Is there another way to introduce limited bandwidth in the Prop and Motor interaction? The use of throttle position as speed command is tied to the feedback signal, which is why I ended up putting that in there as well. I think this is a reasonable incremental improvement.

bcoconni, 2021-02-05:
> Phenomenological models are completely appropriate for flight dynamics models. That's what the whole sim is! The details to model a real motor requires parameters that are not readily knowable. For [...]

I think you misread my statement and missed the first 2 words in _"**C++ hardcoded** phenomenological models"_. I never said phenomenological models should not be used in flight dynamics. I am saying such models should not be hardcoded in the C++ code of JSBSim. That's quite a different statement !

There are a lot of places in JSBSim where no assumption are made about the model including whether it is phenomenological or not. Aerodynamics is one such place where you can use, at your convenience, a linear relationship between lift and AoA or a very complex function that depends on a lot of parameters. I see no reason why this could not be the case for electric motors.
> The filter in feedback is not the same as a filter on the input. I previously had a FGActuator upstream of the motor command, and it does at least prevent the control system from using the infinite [...]

I may miss something but I see no difference between your C++ code and the flight control system below (granted that the properties `propulsion/engine[]/propeller-torque` and `propulsion/engine[]/propeller-power` are missing at the moment - but this could be easily fixed). So I am struggling to see the distinction you are making between filter in feedback and filter on the input.

For better legibility, I have omitted the min/max computations to ensure that RPM is positive and lower than the max value `maxRPM`.

```xml
<system name="motor control">
  <property name="maxRPM"> 10000. </property>
  <property name="maxPOWER"> 10. </property>
  <channel name="motor bandwith">

    <!-- First order lag on RPM measure -->
    <lag_filter name="filtered-rpm">
      <input>propulsion/engine[0]/engine-rpm</input>
      <c1> 2.0 </c1> <!-- tau -->
    </lag_filter>

    <!-- Change in RPM -->
    <fcs_function>
      <function name="delta-rpm">
        <difference>
          <product>
            <property>fcs/throttle-cmd-norm[0]</property>
            <property>maxRPM</property>
          </product>
          <property>filtered-rpm</property>
        </difference>
      </function>
    </fcs_function>

    <!-- Power Command: DeltaPower = DeltaRPM * TorqueReq -->
    <fcs_function>
      <function name="cmd-power">
        <sum>
          <product>
            <property>propulsion/engine[0]/propeller-torque</property>
            <property>delta-rpm<property>
            <value>0.104719</value> <!-- rpm to rad/sec -->
          </product>
          <property>propulsion/engine[0]/propeller-power</property>
        </sum>
      </function>
    </fcs_function>

    <!-- Convert power command to throttle position -->
    <fcs_function>
      <function name="fcs/throttle-pos-norm[0]">
        <quotient>
          <property>cmd-power</property>
          <property>maxPOWER</property>
        </quotient>
      </function>
    </fcs_function>
  </channel>
</system>
```

rega0051, 2021-02-05:
@bcoconni The value for Tau isn't hard-coded. You have an issue with it just being first-order? I'd be happy with a FGFilter instance; but even FGFilter is hard-coded to some degree.

**Input filtering is not the same as feedback filtering**. Lets define the Motor model as K(s) (controller), Prop model as G(s) (plant) in feedback, the and a filter as F(s). The comp sensitivity function with F acting on the input is: T(s) = inv(1 + K(s)G(s))*K(s)F(s); instability at K(s)G(s)=-1. With F(s) in the feedback T(s) = inv(1+K(s)G(s)F(s))*K(s); instability at K(s)G(s)F(s)=-1. 

There is also a case to be made that some damping in the numerics would be helpful. Right now there is no damping mechanism between the Prop and Motor, the bandwidth is limited only by the sim dt value. We've seen cases where the RPM jumps to huge values because of numeric sensitivity.

bcoconni, 2021-02-06:
> The value for Tau isn't hard-coded. You have an issue with it just being first-order? I'd be happy with a FGFilter instance; but even FGFilter is hard-coded to some degree.

I am obviously not making myself clear, sorry about that.  What I am trying to explain is that **you don't need to modify the C++ code of JSBSim for your purpose**.

Whether you are using your own first order filter, `FGFilter`, `FGActuator` or any other control system that exists under the sun is irrelevant to the point I am trying to make. The XML code that I suggested above is the equivalent of your C++ code. I just tried to mimic the behaviour of [your C++ code](https://github.com/rega0051/jsbsim/pull/1), no more no less. Whether or not a first order filter fits your purpose is you to tell, not me.

Once again, I am trying to show you that you can get your phenomenological model in your UAV FDM without altering the C++ code of JSBSim (except perhaps by adding 2 new properties `propulsion/engine[]/propeller-torque` and `propulsion/engine[]/propeller-power` that are currently not available)

The problem being that if a first order filter is coded in `FGElectric` then sooner or later someone will want to alter the C++ code to add, say a second order filter, etc. Fortunately we can already avoid this situation upfront because you (@rega0051) don't need to modify the C++ code of JSBSim to simulate your UAV motors with a first order filter.
> **Input filtering is not the same as feedback filtering**. Lets define the Motor model as K(s) (controller), Prop model as G(s) (plant) in feedback, the and a filter as F(s). The comp sensitivity [...]

Thanks for the control theory refresher :+1: :smiley: 

However my problem remains the same: there is no feedback from `FGPropeller` to `FGElectric`. Check the code by yourself:
https://github.com/JSBSim-Team/jsbsim/blob/33d429e783589c27dccf2d70d231efe98c38f1df/src/models/propulsion/FGElectric.cpp#L86-L103
The motor power does not depend on any input from the propeller (see line 97) so there is no feedback and I don't see how your comment _"Input filtering is not the same as feedback filtering"_ might be any relevant to the subject.

Or may be you did not see that there are two different properties of the controls : one for the command, one for the position ? `FGElectric` uses the throttle **position** as an input and the whole point of the XML code I made is to compute the throttle **position** `fcs/throttle-pos-norm[0]` from the throttle **command** `fcs/throttle-cmd-norm[0]` to drive the motor output power.
> There is also a case to be made that some damping in the numerics would be helpful. Right now there is no damping mechanism between the Prop and Motor, the bandwidth is limited only by the sim dt value. We've seen cases where the RPM jumps to huge values because of numeric sensitivity

I'd like to see examples before jumping to the conclusion that numerical damping is needed.

rega0051, 2021-02-06:
The current state is that all multi-copter models that we've been able to find use External Forces rather than the Propulsion models. I'm just trying to address one of the issues that underlies that reality. I'm saying there should be feedback in that system. I thought a PR would be a good course of action. I'm also not implying that this PR will fix every inevitable issue, nor does it address all the issue that have been brought up in issue 333. I just felt it was a doable increment that preserved backward compatibility could be a nice thing. I believe it captures the characteristic that the proposed BLDC model obtained without requiring a step-wise dynamic inversion as feedback (https://github.com/mvacanti/jsbsim/blob/e6d58b2268c1609b15f362a04520eac9bb3321d5/src/models/propulsion/FGBldc.cpp#L104) and limits to regulate the response (https://github.com/mvacanti/jsbsim/blob/e6d58b2268c1609b15f362a04520eac9bb3321d5/src/models/propulsion/FGBldc.cpp#L106).

A spike in RPM is shown in the nice writeup that @mvacanti did: https://github.com/mvacanti/jsbsim/blob/pr-bldc-validation/SUAS_BLDC.md#propeller-and-engine-torque

bcoconni, 2021-02-06:
> I'm saying there should be feedback in that system.

Did I say otherwise ?

seanmcleod, 2021-02-06:
@rega0051 do you agree or not that @bcoconni's XML example he provided matches/mimics your C++ code that you implemented in your PR?

If it does then he's saying that a JSBSim user wanting to model a multi-copter could use the existing `FGElectric` engine and make use of the XML code he's provided to get the same results as compared to using a modified `FGElectric` model along the lines of your PR.

So if that is the case then are you arguing it would be more convenient for a JSBSim user to make use of your modified `FGElectric` model and providing `MaxRPM` and `tau` in their engine file as opposed to including @bcoconni's XML snippet in their model?

rega0051, 2021-02-06:
@seanmcleod and @bcoconni on Monday I'll do a comparison and add a PR with the additional Get/Set methods. I'm guessing it will be equivalent, or could be made equivalent. I'd really like other folks who thinks they have a working multi-copter model to evaluate and compare as well.

Two lines in the motor definition is more convenient for me than replication of scripting, but I understand the maintenance issue as well. I do believe that all uses of FGElectric should be implementing a time constant or bandwidth limit in feedback (or there should be another relaxation parameter introduced to eliminate the huge RPM spike). And using external forces to mimic the propulsion system works as well... How do you decide were to draw the line on interfacing sub-systems? All options seem like an incomplete band-aide. I want to help, but not waste my time or everybody else's.

rega0051, 2021-02-10:
PR #394 exposes additional properties in FGPropeller:
  propulsion/engine[#]/propeller-power
  propulsion/engine[#]/propeller-torque
  propulsion/engine[#]/propeller-sense

An example of a F450 frame based quad copter is below, along with the required propulsion system definitions. The desired command to the ESC is proportional to the max RPM of the motor. Required propeller torque is used to provide an appropriate scale change from normalized RPM to normalized power for the motor (as required by FGElectric). Max power ends up being defined in two places, in the 'Effectors' system and in the 'electric_engine' below (max motor power could be exposed as a property in FGElectric to avoid replication).  propeller sense (clockwise or counter-clockwise) is defined only in the propulsion defintion, and is pulled into the 'Effectors' systems below through exposing propeller-sense from FGPropeller.

```xml
<system name="Effectors">
    <property value="16000.0"> fcs/maxRPM </property> <!-- Max motor RPM -->
    <property value="320.0"> fcs/maxPwr_Watt </property> <!-- Max motor power -->
    <property value="20"> fcs/lagMotor </property> <!-- lag = tau_s / dt -->
    
  <channel name="Electronic Speed Conrollers">
    <!-- First order lag on RPM measure -->
    <lag_filter name="fcs/filtMeasFR_rpm">
      <input> propulsion/engine[0]/propeller-rpm </input>
      <c1> fcs/lagMotor </c1>
    </lag_filter>
    <lag_filter name="fcs/filtMeasAL_rpm">
      <input> propulsion/engine[1]/propeller-rpm </input>
      <c1> fcs/lagMotor </c1>
    </lag_filter>
    <lag_filter name="fcs/filtMeasFL_rpm">
      <input> propulsion/engine[2]/propeller-rpm </input>
      <c1> fcs/lagMotor </c1>
    </lag_filter>
    <lag_filter name="fcs/filtMeasAR_rpm">
      <input> propulsion/engine[3]/propeller-rpm </input>
      <c1> fcs/lagMotor </c1>
    </lag_filter>

    <!-- Change in RPM:
    deltaCmd_rpm = cmdEsc_nd * maxRPM + filtMeas_rpm -->
    <fcs_function>
      <function name="fcs/deltaCmdFR_rpm">
        <difference>
          <product>
            <property> fcs/cmdEscFR_nd </property>
            <property> fcs/maxRPM </property>
          </product>
          <property> fcs/filtMeasFR_rpm </property>
        </difference>
      </function>
    </fcs_function>
    <fcs_function>
      <function name="fcs/deltaCmdAL_rpm">
        <difference>
          <product>
            <property> fcs/cmdEscAL_nd </property>
            <property> fcs/maxRPM </property>
          </product>
          <property> fcs/filtMeasAL_rpm </property>
        </difference>
      </function>
    </fcs_function>
    <fcs_function>
      <function name="fcs/deltaCmdFL_rpm">
        <difference>
          <product>
            <property> fcs/cmdEscFL_nd </property>
            <property> fcs/maxRPM </property>
          </product>
          <property> fcs/filtMeasFL_rpm </property>
        </difference>
      </function>
    </fcs_function>
    <fcs_function>
      <function name="fcs/deltaCmdAR_rpm">
        <difference>
[... 291 lines cut ...]
        <yaw>0</yaw>
      </orient>
      <sense>-1.0</sense>
      <p_factor>0.0</p_factor>
    </thruster>
  </engine>

  <engine file="DJI_E305" name="aft right">
    <thruster file="DJI_9450">
      <location unit="M">
        <x>0.1651</x>
        <y>0.1651</y>
        <z>0.025</z>
      </location>
      <orient unit="DEG">
        <roll>0</roll>
        <pitch>90</pitch>
        <yaw>0</yaw>
      </orient>
      <sense>-1.0</sense>
      <p_factor>0.0</p_factor>
    </thruster>
  </engine>
</propulsion>
```

bcoconni, 2021-02-13:
Great ! :+1:

Just a couple of tips:
* You can specify the opposite of a property in single line. So the product of
```xml
  <value> -1 </value> <!-- Torque is negative of prop sense -->
  <property> propulsion/engine[1]/propeller-torque </property>
```
can be shortened to (note the minus sign before the property name):
```xml
  <property> -propulsion/engine[1]/propeller-torque </property>  <!-- Torque is negative of prop sense -->
```
* `<fcs_function>` can use `<clipto>` arguments. So you can replace the pair `<fcs_function>` and `<actuator>`:
```xml
    <fcs_function>
      <function name="fcs/cmdPowerFR_nd">
        <quotient>
          <property> fcs/cmdPowerFR_Watt </property>
          <property> fcs/maxPwr_Watt </property>
        </quotient>
      </function>
    </fcs_function>

  <!-- ... -->

    <actuator name="Front Right">
      <input> fcs/cmdPowerFR_nd </input>
      <clipto>
        <min>0.01</min> <!-- Min command shouldn't be zero, commands would be stuck at zero -->
        <max>1.0</max>
      </clipto>
      <output> fcs/throttle-cmd-norm[0] </output>
    </actuator>
```
by a single `<fcs_function>`:
```xml
    <fcs_function>
      <function name="fcs/cmdPowerFR_nd">
        <quotient>
          <property> fcs/cmdPowerFR_Watt </property>
          <property> fcs/maxPwr_Watt </property>
        </quotient>
      </function>
      <clipto>
        <min>0.01</min> <!-- Min command shouldn't be zero, commands would be stuck at zero -->
        <max>1.0</max>
      </clipto>
      <output> fcs/throttle-cmd-norm[0] </output>
    </fcs_function>
```

bcoconni, 2021-02-22:
> PR #394 exposes additional properties in FGPropeller:
  propulsion/engine[#]/propeller-power
  propulsion/engine[#]/propeller-torque
  propulsion/engine[#]/propeller-sense

It finally made its way to the `master` branch. Thanks @rega0051 for your patience (and perseverance 😉).

bcoconni, 2021-12-08:
Hi @pbecchi and thanks for your interest in JSBSim.

I think C++ code is the best option since other engines are coded in JSBSim that way. Basically, you have 2 options:
1. Improve the existing electric engine by adding some features to it https://github.com/JSBSim-Team/jsbsim/blob/master/src/models/propulsion/FGElectric.cpp
2. Build a new engine model just as @mvacanti did in his fork of JSBSim: https://github.com/mvacanti/jsbsim/blob/pr-bldc-validation/src/models/propulsion/FGBldc.cpp

bcoconni, 2021-12-11:
> I only have some C++ background and I need, if possible to get some guidance on how to interface the code, compile and build JSBSIM on VisualStudio.

Sure. For a start, you can find some guidance in [our docs to build JSBSim with Visual Studio](https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/quickstart-building-using-visualstudio/). 

> I will prepare a snippet of the code in my next message , i will appreciate comments and corrections.

We'll be glad to answer your questions.

In order to keep this issue focused on its topic (propellers, quadcopters, etc.), please [open a separate discussion](https://github.com/JSBSim-Team/jsbsim/discussions) regarding Windows/Visual Studio/compilation or any other topics that are unrelated to this ticket.

pbecchi, 2021-12-13:
@bcoconni Thanks again for your help!
I have been able to modify the @mvacanti FGBldc.cpp, to compile and build and to run some of @mvacanti motor benchmark.

Results are good enought : static performances are very similar but for same throttle input resulting RPM is lower as it was expected since a % of the input power is taken by internal losses. Transition from one state to next one result in a small peak in torque given by the prop RPM accelerations:
[image]

The changes to FGBldc are quite simple : it is only necessary to input 2 new properties:
 <coilresistance>0.08</coilresistance>
  <noloadcurrent>1.8</noloadcurrent>
The values can by easily found on motor specs or available motor data base (ecalcl.ch).

Now i would like to test the new FGBldc code on a multirotor example .... how I should proceed? There is an existing multicopter model I can test with?

rega0051, 2021-12-13:
@pbecchi for an existing multicopter model you should look at the DJI F450 that I was using earlier. 
[DJI F450 Airframe](https://github.com/UASLab/OpenFlightSim/tree/master/Simulation/aircraft/F450)
[DJI E305 Motor](https://github.com/UASLab/OpenFlightSim/blob/master/Simulation/engine/DJI_E305.xml)
[DJI 9450 Prop](https://github.com/UASLab/OpenFlightSim/blob/master/Simulation/engine/DJI_9450.xml)

Note that in F450.xml you can change: <system file="EffectorsClosedLoop.xml"/> to <system file="EffectorsClosedLoop.xml"/> in order to remove the loop closure I made around the ESC and motors to get a desired frequency cutoff.

It's good to see that you're using internal resistance and no load current. I had elected to go with just a cut-off frequency because I believe the motor model is otherwise a too stiff ODE and the actual response of a real physical system doesn't have response out nearly high enough to cause issues running JSBSim with a typical dt. Any kind of damping and high-frequency cutoff would help that situation, hopefully the physical motor parameters will be sufficient to mitigate the stiffness without warranting the outer-loop closure method that I ended up using. Both internal resistance and no load current are pretty easy to measure even if the spec sheets don't always list them.

pbecchi, 2021-12-14:
@rega0051 
I am trying to run the F450 quadcopter script, but ,since I am new to JSBSIM, I have problems to fully undestrand and modify effector_closedloop.xml.
It look like propeller RPM are computed there while they should be computed by FGBldc code that expect as an input `ThrottlePos[]` . But it  do not appear to be defined anywhere.
 I guess that all caculations of propeller properties should be commented out and `in.ThrottlePos[EngineNumber]` definition should be added...... but how?
Can you help me?

bcoconni, 2021-12-14:
> I have been able to modify the @mvacanti FGBldc.cpp, to compile and build and to run some of @mvacanti motor benchmark.

Great ! Thanks.

> Results are good enought : static performances are very similar but for same throttle input resulting RPM is lower as it was expected since a % of the input power is taken by internal losses. [...]

The motor output power `/fdm/jsbsim/propulsion/engine/power-hp` seems to be oscillating after t=45s. Is that the expected behaviour ?

pbecchi, 2021-12-15:
Hi guys
I am slowly improving in my learning curve: 
I find the solutions for my previous message questions and I succeed on running some simulations.
I had to increase throttle to avoid retouching the ground and now it perform all the simulation without unexpected events. This what I get:
[image]
I have nothing to compare with and I am unable to judge if BLDC motor perform as it should.....

I will appreciate your comments!
........and thanks for your help and encouragement!

bcoconni, 2021-12-18:
@pbecchi, unless I'm mistaken you have not brought an answer to my question: why is the motor power output oscillating in the previous chart you've shown ? Given the simple model you told you're using, this is quite unexpected.

In addition, I don't see any fork of JSBSim under your GitHub user name so it's quite difficult to provide you with further comments if we can't review your code.

pbecchi, 2021-12-18:
@bcoconni , you are right! Sorry I was busy with the F450 tests....
Anyhow this oscillation is due to a numerical instability of the code in these lines:
`TorqueAvailable = MaxTorque - TorqueRequired;
  DeltaTorque = (((DeltaRPM/60)*(2.0 * M_PI))/(max(0.00001, in.TotalDeltaT))) * ((FGPropeller*)Thruster)->GetIxx();`
`TorqueAvailable = MaxTorque - TorqueRequired;
if (DeltaRPM >= 0){
TargetTorque = min(DeltaTorque, TorqueAvailable) + TorqueRequired;
} else{
TargetTorque = min(abs(DeltaTorque), TorqueRequired) * -1;
}
EnginePower = ((2 * M_PI) * max(RPM, 0.0001) * TargetTorque) / 60;`

Since i am a beginner and i have not any knowledge of JSBSim, I have decided to use the existing code of FGBLdc.cpp written by @mvacanti with minimal modifications. I have worked using his fork and i have run his test scripts.
The FGBldc code his here
[FGBldc.txt](https://github.com/JSBSim-Team/jsbsim/files/7739573/FGBldc.txt)
All my modification are commented.
As you can see the lines that create the instability are an @mvacanti eritage.
I really dont see any need for these lines of code but.....since they where in, i decided to keep it. 
Now I will make a fork of your repo, I will add the code that I have modified from @mvacanti fork and all test scripts . This way the fork will be alligned to your commits and this will give you the possibility to access and review.
Anyhow ..... thanks for your comments.

pbecchi, 2021-12-20:
@bcoconni 
I have corrected the code of FGBLdc.cpp and run some tests. Everything is available on my fork: https://github.com/pbecchi/jsbsim .
Basically I have revised the lines of my previuos message to better control deceleration phase.
The torque in acceleration is given by the propeller aerodynamic torque + the propeller acceleration * prop inertia and is limited by max available motor torque.
During deceleration the can be set on the Electronic Speed Controller and vary from zero braking to maximum braking (the motor generated current is absorbed by internal resistance).   @mvacanti was assuming to simply change the direction of max torque applying a huge braking force . This was the reason of the torque (power) fluctuations.
I have now changed the code assuming that when you reduce the speed the applied torque drop to the reduced torque necessary for the new prop speed, this means no braking.
As you can see the resulting curves are much smoother:
[image]
The braking take few seconds. To adjust breaking time it may be necessary to add a external parameter to be added to motor properties.
I have also rerun the scripts for testing quadrotor F450 and results seams very good (for what i can judge!):
[image]

@bcoconni  @rega0051  I will appreciate your comments! 
Then I will finish and cleanup the code.

bcoconni, 2021-12-23:
Thanks for the updates @pbecchi. This looks very good.

I was a bit surprised that the motor accelerates much faster than it decelerates so I had a look to your code and found the following statement:

```c++
if (DeltaRPM >= 0) {
  TargetTorque = min(DeltaTorque, TorqueAvailable) + TorqueRequired;
}
else {
  TargetTorque = TorqueRequired  - abs(DeltaTorque)*abs(DeltaRPM)/max(RPM,0.001);
  TargetTorque =  TorqueRequired*(CommandedRPM / max(RPM,0.01))* (CommandedRPM / max(RPM, 0.01));
}
```
The different treatment based on the sign of `DeltaRPM` is unexpected. You may have given the explanation about that result but I'm not sure about what you meant since a word seems to be missing in one of your statement:
> During deceleration the ??? can be set on the Electronic Speed Controller

pbecchi, 2021-12-23:
Thaks for your comment  @bcoconni , in general acceleration and deceleration of a BLDC motor are not comparabele.
But in the case of the above plot your comment is correct:
in this case  min deceleration is applied equivalent to no braking torque (see the orange line).  
Since in most cases a more reactive motor control is desired, as anticipated in previous post, I have modified the statement 
`TargetTorque =  TorqueRequired*(CommandedRPM / max(RPM,0.01))* (CommandedRPM / max(RPM, 0.01));`
with a new one
 ` TargetTorque = TorqueRequired  - min(abs(DeltaTorque)/(max(deceleration_time,0.01)*30),RPM*TorqueConstant/VelocityConstant/VelocityConstant/coilResistance);`

where you can set the desired deceleration time as you normally do programming the ESC (??? word is "time").
A value of about 1 second  will lead to a much shorter sweep and will mantain a stable behaviour,
Regarding acceleration i think that the code is correct and give the maximum acceleration that it is  phisically achievable with the motor. This acceleration tested on the F450 copter give result that I consider in line with normal performances of a similar UAV #253.
I will cleanup the code and update shortly the fork on github.com/pbecchi.
Meanwhile i am trying to add the Dj F450 on PX4 sitl and to simulate a fly , it look possible but .............challenging!

Jaeyoung-Lim, 2022-01-04:
@pbecchi Would you be interested in making a PR to https://github.com/Auterion/px4-jsbsim-bridge ?

Regarding logging, we can try to add those in the jsbsim bridge

pbecchi, 2022-01-04:
@Jaeyoung-Lim 
Yes I can do it..... but keep in mind that I use my fork of jsbsim: in github.com/pbecchi/jsbsim.
There I have added the code for the new BLDC electric motor. 
So or you use my fork or I should make a PR also for JSBSim-Team/jsbsim.

seanmcleod, 2022-01-04:
> In jsbsim output is specified in the launch script, but here ,as far as i understand, there is no launch script......

Output can also be specified in the aircraft file as well, e.g. 

https://github.com/JSBSim-Team/jsbsim/blob/d9df84593b40a5bddb67a94f51d57cd1e514c313/aircraft/c172x/c172x.xml#L1207-L1226

bcoconni, 2022-01-06:
> So or you use my fork or I should make a PR also for JSBSim-Team/jsbsim.

@pbecchi, your branch seems to be quite stable, so it could be time to pull your changes. We'd be glad to review your PR 😃.

[11 more replies]