JSBSim-Team/jsbsim #340: Does JSBSim have  the module plotting the fly trajectory?

theorynice, 2020-11-02:


bcoconni, 2020-11-02:
Not sure what you mean by **the** module plotting the fly trajectory ?

JSBSim has 2 plotting utilities in [src/utilities](https://github.com/JSBSim-Team/jsbsim/tree/master/src/utilities). They are legacy code (7 to 10 years old) and are unmaintained for quite some time. The first one seems to use the plotting library [dislin](https://www.dislin.de) and the compilation instruction are given in the comments
https://github.com/JSBSim-Team/jsbsim/blob/12f83baf66c5e70643b18ceddcc53baaa667331f/src/utilities/main.cpp#L2-L10

The second utility calls [gnuplot](http://www.gnuplot.info) to plot the data. There again compilation instructions are given in the comments.
https://github.com/JSBSim-Team/jsbsim/blob/12f83baf66c5e70643b18ceddcc53baaa667331f/src/utilities/prep_plot.cpp#L59-L64
I have never tried myself to compile these utilities (needless to say to use them) so they might or might not compile with modern compilers.

theorynice, 2020-11-05:
Thank you very much for your reply. what I mean is whether jsbsim has a class to draw the path that the plane passes .

theorynice, 2020-11-05:
> Not sure what you mean by **the** module plotting the fly trajectory ?
>
> JSBSim has 2 plotting utilities in [src/utilities](https://github.com/JSBSim-Team/jsbsim/tree/master/src/utilities). They are [...]

Thank you very much for your reply. what I mean is whether jsbsim has a class to draw the path that the plane passes. My purpose is to visualize the path that the aircraft passed.

seanmcleod, 2020-11-05:
JSBSim doesn't have a built-in module as part of the JSBSim executable that will allow you to plot a flight's trajectory.

So you need to log the flight's trajectory and then use an external tool to then render the flight's trajectory.

@bcoconni has mentioned two legacy external plotting tools that are included in the JSBSim repo.

If you're familiar with python you could write a fairly simple python program to read the lat, lon position from the .csv file that logs the data you require and use something like matplotlib to render the trajectory after the flight.

Do you just want to render a 2D trajectory of the lat, lon or are you looking to render a 3D trajectory?

theorynice, 2020-11-05:
> JSBSim doesn't have a built-in module as part of the JSBSim executable that will allow you to plot a flight's trajectory.
>
> So you need to log the flight's trajectory and then use an external tool to [...]

Thanks for your answer very much.  Your answer about rendering 2D trajectory is helpful to me. As you mention above, how do I render a 3D trajectory? Can FlightGear visaulize the trajectory？

bcoconni, 2020-11-07:
> As you mention above, how do I render a 3D trajectory? Can FlightGear visaulize the trajectory

Yes, you can use JSBSim as an external FDM of FlightGear. The procedure is described on our README: [How can I interface JSBSim with FlightGear ?](https://github.com/JSBSim-Team/jsbsim#how-can-i-interface-jsbsim-with-flightgear-)

theorynice, 2020-11-09:
> Yes, you can use JSBSim as an external FDM of FlightGear. The procedure is described on our README: [How can I interface JSBSim with FlightGear [...]

Thanks very much

closed 2020-11-09 by theorynice