JSBSim-Team/jsbsim #341: Controller optimization

maedehmohammadi, 2020-11-05:
Hello all,
I want to optimize the PID controller of an aircraft in JSBSim. How can I do that?
I did a manual trial and error. However, I was not successful, and I am wondering whether there is an automated way for doing this in JSBSim such as using the optimization methods. Any ideas/samples/references?
Thanks in advance.

bcoconni, 2020-11-07:
Hi @maedehmohammadi

Unfortunately, JSBSim does not provide automatic optimization methods for PIDs. Either the parameters are tuned manually by the user, or they use an external tool. I am not knowledgeable in control opimization but I guess a good entry point is Wikipedia page about PID controllers, and more specifically the [loop tuning section](https://en.wikipedia.org/wiki/PID_controller#Loop_tuning).

maedehmohammadi, 2020-11-08:
Thank you for your answer @bcoconni.

closed 2020-11-08 by maedehmohammadi