JSBSim-Team/jsbsim #342: JSBSim integration in a custom visual software

Marthi0, 2020-11-06:
Hello,

I am working on a simulator using JSBSim and involving a custom visual software developped under Unity3D.
My problem is about ground reaction and above ground level calcultation. My visual is able to send me above ground level under each contact point but JSBSim Ground Callback is requesting this information for precise latitude and longitude.
I cannot use request like FlightGear because JSBSim and my visual are on different computers and it is not compatible with my communication protocol.

I tried to figure out a pattern in callback calls in order to give to JSBSim the right altitude in the right time but without success.

My questions are :
- Is there a pattern in ground callback calls ?
- Do you have any other idea to connect JSBSim to my visual system ?

Thank you for your help,

Marthi

bcoconni, 2020-11-07:
Hi @Marthi0 

Unfortunately, calls to ground callbacks are not limited to the evaluation of the AGL (altitude above ground level) of contact points. They are also used to evaluate the terrain height below the CG. In addition JSBSim, as a project, cannot commit to the fact that the calls pattern of ground callbacks will be stable over the releases so it is not be advisable to write code that relies on that pattern.

However there is a means to determine which contact is used during a call to the ground callback: the `FGLocation` instance passed to the callback contains the position below which the ground altitude shall be interrogated and this location can be converted to a vector of coordinates in the body frame that expresses the position of the contact point with respect to the CG:
```c++
  FGPropagate* propagate = fdmex->GetPropagate();
  // Get the vehicle CG location
  FGLocation CG_location = propagate->GetLocation();
  // Get the transformation matrix from the local horizontal frame to the aircraft body frame
  FGMatrix33 Tl2b = propagate->GetTl2b();
  // Get the position of the contact point relative to the CG in the body frame
  FGColumnVector3 contact_point = Tl2b * CG_location.LocationToLocal(contact_point_location);
```
It is assumed in this code that the instance of `FGFDMExec` is pointed to by the `fdmex` variable and that the contact point location is contained in the `FGLocation` instance named `contact_point_location`. Once the code above is executed, you should be able to tell if the AGL is evaluated at the CG (X, Y, Z are all zero), at the front gear (Y, Z are zero and X is positive), etc.

You can find the [definition of the multiple frames used by JSBSim in our documentation](https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/user-manual-frames-of-reference/).

Note that the CG location might not be fixed in the structural frame, for instance if your aircraft burns fuel then its mass distribution changes over time and so does the CG. As a consequence, the coordinates returned by the code evolve might change over time. If you need stable coordinates then you can convert the coordinates in the structural frame:
```c++
  FGMassBalance* mass_balance = fdmex->GetMassBalance();
  // Get the CG position in the body frame (multiplied by 12 to convert from feet to inches).
  FGColumnVector3 CG_position = mass_balance->StructuralToBody(contact_point*12.);
```
Notes:
* In the code above, all coordinates are expressed in feet.
* The instances of `FGPropagate` and `FGMassBalance` can be stored as members of the ground callback class to avoid calling `FGFDMExec::GetPropagate` and `FGFDMExec::GetMassBalance` at each iteration.

bcoconni, 2020-11-07:
Just to clarify a bit what the code above does: if you have a contact point with the following coordinates in your aircraft definition
https://github.com/JSBSim-Team/jsbsim/blob/1ab61426ddd55da512def05cb1f5dbc94fa6925c/aircraft/c172r/c172r.xml#L103-L107
then the variable `contact_position` in the code above will contain `{58.2, -43., -17.9}` i.e. the exact same coordinates than in the XML file.

Marthi0, 2020-11-10:
Hi,

Thank you for your help and your reactivity. It is quite clear to me now.

closed 2020-11-10 by Marthi0