JSBSim-Team/jsbsim #343: How does JSBSim transfer heading error into control face deflection in autopilot?

theorynice, 2020-11-09:
As mentioned above, the airctaft can caculate state variable by the motion equation. But in autopilot, how does JSBSim transfer the heading error into Aileron deflection？

seanmcleod, 2020-11-09:
Yes.

theorynice, 2020-11-09:
Thanks very much. Where in source code(or in which class) can I find the relevant code?

seanmcleod, 2020-11-09:
There isn't any specific autopilot source code in the JSBSim source code.

Autopilots are implemented in the xml files for the aircraft. Take a look in the JSBSim Reference Manual, there are a bunch of examples of autopilot implementations.

http://jsbsim.sourceforge.net/JSBSimReferenceManual.pdf

theorynice, 2020-11-09:
> There isn't any specific autopilot source code in the JSBSim source code.
> 
> Autopilots are implemented in the xml files for the aircraft. Take a look in the JSBSim Reference Manual, there are a bunch of examples of autopilot implementations.
> 
> http://jsbsim.sourceforge.net/JSBSimReferenceManual.pdf

OK, thank you very much!

bcoconni, 2020-11-19:
@theorynice, shall we consider this item closed or do you need further help ?

closed 2020-11-23 by theorynice