JSBSim-Team/jsbsim #344: Scheduled gain

maedehmohammadi, 2020-11-10:
Hello all,

I have a question. In the X15 aircraft XML file, there is a scheduled gain. Do you know how we can define a scheduled gain for an aircraft? Is there any formula for that?

Thanks in advance.

seanmcleod, 2020-11-10:
JSBSim doesn't have any facilities for calculating the gain values, you would need to use other software like MATLAB etc. to help you calculate the gain values to use in your scheduled gain which you could then plug into JSBSim as you've seen in the X-15 example.

maedehmohammadi, 2020-11-11:
Thank you so much for your quick response.

maedehmohammadi, 2020-11-11:
For my own aircraft, it is complex and time-consuming to simulate it in MATLAB to obtain scheduled-gain. Do you think can I calculate scheduled-gain using [MATLAB-JSBSim](https://github.com/podhrmic/JSBSim-Matlab)?
Thanks in advance @seanmcleod.

seanmcleod, 2020-11-11:
I'm aware of the MATLAB-JSBSim integration but I've never used it myself. Take a look at the following issue which has some discussion of trying to get it to work with more recent versions of Matlab etc.

https://github.com/JSBSim-Team/jsbsim/issues/70

maedehmohammadi, 2020-11-17:
Thank you so much for your response @seanmcleod.

bcoconni, 2020-11-19:
@brats2017, shall we consider this item closed or do you need further help ?

brats2017, 2020-11-19:
Thank you so much. My problem has been solved with your help.

closed 2020-11-19 by bcoconni