JSBSim-Team/jsbsim #345: c310.xml file problem

maedehmohammadi, 2020-11-18:
Hello all,
I am running the **c3101.xml** file. When I change the PID controller gains, the outputs do not change.
For example:
    <property value="50.0"> ap/roll-pid-kp </property>
    <property value="5"> ap/roll-pid-ki </property>
    <property value="17.0"> ap/roll-pid-kd </property> 
The phi is as follow: 
[image]

When I change the PID controller gains to 0 or other values, the phi does not change. Why?
Thanks in advance.

brats2017, 2020-11-19:
Thank you so much for your detailed answer @bcoconni. It works now.

bcoconni, 2020-11-19:
Glad it helps !

closed 2020-11-19 by bcoconni