JSBSim-Team/jsbsim #346: waypoint navigation system

brats2017, 2020-11-20:
Hello all,
I want to design a waypoint navigation system for the aircraft x15. I read the manual and I run the c3104.xml file without any problem. However, for designing the x15 waypoint navigation system, I face problems. The extra works that I did to design a waypoint navigation system are as follows:
1) I copied the c3104.xml file from the jsbsim-master\scripts and pasted it with a different name (x15way.xml) in jsbsim-master\scripts.
2) I changed the 8th line of x15way.xml to <use aircraft="X15" initialize="ellington"/>  and saved it.
3) I copied the ellington.xml file at this location: jsbsim-master\aircraft\X15.

When I run the x15way.xml, I have errors that said some properties are not defined. Consequently:
4) I add these properties to the x15ap.xml file and run the x15way.xml file. At this step, I do not have any error, but in the command prompt window in front of the distance to the wp at events, is written 0. Why?

The properties that I added, are:

>   <property>ap/attitude_hold</property>
>   <property>ap/heading_hold</property>
>   <property>ap/yaw_damper</property>
>   <property>ap/active-waypoint</property>
>   <property>ap/altitude_setpoint</property>
>   <property>ap/heading_setpoint</property>
>   <property>ap/heading-setpoint-select</property>
>   <property>ap/aileron_cmd</property>
>   <property>ap/elevator_cmd</property>
>   <property>ap/rudder_cmd</property>
>   <property>ap/airspeed_setpoint</property>
>   <property>ap/airspeed_hold</property>
>   <property>ap/throttle-cmd-norm</property>
>   <property>guidance/wp-distance</property>
>   <property>fcs/mixture-cmd-norm[1]</property>
>   <property>fcs/advance-cmd-norm[1]</property>
>   <property>propulsion/magneto_cmd</property>
>   <property>fcs/throttle-cmd-norm[1]</property>
>   <property>fcs/propulsion/starter_cmd</property>
>   <property>fcs/propulsion/starter_cmd</property>
>   <property>propulsion/starter_cmd</property>
>   <property>guidance/target_wp_latitude_rad</property>
>   <property>guidance/target_wp_longitude_rad</property>
>   <property>ap/altitude_hold</property>

I wanted to kindly ask you the steps that I did to design a waypoint navigation system are correct?
I am a beginner with JSBSim and your help will be greatly appreciated.
Thanks in advance

bcoconni, 2020-11-21:
Hi @brats2017,

The flight controls for waypoint guidance are located in [systems/GNCUtilities.xml](https://github.com/JSBSim-Team/jsbsim/blob/master/systems/GNCUtilities.xml). The c310 loads this autopilot feature with the following code but the X15 does not.
https://github.com/JSBSim-Team/jsbsim/blob/82de86e7dcf6b8029bbbf676dbff99e6bdc30b9d/aircraft/c310/c310.xml#L343-L345

Also please bear in mind that the X15 has very limited range and maneuverability and might not be able to perform the flight plan described in the script `c3104.xml`.

brats2017, 2020-11-23:
Thank you so much for your quick response @bcoconni. I could design a waypoint navigation system for x15 with the points that you mentioned. Now, I can work on constraints to get good results.

brats2017, 2020-11-23:
I wanted to kindly ask you how can I consider a constraint for the maximum speed?
Thanks in advance

bcoconni, 2020-11-23:
> how can I consider a constraint for the maximum speed?

Not sure what you mean. If you want to cap the maximum speed, my suggestion would be to use a PID that would take over the throttle as soon as the aircraft velocity goes beyond a given threshold. Then the PID would be switched off as soon as the velocity drops below that threshold. Just an idea off the top of my head...

brats2017, 2020-11-24:
Thank you so much for your response @bcoconni.

bcoconni, 2020-11-27:
@brats2017 according to your discussion in issue #349 I infer that this issue can be closed ?

brats2017, 2020-11-27:
Yes, sure. Thank you so much @bcoconni.

closed 2020-11-27 by bcoconni

brats2017, 2020-12-08:
> Not sure what you mean. If you want to cap the maximum speed, my suggestion would be to use a PID that would take over the throttle as soon as the aircraft velocity goes beyond a given threshold. [...]

Sorry @bcoconni, the issue is closed. However, my problem has not been solved yet. Can you please give me advice that what are the exact **properties name** (input and output) that I should consider in the PID controller to cap the maximum speed?
Your help will be greatly appreciated.

seanmcleod, 2020-12-08:
You do realise that the X-15 is carried aloft under the wing of a B-52, then dropped from altitude, lights its rocket engine which has a limited run time and then becomes a glider.

So you would only be able to use the throttle of the rocket engine for a small portion of the flight to control speed. When it becomes a glider the throttle will be useless, you would need to use alpha (within limits) to control speed. The X-15 also had speed brakes (split tail) which you could use to help in managing speed, but the current X-15 JSBSim model doesn't seem to model the speed brake.

So which part of the flight do you want to limit speed for? If it's just while the rocket is firing then you can use the throttle to control that.

brats2017, 2020-12-08:
> You do realise that the X-15 is carried aloft under the wing of a B-52, then dropped from altitude, lights its rocket engine which has a limited run time and then becomes a glider.
>
> So you would [...]

Thank you so much for your response. The information that you mentioned is very helpful. About the question that you asked me, I am designing a waypoint navigation system for X-15, and I want to control its speed when it is following the waypoints. Its speed is too high when it heads to the first waypoint and it causes the aircraft to deviate. Do you have any idea about that?
Thanks in advance.

seanmcleod, 2020-12-08:
You haven't answered the question I asked. 

> So which part of the flight do you want to limit speed for?

You mention the leg to the 1st waypoint, but that doesn't specify whether it's under rocket power or whether it's gliding, and as I mentioned you will need to use a different strategy for the different phases of flight in general.