JSBSim-Team/jsbsim #347: I can't build jsbsim python3.7 ubuntu18.04 on vmware

kkz003, 2020-11-22:
I cannot build this part on python3.7 and ubuntu 18.04.
I am very beginner. Some issue discussions are too difficult to understand to me.
I just change Py_InitModeule to Py_Initialize but it caused another complie errors.
Please give me any advice to solve this issue and build jsbsm.

cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
/3rd-party/src/jsbsim/tests/fpectlmodule.cpp: In function ‘PyObject* initfpectl()’:
/3rd-party/src/jsbsim/tests/fpectlmodule.cpp:130:17: error: ‘Py_InitModule’ was not declared in this scope
PyObject m = Py_InitModule("fpectl", fpectl_methods);
^~~~~~~~~~~~~
/3rd-party/src/jsbsim/tests/fpectlmodule.cpp:130:17: note: suggested alternative: ‘Py_Initialize’
PyObject m = Py_InitModule("fpectl", fpectl_methods);
^~~~~~~~~~~~~
Py_Initialize
/3rd-party/src/jsbsim/tests/fpectlmodule.cpp:132:5: error: return-statement with no value, in function returning ‘PyObject {aka _object}’ [-fpermissive]
return;
^~~~~~
error: command 'gcc' failed with exit status 1

bcoconni, 2020-11-22:
Hi @kkz003 ,

According to the code extract you are showing above, it looks you are using quite an outdated version of JSBSim. The code that is currently used in JSBSim in [python/fpectl/fpectlmodule.cpp](https://github.com/JSBSim-Team/jsbsim/blob/49262a7215402da1638caf75f17103a842a234fa/python/fpectl/fpectlmodule.cpp) is shown below:
https://github.com/JSBSim-Team/jsbsim/blob/49262a7215402da1638caf75f17103a842a234fa/python/fpectl/fpectlmodule.cpp#L143-L185
This is quite different from the code you showed (which is not compatible with Python 3 hence the compilation errors you get).

May I ask where you got the code you are trying to compile ?

Also, if you want to avoid building JSBSim yourself, be aware that we are providing:
* Ubuntu packages for our [latest release v1.1.2](https://github.com/JSBSim-Team/jsbsim/releases/tag/v1.1.2) .
* Python packages that can be installed with `pip`.

So unless you are planning to hack the code of JSBSim I would encourage you to use one of the packages above.

On the other hand, if you are planning to modify JSBSim code then we will be more happy to provide help as well.

kkz003, 2020-11-22:
Dear Bertrand Coconnier

Thank you for your support by the following email.
I try to rewrite the cpp to a python3 acceptable version but I am failing so far.

> May I ask where you got the code you are trying to compile ?

I have tried to build Scrimmage which is an UAV simulator and I set its makefile's repository of JSBSIM as https://github.com/JSBSim-Team/jsbsim.git <https://github.com/JSBSim-Team/jsbsim.git>
If you can, please tell me the repository address of the newest version.

Thank you very much. I forget to use pip. I did “pip install jsbsim” just now and
 it was successfully finished. So the newest version of jsbsim does not occur 
these errors.

bcoconni, 2020-11-22:
> I try to rewrite the cpp to a python3 acceptable version but I am failing so far.

You don't need to as this has already been fixed in JSBSim code base.
> I have tried to build Scrimmage which is an UAV simulator and I set its makefile's repository of JSBSIM as https://github.com/JSBSim-Team/jsbsim.git <https://github.com/JSBSim-Team/jsbsim.git>
If you can, please tell me the repository address of the newest version.

That's weird, you are using the correct git repo for JSBSim but once again, the code you showed is quite outdated. Have you tried to build JSBSim from a fresh git clone ?
> Thank you very much. I forget to use pip. I did “pip install jsbsim” just now and
 it was successfully finished. So the newest version of jsbsim does not occur
these errors.

Great ! Do you still need support to build JSBSim ? Or are you planning to use the Python package ?

kkz003, 2020-11-22:
Dear bcoconni
I want to start  a fresh git clone  but I fail.
Could you give me a suggestion?
Is my tag setting correct?
In the CMakeList.txt 
  set(EXTERNAL_TARGET_JSBSIM jsbsim)
  ExternalProject_Add (
    jsbsim
    PREFIX ${CMAKE_CURRENT_BINARY_DIR}/3rd-party
    CMAKE_COMMAND cmake -DCMAKE_INSTALL_PREFIX=${CMAKE_CURRENT_BINARY_DIR}/3rd-party ..
    GIT_REPOSITORY "https://github.com/JSBSim-Team/jsbsim.git"
    #GIT_REPOSITORY "https://git.code.sf.net/p/jsbsim/code"
    GIT_TAG "v1.1.1"
    #GIT_TAG "36de9ac63c959cef5d7b2c56fb49c1a57fd46369"
    PATCH_COMMAND ${SCRIMMAGE_SETUP_DIR}/apply-patch.sh ${CMAKE_SOURCE_DIR}/setup/patches/jsbsim.patch
    UPDATE_COMMAND ${SCRIMMAGE_SETUP_DIR}/install-aircraft.sh "${CMAKE_CURRENT_BINARY_DIR}/3rd-party/src/jsbsim"
    INSTALL_COMMAND make install
    )

bcoconni, 2020-11-22:
OK I think I understand what you are trying to do.

First thing first, I would try to build JSBSim alone from a fresh copy of the git repo:
```bash
> mkdir test_JSBSim
> cd test_JSBSim
> git clone https://github.com/JSBSim-Team/jsbsim.git
> cd jsbsim
> mkdir build
> cd build
> cmake ..
> make
> ctest
```
The last command `ctest` may take several minutes to complete but you can use the `-j` option to run several tests in parallel: `ctest -j4` will run 4 tests in parallel if your CPU has 4 cores. The same option is applicable to `make` by the way.

Only once the commands above have completed successfully can we tell that your system is capable to compile JSBSim. If something fails at this stage, please provide the error messages so that we can understand what is going wrong.

Regarding the CMake command you supplied, I am not familiar with `ExternalProject_Add` but I have a few comments (which might or might not help):
* You might not want JSBSim to build its Python module so you might want to add `-DBUILD_PYTHON_MODULE=OFF` to the `CMAKE_COMMAND`
* Regarding the parameter `GIT_TAG`, you were previously referring to the commit 36de9ac63c959cef5d7b2c56fb49c1a57fd46369 which is quite old (Sep 25, 2016). Now you've modified it to `v1.1.1` which hopefully should help with your problem. By the way, we've just released `v1.1.2` yesterday so you might rather want to use the tag `v1.1.2` for your build process.
* I have checked the [patch `jsbsim.patch` from the project Scrimmage](https://github.com/gtri/scrimmage/blob/master/3rd-party/patches/jsbsim.patch). You might want to update its content since it is referring to files that no longer exist (`Makefile.am`, `.cvsignore`, etc.)

Other than these comments, you command `ExternalProject_Add` seems fine to me. Let us know the error messages you get if this fails.

bcoconni, 2020-12-05:
@kkz003 Has your problem been fixed ? Or do you need further support ?

kkz003, 2020-12-05:
Dear Bertrand Coconnier

Thank you for your email.

I cannot install scrimmage so far but your suggestion about JSBsim
solved one of my problem! I appreciate it.

sincerely yours

closed 2020-12-10 by bcoconni