JSBSim-Team/jsbsim #348: Can JSBSim be used as dynamics constraint in trajectory optimization software, such as GPOPS, PSOPT?

theorynice, 2020-11-23:


bcoconni, 2020-11-23:
Hi @theorynice,

Could you please give more details about what these softwares are ? What kind of input should they be supplied with ? etc. With that few details, I am afraid that it will be difficult to answer your question.

theorynice, 2020-12-03:
> Hi @theorynice,
> 
> Could you please give more details about what these softwares are ? What kind of input should they be supplied with ? etc. With that few details, I am afraid that it will be difficult to answer your question.

Sorry for replying too late due to my business trip!  In aircraft trajectory optimization problem, orthogonal collocation method(or called pseudospectral method) is popular but need to  be supplied with the concrete algebraic formula, which is differential form of equation of dynamics. Can JSBSim transfer into the concrete equation metioned above?

bcoconni, 2020-12-04:
> Sorry for replying too late due to my business trip! In aircraft trajectory optimization problem, orthogonal collocation method(or called pseudospectral method) is popular but based on the concrete differential form of equation of dynamics. Can JSBSim transfer into the concrete differential equation metioned above?

Unfortunately, there is no such feature in JSBSim. But there is however a discussion about model linearization that is going on in the issue #227. Once the feature will be implemented, JSBSim will be able to generate the matrices A and B for a differential equations system such as 

<a href="https://www.codecogs.com/eqnedit.php?latex=\dot{x}(t)=Ax(t)&plus;Bu(t)" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\dot{x}(t)=Ax(t)&plus;Bu(t)" title="\dot{x}(t)=Ax(t)+Bu(t)" /></a>

This may be helpful for trajectory optimization ?

theorynice, 2020-12-07:
> > Sorry for replying too late due to my business trip! In aircraft trajectory optimization problem, orthogonal collocation method(or called pseudospectral method) is popular but based on the concrete differential form of equation of dynamics. Can JSBSim transfer into the concrete differential equation metioned above?
> 
> Unfortunately, there is no such feature in JSBSim. But there is however a discussion about model linearization that is going on in the issue #227. Once the feature will be implemented, JSBSim will be able to generate the matrices A and B for a differential equations system such as
> 
> [![](https://camo.githubusercontent.com/6884c179687f518b7ab21642bc85f8b4bce167c6ca6e4ae99264f7d49a2aa9b6/68747470733a2f2f6c617465782e636f6465636f67732e636f6d2f6769662e6c617465783f5c646f747b787d2874293d41782874292b4275287429)](https://www.codecogs.com/eqnedit.php?latex=\dot{x}(t)=Ax(t)+Bu(t))
> 
> This may be helpful for trajectory optimization ?
Of course it is! Thanks for your help!

closed 2020-12-09 by theorynice