JSBSim-Team/jsbsim #349: no-fly zone

brats2017, 2020-11-24:
Hello all,
I want to design a trajectory for an aircraft in JSBSim with **no-fly zone constraints** like this:
[image]

Is it possible? 
Your help will be greatly appreciated.
Thanks in advance

seanmcleod, 2020-11-24:
Any reasons you can think of that would make it difficult or impossible?

I don't see it as a JSBSim specific issue. You have an auto-pilot which can follow waypoints, so it really comes down to a geometry problem. If the route between 2 waypoints intersects a no-fly zone then add in some intermediate waypoints based on your geometry calculations.

brats2017, 2020-11-24:
Thank you so much for your response @seanmcleod. The aircraft that I am working on it, works on Mach 9. As you mentioned, it is possible to set waypoints that are close to each other to avoid entering the no-fly zone, but it is too difficult for aircraft to follow them. Actually, I think the aircraft can not follow these close waypoints because its velocity is high. For the special trajectory that I have, I think I have to work on the controller and I do not have another option.

seanmcleod, 2020-11-25:
You can't get around the basic physics. So when I mentioned that it's a 'geometry problem' what I was including in that were any physics constraints, e.g. based on the speed and the maximum load factor (g) that will determine the turning radius of any turns that need to be executed. Whether those turns are initiated by the current waypoint controller or any other controller you come up with.

And there may well be cases where it is impossible to meet all the constraints. For example say waypoint 1 in the image above was at 10N 65E and given your Mach 9 speed and maximum load factor it may be physically impossible to fly from waypoint 1 to waypoint 2 without encroaching on no-fly zone 1.

brats2017, 2020-11-25:
Thank you so much @seanmcleod. I will consider the points that you mentioned. You said the speed and the maximum load factor (g) determine the turning radius. Can you please explain more about that? Actually, I want the aircraft to reach the target in a straight trajectory as in the image above. For example, in the c3104.xml file, I considered some waypoints as follows (blue line). However, the aircraft turns when it wants to go to the third waypoint from the second waypoint. Can you please help me how can I determine the turn angle?

[image]

brats2017, 2020-11-26:
Thank you so much @seanmcleod. I will consider the points that you mentioned. You said the speed and the maximum load factor (g) determine the turning radius. Can you please explain more about how can I do that?

Also, I used the c310 files as a sample and I could run a waypoint navigation system for **X15** aircraft. However, when I run it, the nonacceptable results are as follows:
[image]

As we can see, in the second waypoint, the height is negative and the velocity is too much. Actually, the velocity and altitude should be controlled, but I do not know how can I do that. Also, only the `<condition>velocities/vc-fps ge 145</condition>` works in my project, and when I change other conditions like altitude for raise landing gear and head to the first waypoint, the results do not change.  Is there any idea to solve my problem? Which parameters should be tuned to solve this problem? The X15 has good PID controllers, and I expected that it could follow waypoints.

seanmcleod, 2020-11-26:
> You said the speed and the maximum load factor (g) determine the turning radius. Can you please explain more about how can I do that?

Are you looking for a formula on how to calculate the turn radius for a given speed and load factor? 

In terms of example run above, take note of the **GEAR_CONTACT** messages, that means the aircraft has impacted the ground.

brats2017, 2020-11-26:
Thank you so much for your response @seanmcleod . No, I am familiar with its formula. However, my question is whether the turning radius is considered for c310, that I can use its code in my own project (X15 aircraft) or not.

Yes, as you mentioned the gear contact occurred. Because I think I did not tune some parameters that exist in my autopilot file (I copied them from c310ap.xml file), and I do not know how can I tune them:

```
  <pid name="fcs/heading-pi-controller">
    <input> fcs/heading-roll-error-switch </input>
    <kp> 6 </kp>
    <ki> 0.13 </ki>
    <kd> 6.0 </kd>
  </pid>
```

```
  <integrator name="fcs/integral">
    <input> fcs/ap-alt-hold-switch </input>
    <trigger> fcs/windup-trigger </trigger>
    <c1> 0.001 </c1>
  </integrator>
```

```
  <pure_gain name="fcs/proportional">
    <input> fcs/ap-alt-hold-switch </input>
    <gain> 0.03 </gain>
  </pure_gain>
```

Should I tune them with trial and error methods?
Your help will be greatly appreciated.

seanmcleod, 2020-11-27:
> However, my question is whether the turning radius is considered for c310

Did you see any references to calculating and taking into account a turn radius in the c310 auto-pilot?

I'm not familiar with the c310 auto-pilot, but taking a quick look at it I don't see it calculating or taking turn radius into account. It simply calculates a heading error and drives the ailerons to zero out the heading error using a PID controller.

brats2017, 2020-11-27:
Thank you so much @seanmcleod. I will consider the points that you mentioned.

seanmcleod, 2020-11-28:
@brats2017 talking about the PID gains

> Should I tune them with trial and error methods?

Something else you need to potentially take into account is the concept of **gain scheduling**. It's very difficult to come up with a single set of PID gains that work well for an aircraft that operates across a wide range of speeds. So if you're trying to control something like an X-15 operating between say 700KIAS down to 200KIAS when coming in to land etc. then you'll need the gains to be scheduled based on speed etc.

brats2017, 2020-12-03:
Thank you so much for your answer @seanmcleod. I used [this paper](https://ieeexplore.ieee.org/document/260670) and I designed a fuzzy scheduled PID controller for the aircraft with the help of MATLAB. However, I could not get a good result, and I have the gear contact message. The gain scheduling that I got from MATLAB is based on time, and when I want to consider this gain scheduling based on velocity I am not sure what should I do. Actually, I wrote gain scheduling based on velocity according to the two tables that I had: a table which is velocity vs time, and a table with is gain scheduling based on time.  I want to kindly ask you do you have any idea/reference about designing a gain scheduling?
your help will be greatly appreciated.

bcoconni, 2020-12-04:
@brats2017 regarding the gear contact this almost certainly means that either you don't have an altitude/pitch hold autopilot or that it is disabled or failing.

closed 2020-12-10 by bcoconni