JSBSim-Team/jsbsim #350: How to build JSBSIM with Cygwin on windows

moreba1, 2020-11-27:
Hi

How we can build jsbsim with cygwin ? 

is related to :
https://github.com/JSBSim-Team/jsbsim/issues/225

bcoconni, 2020-11-27:
Hi @moreba1,

JSBSim is using CMake and make to build. In theory, it should work with Cygwin as well. The process to build [JSBSim with CMake is described in our developer docs](https://github.com/JSBSim-Team/jsbsim/blob/master/doc/DevelopersDocs.md#building-with-cmake).

Let us know if the building process fails for any reason, we will gladly answer your questions (do not forget to provide the error messages you get when reporting an issue).

moreba1, 2020-11-28:
https://github.com/PX4/PX4-Autopilot/issues/16294

closed 2020-12-10 by bcoconni