JSBSim-Team/jsbsim #352: HoldDown

jonsberndt, 2020-12-09:
Some years ago I believe there was a HoldDown parameter in FGAircraft.cpp - at least there was in my local source tree. Anyone know if this was removed at some point? There was an associated property, "hold-down". I might not have actually ever submitted that change.

seanmcleod, 2020-12-09:
It's in the current source, e.g. https://github.com/JSBSim-Team/jsbsim/blob/5530ed26f07124ffe7ca1d609a83da1d90770454/src/FGFDMExec.cpp#L166

jonsberndt, 2020-12-09:
Interesting. The implementation is different. In 2007 it was in FGAircraft. I can't remember if I changed that myself or someone else. LoL. Oh well, that it is still there is the important thing. I need it. :-)

seanmcleod, 2020-12-09:
There are a couple of scripts in the current repo that make use of it and @bcoconni also developed a test for it.

https://github.com/JSBSim-Team/jsbsim/search?q=hold-down

bcoconni, 2020-12-09:
@jonsberndt 

> Some years ago I believe there was a HoldDown parameter in FGAircraft.cpp - at least there was in my local source tree. Anyone know if this was removed at some point? There was an associated property, "hold-down". I might not have actually ever submitted that change.

You did 😄. It is just that I moved the feature first to `FGAccelerations` (commit 6ce8a7ee0) and then to `FGFDMExec` (commit 2bf0140da2). If my memory serves me well, I have moved the HoldDown feature first to `FGAccelerations` because the purpose of the feature is really to nullify accelerations, not filtering out some forces. This first change gave more stable results which allowed the vehicle to remain motionless for a longer time when `forces/hold-down` was set to true. The second change has been implemented because you can still have non zero velocities even when accelerations are zero. Obviously the point of holding down a vehicle is to force both velocities and accelerations to be zero which implies some coordination between `FGAccelerations`and `FGPropagate`. As a result `FGFDMExec` seemed a better place to manage the feature.

closed 2020-12-09 by bcoconni