JSBSim-Team/jsbsim #357: Negative drag due to elevator deflection

seanmcleod, 2020-12-11:
While debugging a difference in trim results I noticed that the drag due to elevator deflection was negative for the 737 model.

```sh
    aero/coefficient/CD0 = 2801.679561
    aero/coefficient/CDi = 3813.129823
    aero/coefficient/CDflap = 5689.318388
    aero/coefficient/CDgear = 1928.582504
    aero/coefficient/CDsb = 0.000000
    aero/coefficient/CDsb = 0.000000
    aero/coefficient/CDsp = 0.000000
    aero/coefficient/CDbeta = 0.007089
    aero/coefficient/CDde = -181.721539
```

Taking a look at the 737 model's calculation for elevator deflection drag it's using `elevator-pos-norm` `[-1, +1]` without converting it to an absolute value, hence the reason you can see negative drag values.

```xml
<function name="aero/coefficient/CDde">
    <description>Drag_due_to_Elevator_Deflection</description>
    <product>
        <property>aero/qbar-psf</property>
        <property>metrics/Sw-sqft</property>
        <property>fcs/elevator-pos-norm</property>
        <value>0.059</value>
    </product>
</function>
```

In context that means at around the critical AoA of ~12.5 degrees the negative drag from a full elevator deflection will completely cancel out the total induced drag!

[image]

Also if the aircraft was trimmed at say ~2 degrees AoA and you put in a sudden step of full elevator deflection the aircraft would actually start accelerating with this extra negative drag 'thrust'.

In addition to the sign issue with the elevator deflection the use of a normalized position `[-1, +1]` is unusual since the coefficient is normally per radian. It's not impossible that it's been converted/scaled, however the value of `0.059` is very similar to the typical values published that are per radian.

The 737 model for example has a maximum elevator deflection of `0.3 rad` so if the figure of `0.059` is per radian then that would equate to a value of `0.018` for a normalized range.

I did a search through all the JSBSim aircraft for `/CDde` to see whether the 737 model was an exception or not.

Unfortunately it's not. The following aircraft models follow the same setup as the 737 model.

## Incorrect use of `elevator-pos-norm`

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/sgs233/sgs233.xml#L321

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/F4N/F4N-jsbsim.xml#L403

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/L17/L17.xml#L378

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/sgs126/sgs126.xml#L363

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/OV10/OV10.xml#L502

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/MD11/MD11.xml#L495

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/A4/A4.xml#L541

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/C130/C130.xml#L477

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/pogo-jsbsim/pogo-jsbsim.xml#L614

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/B747/B747.xml#L533

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/T38/T38.xml#L570

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/XB-70/XB-70.xml#L604

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/DHC6/DHC6.xml#L324

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/737/737.xml#L632

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/T37/T37.xml#L469

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/f22/yf22.xml#L529

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/B17/B17.xml#L910

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/fokker50/fokker50.xml#L695

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/L410/L410.xml#L1187

## Correct use of `mag-elevator-pos-rad`

The following aircraft make use of the magnitude that the FCS provides - `mag-elevator-pos-rad` which is an absolute version and radian based.

```xml
            <function name="aero/coefficient/CDde">
                <description>Drag_due_to_elevator_Deflection</description>
                <product>
                    <property>aero/qbar-psf</property>
                    <property>metrics/Sw-sqft</property>
                    <property>fcs/mag-elevator-pos-rad</property>
                    <value>0.0500</value>
                </product>
            </function>
```

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/t6texan2/t6texan2.xml#L327

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/pa28/pa28.xml#L431

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/c172x/c172x.xml#L911

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/f15/f15.xml#L594

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/fokker100/xf100.xml#L720

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/A320/A320.xml#L602

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/c172r/c172r.xml#L500

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/fokker100/fokker70.xml#L727

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/fokker100/fokker100.xml#L776

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/dr1/dr1.xml#L665

## Coefficient of 0

The following aircraft make use of `mag-elevator-pos-rad` but have a coefficient of 0.

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/c182/c182.xml#L483

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/c172p/c172p.xml#L551

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/c310/c310.xml#L670

## Sort of correct use of `<abs> elevator-pos-norm </abs>`

Some aircraft make use of an absolute version of `elevator-pos-norm` using `<abs />` but should more than likely be using `pos-rad`.

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/p51d/p51d.xml#L960

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/Boeing314/Boeing314.xml#L1313

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/Concorde/Concorde.xml#L1526

## Correct sign but confusing `elevator-pos-rad` and `elevator-pos-norm`?

The following two aircraft use a table to get the sign correct, but they appear to be confusing `elevator-pos-rad` with `elevator-pos-norm`.

```xml
                          <independentVar>fcs/elevator-pos-rad</independentVar>
                          <tableData>
                              -1.0000	0.1140
                              0.0000	0.0000
                              1.0000	0.1140
                          </tableData>
```

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/minisgs/minisgs.xml#L331

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/SGS/SGS.xml#L332

## Sign issue with `elevator-pos-rad`

This aircraft uses `elevator-pos-rad` but forgets about the sign issue.

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/F80C/F80C.xml#L547

## Table based `elevator-pos-rad` that gets the sign correct

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/aircraft/f22/f22.xml#L1931

## Aeromatic

Lastly I took at look at the Aeromatic source code to see if it was possibly the reason for most of the incorrect cases, but at least currently, I didn't look at it's source code history, it does the correct thing, outputting - `<abs><property>fcs/elevator-pos-rad</property></abs>`

## To Do

I'm busy preparing a pull request to correct all the examples listed above that I labelled problematic, unless someone tells me I've got it all wrong :wink:.

bcoconni, 2020-12-11:
Looks good to me ! :+1:

Out of curiosity, I have checked the [official c172p model for FlightGear](https://github.com/c172p-team/c172p/blob/121a280a111c8df36ea5ba00d273694b16c62cfc/c172p.xml#L1304) and they are using the exact same function than our `c172p` model: the drag is computed correctly with `fcs/mag-elevator-pos-rad` and cancelled out with a `0` value.
```xml
            <function name="aero/coefficient/CDDe">
                <description>Drag_due_to_Elevator_Deflection</description>
                <product>
                    <property>aero/qbar-psf</property>
                    <property>metrics/Sw-sqft</property>
                    <property>fcs/mag-elevator-pos-rad</property>
                    <value>0.0000</value>
                </product>
            </function>
```

seanmcleod, 2020-12-22:
Example of the increase in IAS as a step input to the elevator is made with the 737 model.

Same 737 model except that the orange version `/fdm/jsbsim/velocities/vc-kts` uses `<abs>` for elevator deflection drag.

[image]

Octal450, 2021-01-30:
Its definitely wrong. Negative drag should not. Either use the property that has absolute value or do it yourself.. for example in my MD-11:

`		<function name="aero/coefficient/CDde_lo">
			<description>Drag due to left outer elevator</description>
			<product>
				<property>aero/qbar-psf</property>
				<property>metrics/Sw-sqft</property>
				<abs><toradians><property>hydraulics/elevator-out-l/final-deg</property></toradians></abs>
				<value>0.003625</value>
			</product>
		</function>`

Kind Regards,
Josh

bcoconni, 2021-01-30:
@Octal450 What's wrong ?

bcoconni, 2021-02-20:
@seanmcleod it seems this issue has been fixed. Shall we close it now ?

seanmcleod, 2021-02-20:
Yep, closing.

closed 2021-02-20 by seanmcleod

Octal450, 2021-02-20:
> @Octal450 What's wrong ?

Just agreeing with that it was wrong, that's all.

Kind Regards,
Josh