JSBSim-Team/jsbsim #360: Use gamma to help trim solver?

seanmcleod, 2020-12-13:
@bcoconni I'd noticed you periodically having to change the initial theta angle in the initialization file for the Short S23 aircraft in order to ensure that the trim solver found a solution - https://github.com/JSBSim-Team/jsbsim/commit/e0f1449ca56bcfd98dabe8a4d87d580b1bedeeb5

When I was debugging the trim differences I was seeing with the 737 model (negative elevator drag etc.) I noticed fairly different trim results depending on the initial theta value specified. In a much older (many years) version of JSBSm I had specified an initial theta of 4.1 degrees and the trim solution was level flight with a theta of ~ -0.5 degrees.

But with the latest JSBSim it results in a climb of ~2 degrees with a similar value of ~ -0.5 theta degrees. But if I specified an initial gamma value of 0 degrees then the initial value of theta appears to be ignored and I get a trim result very close to the older version of JSBSim.

So I tested out using gamma with `Short_S23\reset01.xml` and the script `scripts\Short_S23_2.xml`.

## Theta -1.5
Using the currently comitted version of `reset01.xml`

```sh
  Trim successful
  Trim Results:
       Angle of Attack:  -1.27  wdot:  8.22e-04 Tolerance: 1e-03  Passed
              Throttle:   0.58  udot: -3.43e-04 Tolerance: 1e-03  Passed
            Pitch Trim:   0.14  qdot: -3.74e-08 Tolerance: 1e-04  Passed
            Roll Angle:  -0.04  vdot:  4.54e-12 Tolerance: 1e-03  Passed
              Ailerons:  -0.01  pdot: -3.68e-08 Tolerance: 1e-04  Passed
                Rudder:   0.00  rdot:  1.33e-20 Tolerance: 1e-04  Passed

Trim (Event 4) executed at time: 1.008333
    velocities/u-aero-fps = 253.108948
    velocities/v-aero-fps = 0.000596
    velocities/w-aero-fps = -5.627488
    attitude/psi-rad = 3.141593
    attitude/theta-rad = -0.048410
    attitude/phi-rad = -0.000681
```

## Theta -1.0

```sh
  Full Trim
  Sorry, wdot doesn't appear to be trimmable

  Trim failed
  Trim Results:
       Angle of Attack:  -1.27  wdot: -8.35e+00 Tolerance: 1e-03  Failed
              Throttle:   0.65  udot: -5.65e-01 Tolerance: 1e-03  Failed
            Pitch Trim:   0.14  qdot:  2.86e-01 Tolerance: 1e-04  Failed
            Roll Angle:  -0.04  vdot:  2.22e-02 Tolerance: 1e-03  Failed
              Ailerons:  -0.01  pdot:  2.10e-02 Tolerance: 1e-04  Failed
                Rudder:   0.00  rdot: -1.44e-04 Tolerance: 1e-04  Failed
FATAL ERROR: JSBSim terminated with an exception.
The message was: Trim Failed
```

## Theta -1.0  Gamma 0.0

```sh
  Full Trim

  Trim successful
  Trim Results:
       Angle of Attack:  -1.26  wdot:  6.85e-04 Tolerance: 1e-03  Passed
              Throttle:   0.78  udot: -9.15e-04 Tolerance: 1e-03  Passed
            Pitch Trim:   0.13  qdot:  5.02e-08 Tolerance: 1e-04  Passed
            Roll Angle:  -0.04  vdot: -2.22e-11 Tolerance: 1e-03  Passed
              Ailerons:  -0.02  pdot: -1.03e-09 Tolerance: 1e-04  Passed
                Rudder:   0.00  rdot:  8.79e-20 Tolerance: 1e-04  Passed

Trim (Event 4) executed at time: 1.008333
    velocities/u-aero-fps = 253.110318
    velocities/v-aero-fps = 0.000583
    velocities/w-aero-fps = -5.565552
    attitude/psi-rad = 3.141593
    attitude/theta-rad = -0.021985
    attitude/phi-rad = -0.000705
```

## Theta -10.0  Gamma 0.0

```sh
  Full Trim

  Trim successful
  Trim Results:
       Angle of Attack:  -1.26  wdot:  6.85e-04 Tolerance: 1e-03  Passed
              Throttle:   0.78  udot: -9.15e-04 Tolerance: 1e-03  Passed
            Pitch Trim:   0.13  qdot:  5.02e-08 Tolerance: 1e-04  Passed
            Roll Angle:  -0.04  vdot: -2.22e-11 Tolerance: 1e-03  Passed
              Ailerons:  -0.02  pdot: -1.03e-09 Tolerance: 1e-04  Passed
                Rudder:   0.00  rdot:  2.96e-21 Tolerance: 1e-04  Passed

Trim (Event 4) executed at time: 1.008333
    velocities/u-aero-fps = 253.110318
    velocities/v-aero-fps = 0.000583
    velocities/w-aero-fps = -5.565552
    attitude/psi-rad = 3.141593
    attitude/theta-rad = -0.021985
    attitude/phi-rad = -0.000705
```

## Theta -10.0  Gamma 0.0

```sh
Theta 0.0  Gamma 0.0

  Trim successful
  Trim Results:
       Angle of Attack:  -1.26  wdot:  6.85e-04 Tolerance: 1e-03  Passed
              Throttle:   0.78  udot: -9.15e-04 Tolerance: 1e-03  Passed
            Pitch Trim:   0.13  qdot:  5.02e-08 Tolerance: 1e-04  Passed
            Roll Angle:  -0.04  vdot: -2.22e-11 Tolerance: 1e-03  Passed
              Ailerons:  -0.02  pdot: -1.03e-09 Tolerance: 1e-04  Passed
                Rudder:   0.00  rdot:  6.20e-20 Tolerance: 1e-04  Passed

Trim (Event 4) executed at time: 1.008333
    velocities/u-aero-fps = 253.110318
    velocities/v-aero-fps = 0.000583
    velocities/w-aero-fps = -5.565552
    attitude/psi-rad = 3.141593
    attitude/theta-rad = -0.021985
    attitude/phi-rad = -0.000705
```

I haven't double-checked the trim source code and logic, but it appears based on these handful of tests if we specify the required gamma that the trim solver might not be as brittle/fragile and we won't have to periodically tweak the initial theta value.

bcoconni, 2020-12-14:
> I'd noticed you periodically having to change the initial theta angle in the initialization file for the Short S23 aircraft in order to ensure that the trim solver found a solution - e0f1449

Yeah, that's a pain. I have always attributed this to the limited abilities of the default trim solver. As you are pointing out, each time the Short S23 trim fails, I tweak the initial value of `theta` until it converges again and I check in the process that the converged solution is still the same, or is very close to the previous one.

The mere fact that trim convergence breaks following minor modifications to the code has strenghtened over the time my feeling that the default trim solver is not robust. So each time the Short S23 broke I jumped to the conclusion that the cause was the trim solver brittleness rather than the code modifications I made. I may have been wrong, I don't know.
> When I was debugging the trim differences I was seeing with the 737 model (negative elevator drag etc.) I noticed fairly different trim results depending on the initial theta value specified. In a [...]

That's the kind of result that makes me uneasy about the exact cause of the trim convergence issues. Does the trim solver fail because its numerical algorithms are not robust enough ? Or does it fail because the code modifications I made were incorrect ?

However the 737 should not be affected by the recent changes to the propeller code. So this problem with the 737 might have been existing for quite some time. Have you tried to bisect the code changes to find at which point it has started to converge to a climb of 2 degrees ?
> But if I specified an initial gamma value of 0 degrees then the initial value of theta appears to be ignored and I get a trim result very close to the older version of JSBSim.

That's interesting ! Could you please provide an example of an XML file `reset*.xml` that specifies initial conditions with `gamma` ? I infer from your examples that you specify both `theta` and `gamma` in the initial conditions. By the way, your last 2 examples are both named '**Theta -10.0 Gamma 0.0**' but I understand from the console output of the last example that its title should be '**Theta 0.0 Gamma 0.0**':
```bash
Theta 0.0  Gamma 0.0

  Trim successful
  Trim Results:
       Angle of Attack:  -1.26  wdot:  6.85e-04 Tolerance: 1e-03  Passed
  [...]
```
> I haven't double-checked the trim source code and logic, but it appears based on these handful of tests if we specify the required gamma that the trim solver might not be as brittle/fragile and we won't have to periodically tweak the initial theta value.

Neither have I. However I suspect the specification of `gamma` to take over `theta` specification, the latter being ignored. I will check the code later this week, unless you beat me 😄.

seanmcleod, 2020-12-14:
> Could you please provide an example of an XML file...
>  I infer from your examples that you specify both theta and gamma in the initial conditions.

Yep, I left `theta` in and simply tried different values, didn't try removing it completely, and added in `gamma`.

```xml
<?xml version="1.0"?>
<initialize name="reset01">
  <!--
    This file sets up the aircraft to cruise at 143kt indicated at 3000 ft.
  -->
  <ubody unit="KTS">     150.0   </ubody>
  <vbody unit="FT/SEC">    0.0   </vbody>
  <wbody unit="FT/SEC">    0.0   </wbody>
  <latitude unit="DEG">   37.86  </latitude>
  <longitude unit="DEG">-122.34  </longitude>
  <phi unit="DEG">         0.0   </phi>
  <theta unit="DEG">       0.0   </theta>
  <gamma unit="DEG">       0.0   </gamma>
  <psi unit="DEG">       180.0   </psi>
  <altitude unit="FT">  3000.0   </altitude>
</initialize>
```

> By the way, your last 2 examples are both named 'Theta -10.0 Gamma 0.0' 

Yep, sorry it was a typo.

> Have you tried to bisect the code changes to find at which point it has started to converge to a climb of 2 degrees?

I haven't tried that (yet?). The original JSBSim code is from late 2014 that I'm using with my VSS sim and I'm in the process of updating to the latest JSBSim code. I did glance at the FGTrim code from 2014 and the latest and set some breakpoints and stepped through it to see if I could see any obvious differences that may account for the difference I was seeing but didn't spot any at the time. But I didn't spend too much time on it, partly since I had been planning on adding a flight path angle option to the VSS sim anyway.

Did wonder if the code from 2014 was defaulting to a gamma of 0 and using it even if it wasn't specified in the initialization file whereas the latest code wasn't using it at all unless explicitly listed in the initialization file.

bcoconni, 2020-12-18:
Following your findings, I have checked if `gamma` was already used in our models and there are only a few models that use it in their initial conditions:
https://github.com/JSBSim-Team/jsbsim/blob/647510208c3e88ee9393b68351d591c09c2beecf/aircraft/737/cruise_init.xml#L9
https://github.com/JSBSim-Team/jsbsim/blob/647510208c3e88ee9393b68351d591c09c2beecf/aircraft/737/cruise_steady_turn_init.xml#L9
https://github.com/JSBSim-Team/jsbsim/blob/647510208c3e88ee9393b68351d591c09c2beecf/aircraft/c172x/elevator_doublet_init.xml#L12
and a couple of them use a value for `gamma` which is not zero.

https://github.com/JSBSim-Team/jsbsim/blob/647510208c3e88ee9393b68351d591c09c2beecf/aircraft/c172p/reset01.xml#L19
https://github.com/JSBSim-Team/jsbsim/blob/647510208c3e88ee9393b68351d591c09c2beecf/aircraft/c172r/reset01.xml#L19
According to the code in `FGInitialCondition`, setting `gamma` to 0 is equivalent to setting the vertical component of the velocity to 0.0. This may be equivalent to setting `theta` to 0 and α to 0 (but I need to understand what does `FGInitialCondition::calcThetaBeta` exactly do to be sure).
https://github.com/JSBSim-Team/jsbsim/blob/647510208c3e88ee9393b68351d591c09c2beecf/src/initialization/FGInitialCondition.cpp#L284-L306
Regarding your last example '**Theta 0.0 Gamma 0.0**', I have tried `theta` to 0.0 (without specifying `gamma` at all) and the trim solver converges. So this might be an indication that only the vertical component of the velocity needs to be zero.

seanmcleod, 2020-12-20:
I took a quick look with my 737 example, using the following initialization file with `gamma` set to 0 and with the `gamma` element removed.

```xml
<?xml version="1.0"?>
	<initialize name="Sortie_1_-_Task_1-ic">
	  <ubody unit="KTS">  194  </ubody>
	  <vbody unit="KTS">  0.0  </vbody>
	  <wbody unit="KTS">  0.0  </wbody>
	  <latitude unit="DEG">  -33.9873333333333  </latitude>
	  <longitude unit="DEG"> 18.6086666666667  </longitude>
	  <phi unit="DEG">       0.0  </phi>
	  <theta unit="DEG">     4.1  </theta>
	  <psi unit="DEG">       0    </psi>
          <gamma unit="DEG">		 0.0  </gamma>    
	  <altitude unit="FT">   5000 </altitude>
	  <winddir unit="DEG">   0.0  </winddir>
	  <vwind unit="FT/SEC">  0.0  </vwind>
	  <running> -1 </running>
	</initialize>
```

Setting a breakpoint in `FGTrim::DoTrim()` before it starts trimming I compared the initial condition values in `fgic`.

### Gamma = 0, Theta 4.1

```sh
vUVW_NED: {327.43513999999982, 0.0000000000000000, -3.5527136788005009e-15} 
vt: 327.43513999999982
alpha: -3.2550388563675582e-17
```

### Gamma not specified, Theta=4.1
```sh
vUVW_NED: {326.59716240070310, 0.0000000000000000, -23.410775694715209} 
vt: 327.43513999999993
alpha: 2.1700259042450376e-17
```

So yes with `gamma = 0` the vertical velocity component is practically set to 0.

With `gamma` not specified it looks like the vertical velocity component is set to `vt * sin(theta)` which is why I see a trim result in this case which results in a climb.

bcoconni, 2021-05-28:
As you may have noticed, I had to change yet another time the files ` aircraft/Short_S23/reset01.xml` and ` aircraft/Short_S23/reset02.xml`  in commit c0b46aa74d19ad79aa91c551e4ed5ad548d55863

I tried adding `gamma = 0` but to no avail.