JSBSim-Team/jsbsim #369: Simplified c172_cruise script: altitude is not hold even with autopilot

kevung, 2020-12-30:
Hello Jsbsim community, 

In order to learn to use the c172 with autopilot, I a simplified version of the script the [c172_cruise_8K.xml](https://github.com/JSBSim-Team/jsbsim/blob/master/scripts/c172_cruise_8K.xml), with only heading and altitude autopilot commands.

I expected the altitude to stay constant to 4000ft but it decreases with time. Am I doing something wrong?

Thank you very much for your help.

The script I use:
```
<?xml version="1.0" encoding="UTF-8"?>
<?xml-stylesheet type="text/xsl" href="http://jsbsim.sf.net/JSBSimScript.xsl"?>
<runscript xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
    xsi:noNamespaceSchemaLocation="http://jsbsim.sf.net/JSBSimScript.xsd"
    name="C310-01A takeoff run">
    
  <description>Flying a plane in a straight line</description>
  <!-- <use aircraft="c172x" initialize="airborne0"/> -->
  <!-- <use aircraft="c172x" initialize="reset00"/> -->
  <use aircraft="c172x" initialize="reset01"/>

  <output name="straight_line.csv" type="CSV" rate="0.1">
  <!-- <output name="straight_line.csv" type="CSV" rate="60"> -->
    <rates>            ON </rates>
    <velocities>       ON </velocities>
    <position>         ON </position>
    <atmosphere>       ON </atmosphere>
    <fcs>              OFF </fcs>
    <ground_reactions> OFF </ground_reactions>
    <propulsion>       OFF </propulsion>
    <simulation>       OFF </simulation>
    <massprops>        OFF </massprops>
    <forces>           ON </forces>
    <moments>          ON </moments>
    <aerosurfaces>     OFF </aerosurfaces>
    <coefficients>     ON </coefficients>
  </output>

  <run start="0.0" end="2000" dt="0.008333333">

     <event name="Start engine">
        <condition>
           simulation/sim-time-sec  ge  0.01
        </condition>
        <set name="fcs/throttle-cmd-norm" action="FG_RAMP" value="0.50" tc="0.05"/>
        <set name="fcs/mixture-cmd-norm" action="FG_RAMP" value="1.0" tc="0.05"/> 
        <set name="propulsion/magneto_cmd" value="3"/>
        <set name="propulsion/starter_cmd" value="1"/>
        <notify/>
     </event>

     <event name="Trim">
        <condition>
           simulation/sim-time-sec  gt  1.0
        </condition>
        <set name="simulation/do_simple_trim" value="1"/>
        <notify>
           <property> velocities/u-aero-fps </property>
           <property> velocities/v-aero-fps </property>
           <property> velocities/w-aero-fps </property>
           <property> attitude/psi-rad </property>
           <property> attitude/theta-rad </property>
           <property> attitude/phi-rad </property>
           <property> position/h-agl-ft </property>
           <property> accelerations/pdot-rad_sec2 </property>
           <property> accelerations/qdot-rad_sec2 </property>
           <property> accelerations/rdot-rad_sec2 </property>
           <property> propulsion/engine/thrust-lbs </property>
           <property> propulsion/engine/propeller-rpm </property>
           <property> velocities/p-rad_sec </property>
           <property> velocities/q-rad_sec </property>
           <property> velocities/r-rad_sec </property>
           <property> fcs/elevator-cmd-norm </property>
           <property> fcs/aileron-cmd-norm </property>
           <property> fcs/rudder-cmd-norm </property>
           <property> fcs/elevator-pos-norm </property>
           <property> fcs/right-aileron-pos-norm </property>
           <property> fcs/rudder-pos-norm </property>
           <property> fcs/throttle-pos-norm[0] </property>
           <property> fcs/mixture-pos-norm[0] </property>
           <property> inertia/weight-lbs </property>
           <property> velocities/vc-kts </property>
           <property> velocities/ve-kts </property>
           <property> velocities/vtrue-kts </property>
           <property> atmosphere/T-R </property>
           <property> propulsion/engine/fuel-flow-rate-gph </property>
           <property> propulsion/engine/power-hp </property>
        </notify>
     </event>

  <event name="Hold heading and altitude">
     <condition>simulation/sim-time-sec ge 5</condition>
     <set name="ap/heading_setpoint" value="200"/>
     <set name="ap/heading_hold" value="1"/>
     <set name="ap/altitude_setpoint" value="4000.0"/>
     <set name="ap/altitude_hold" value="1"/>
     <notify/>
  </event>

  </run>

</runscript>

```

kevung, 2020-12-30:
Is it due to the length of the simulation? Is it due to aircraft fuel capacity?

seanmcleod, 2020-12-30:
This is what I see when I run your script.

[image]

And zooming in on the altitude it doesn't look like the autopilot is very stable.

[image]

And looking at the airspeed and elevator.

[image]

And round the 328s mark things go pear shaped resulting in a stall.

So my initial guess is the PID gains aren't very well tuned, and maybe there is an integrator error that builds up eventually resulting in full elevator deflection leading to a stall.

I'm not familiar with this particular auto-pilot, so I'm not sure if it used to be more stable in the past and something has changed or whether it's always performed like this.

kevung, 2020-12-30:
> This is what I see when I run your script.
>
> [image]
>
> And zooming in on the altitude it doesn't look like the autopilot is very stable.
>
> [image]
>
> And looking at the airspeed and elevator.
>
> [image] [...]

Thank you very much for your help! I'm quite new to JSBSim and to aircraft control and that was the reason why I chose to use the autopilot for a first introduction and the c172x plane seemed to be the most basic/simple one :) . I'm going to have a look into the reference manual if autopilot tuning is tackled.

Thanks again and all the best for the new year.

closed 2021-01-22 by bcoconni