JSBSim-Team/jsbsim #371: JSBSim.minimal.cpp file not compiling

viren3999, 2021-01-13:
Hello!

I have been recently familiarised to this ambitious project. Needless to say, I am facing a few issues while trying to compile the **JSBSim.minimal.cpp file.** 

I have been successful in building the standalone .exe and the .lib files using MS Visual Studio 2017.

```
#include <FGFDMExec.h>

int main(int argc, char **argv)
{
  JSBSim::FGFDMExec FDMExec;
  FDMExec.LoadScript(argv[1]);
  FDMExec.RunIC();
  bool result = true;
  while (result) result = FDMExec.Run();
}
```

I tried to compile this file using g++, but it gives the following errors :

```
FGFDMExec.h:291:8: note: candidate: 'bool JSBSim::FGFDMExec::LoadScript(const SGPath&, double, const SGPath&)'
   bool LoadScript(const SGPath& Script, double deltaT=0.0)
```

`FGFDMExec.h:291:8: note:   **no known conversion for argument 1 from 'char*' to 'const SGPath&'**`

The above error occurs for this line : **FDMExec.LoadScript(argv[1]);**

Kindly help!

seanmcleod, 2021-01-13:
Take a look at the ctors for `SGPath` and some examples in `JSBSim.cpp`, e.g.

https://github.com/JSBSim-Team/jsbsim/blob/868fcd59f4fea8bc051bd44b18a913c1d86de40c/src/JSBSim.cpp#L350

viren3999, 2021-01-15:
> Take a look at the ctors for `SGPath` and some examples in `JSBSim.cpp`, e.g.
> 
> https://github.com/JSBSim-Team/jsbsim/blob/868fcd59f4fea8bc051bd44b18a913c1d86de40c/src/JSBSim.cpp#L350

viren3999, 2021-01-15:
Sorry that was by mistake sir.

In the examples provided, the string is either casted explicitly to SGPath or converted to the required format using some string functions. 

However, what I fail to understand is that when I try to build using Visual Studio, I have replaced the already existent JSBSim.cpp with the one shown above (i.e. JSBSim.minimal.cpp), yet it fails to build. (This was suggested on the flight gear forum)

seanmcleod, 2021-01-15:
The issue is that `JSSim.minimal.cpp` was last modified virtually 10 years ago.

https://github.com/JSBSim-Team/jsbsim/blob/master/src/JSBSim.minimal.cpp

So 10 years ago `LoadScript()` took a plain `char*`, but in the meantime it was updated to take a `SGPath` and the minimal example wasn't updated.

viren3999, 2021-01-15:
@seanmcleod Got it, Sir. Thank you for the support 👍👍

closed 2021-01-15 by viren3999

seanmcleod, 2021-01-15:
@viren3999 I've updated `JSBSim.minimal.cpp` so that it now compiles, see the following commit - https://github.com/JSBSim-Team/jsbsim/commit/4f69b4de8e0a0998e6076dfc4ab7556d191a2834