JSBSim-Team/jsbsim #376: Is the website still in use?

valtsu41, 2021-01-18:
Not completely sure if this is the right place to report this, but it seems like [http://jsbsim.sourceforge.net/](url) has multiple links to the old Sourceforge repository (for example the "suggestions" -button). If the website is still used, I guess those could be updated?

bcoconni, 2021-01-18:
Hi @valtsu41, thanks for your interest in JSBSim.

You are correct on both topics: this is the right place to report this issue and indeed the website is quite outdated.

In the very short term, I'll try to fix the broken links later this week.

In the longer term, the problem is that there are only @jonsberndt and myself who can modify the website files in SourceForge (one needs an SSH access to do so). I guess we need to migrate the website in a repo (it's a static site anyway) so that more people can participate to its maintenance.

bcoconni, 2021-01-22:
This should be fixed now and the website now redirects to GitHub wherever it makes sense.