JSBSim-Team/jsbsim #377: Changing the fuel feed

valtsu41, 2021-01-18:
I have understood that currently it is possible to set the tanks an engine feeds from before running the simulation, but not during the simulation. Is there a technical reason for this, or is this just simply never thought to be worth to implement? https://sourceforge.net/p/jsbsim/feature-requests/59/ mentions the use of priorities, but that won't work in all cases.

seanmcleod, 2021-01-18:
So you can remove a tank from feeding an engine by setting it's priority to 0 while the simulation is running, and changing it to > 0 when you want the tank to be used. In what situations would this not work?

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/src/models/propulsion/FGTank.h#L310

https://github.com/JSBSim-Team/jsbsim/blob/eccfd64bbe09fc209c399bead031a8ccc5873118/src/models/propulsion/FGTank.cpp#L467-L469

valtsu41, 2021-01-18:
In case other engines use the tank as well, but behind different valves.

seanmcleod, 2021-01-18:
We would need to expose the engine's `feed list` and allow the `feed list` to be modified via properties. 

At the moment the `feed list` is created when the aircraft config file is read in, basically a list of integers referencing the tank indices.

```xml
<feed> {integer} </feed>
... optional more feed tank index numbers ... 
```

For example from the 737, both feed from the center tank (index 2) and each engine also feeds from the wing tank on the same wing as the engine.

```xml
    <propulsion>
        <engine file="CFM56">
            <location unit="IN">
                <x>  540 </x>
                <y> -193 </y>
                <z>  -40 </z>
            </location>
            <feed>0</feed>
            <feed>2</feed>
            <thruster file="direct">
                <location unit="IN">
                    <x>  540 </x>
                    <y> -193 </y>
                    <z>  -40 </z>
                </location>
                <orient unit="DEG">
                    <roll>  0 </roll>
                    <pitch> 0 </pitch>
                    <yaw>   0 </yaw>
                </orient>
            </thruster>
        </engine>
        <engine file="CFM56">
            <location unit="IN">
                <x> 540 </x>
                <y> 193 </y>
                <z> -40 </z>
            </location>
            <feed>1</feed>
            <feed>2</feed>
            <thruster file="direct">
                <location unit="IN">
                    <x> 540 </x>
                    <y> 193 </y>
                    <z> -40 </z>
                </location>
                <orient unit="DEG">
                    <roll>  0 </roll>
                    <pitch> 0 </pitch>
                    <yaw>   0 </yaw>
                </orient>
            </thruster>
        </engine>
        <tank type="FUEL"><!-- Left wing tank -->
            <location unit="IN">
                <x> 520 </x>
                <y> -80 </y>
                <z> -18 </z>
            </location>
            <type>JET-A</type>
            <capacity unit="LBS"> 10200 </capacity>
            <contents unit="LBS"> 10000 </contents>
        </tank>
        <tank type="FUEL"><!-- Right wing tank -->
            <location unit="IN">
                <x> 520 </x>
                <y>  80 </y>
                <z> -18 </z>
            </location>
            <type>JET-A</type>
            <capacity unit="LBS"> 10200 </capacity>
            <contents unit="LBS"> 10000 </contents>
        </tank>
        <tank type="FUEL"><!-- Center tank -->
            <location unit="IN">
                <x> 480 </x>
                <y>   0 </y>
                <z> -18 </z>
            </location>
            <type>JET-A</type>
            <capacity unit="LBS"> 15000 </capacity>
            <contents unit="LBS">  4000 </contents>
        </tank>
    </propulsion>
```

valtsu41, 2021-01-18:
Properties cannot be lists, right?

seanmcleod, 2021-01-18:
We could expose the feed list as an indexed property, e.g.

```sh
engine[0]/feed[0] = 0
engine[0]/feed[1] = 2
engine[1]/feed[0] = 1
engine[1]/feed[1] = 2
```

So you would be able to change the source tanks for each engine, and writing say -1 to a feed would disable the feed.

valtsu41, 2021-01-18:
Would those be only the predefined feeds then?

seanmcleod, 2021-01-18:
Yep, but you could add additional feeds, e.g. you could add a 3rd feed to the 737 example that is initially set to -1 to allow you the flexibility later on to say feed an engine from all 3 tanks at once if you wanted to.

valtsu41, 2021-01-18:
That would probably work, but I wonder how logical that would feel to use...

seanmcleod, 2021-01-18:
The logic would be that you first decide on the maximum number of feeds/tanks that an engine can be connected to. Then engine could then be connected to any of the tanks, and you can change the tanks it's connected to and leave it connected to a tank but cut the feed (close a valve in effect) by writing -1 to the specific feed.

valtsu41, 2021-01-18:
The thing that does not feel logical is that you have a certain number of feed "lines" and no more or no less. From the technical implementation perspective I understand it, but when thinking about it from a user perspective it just feels weird, especially since things like fuel lines are not actually modeled.

bcoconni, 2021-01-19:
> from a user perspective it just feels weird, especially since things like fuel lines are not actually modeled.

Indeed JSBSim does not hard code a fuel system. But it does much better than that: it allows you to code one, as detailed as you wish. The [c172p team](https://github.com/c172p-team/c172p) for instance has implemented the fuel system of the c172p in JSBSim. See their implementation in their file [Systems/fuel.xml](https://github.com/c172p-team/c172p/blob/master/Systems/fuel.xml) for example.

Basically they are using a "float chamber" which is kind of a fuel buffer and which is the only tank that feeds the engine (see line 578 in their file [c172p.xml](https://github.com/c172p-team/c172p/blob/master/c172p.xml)). The fuel systems then manages how the fuel flows from one tank to another and ultimately fill the float chamber.

I think @dany93 can give you more details about this, if you need to.

valtsu41, 2021-01-19:
I know that that method exists, and the reason I opened this suggestion/issue in the first place is the fact that that method is so "hacky" in my opinion. I'm sorry if I sound like someone who complains about everything, but these things just really bother me.

dany93, 2021-01-19:
+1 with @bcoconni.

The [DOUGLAS DC C-47 JSBSim](http://pattenflightgear.wifeo.com/douglas-dc3-c47.php) from Patten's aircraft hangar manages the feeding of two engines from 4 tanks at choice.
This has been done thanks to a "float chamber" (one per engine, given in the FDM). Also called collector tank.
In-sim, you do not change the float chamber which feeds a given engine, but you can select which one(s) the four main tanks this float chamber is fed from.

See FG Wiki [Howto:Write a fuel system in JSBSim, Douglas DC3 (2 engines, 4 tanks)](http://wiki.flightgear.org/Howto:Write_a_fuel_system_in_JSBSim#Douglas_DC3_.282_engines.2C_4_tanks.29)

seanmcleod, 2021-01-19:
> I'm sorry if I sound like someone who complains about everything...

That's fine, but the issue is you haven't really given us a detailed requirement, basically this is all we have to work with:

> set the tanks an engine feeds from before running the simulation, but not during the simulation.

Then with a couple of suggestions of how to modify which tanks an engine feeds from at runtime you've just replied with replies along these lines:

> I wonder how logical that would feel to use...
> method is so "hacky" in my opinion

So if you formulate a more detailed requirement we'll have a better chance of coming up with a suggestion that may meet it.

valtsu41, 2021-01-19:
> So if you formulate a more detailed requirement we'll have a better chance of coming up with a suggestion that may meet it.

Basically, I just would like to control the fuel feed for individual engines while the simulation is running, without the need of adding those "buffers".

> Then with a couple of suggestions of how to modify which tanks an engine feeds from at runtime you've just replied with replies along these lines:

I could give a counter-suggestion, the feed properties could be just boolean properties as well. Usually the fuel system won't change on the fly, so setting the feed tanks in the configuration file and then enabling or disabling them for the engine when needed could be a more "clean" solution (at least I think so). Opinions are welcome.

seanmcleod, 2021-01-19:
> control the fuel feed for individual engines while the simulation is running, without the need of adding those "buffers".

The suggestion of exposing the `feed list` as an indexed property would allow this.

> feed properties could be just boolean properties as well. 

We would then need a duplicate list, the feed list listing which tank a feed (integer) is connected to, i.e. the current `feed list` and then another one containing booleans.

So the suggestion was to overload the feed list such that -1 represented a closed valve and 0 and higher represented the tank the feed was connected to. It does mean the system enabling/disabling would need to remember the tank number when wanting to re-enable the feed as opposed to simply writing `true`.

> Usually the fuel system won't change on the fly...

So take the C-47 example.

> The real aircraft has two engines and four tanks. Two tank selectors (one per engine) allows each engine to be fed by each tank, separately and one tank at a time for a given engine. The diagram of the fuel system can be seen in this document. For FG, the engines are numbered [0] and (1], the 'true' feed tanks [0], [1], [2], [3]. Each tank selector is a 5-positions ('OFF' + 4 tanks) rotary switch. Its positions are numbered -1 (= 'OFF'), 0, 1, 2, 3, respectively for 'no tank selected', tank[0], [1], [2], [3].

So with the indexed fuel list property you could implement the rotary switch option fairly easy without the need for 'buffer' tanks etc. Each engine would have 1 feed, for the rotary switch in the OFF position the feed would have a value of -1, and when the rotary switch was rotated to a specific tank (0, 1, 2, 3) then the feed value would equal the tank number.

This matches the mention of an engine only being fed by 1 tank at a time.

But for other aircraft, e.g. the 737 we can also support an engine being fed by multiple tanks at the same time.

valtsu41, 2021-01-19:
If you think that is the best way to implement it, I am fine with it.

seanmcleod, 2021-01-19:
I'd like to get some feedback/thoughts from @bcoconni before rushing into an implementation.

andgi, 2021-01-19:
Hi,

Take a look at the Short_S23 too.
It has 4 engines and supports 3 different fuel tank configurations (1 tank
per wing; 2 tanks per wing; 3 tanks per wing and 2 in the fuselage if I
recall correctly).
This is done by giving each engine a small tank that is refilled from the
other tanks.
Fuel is taken from the tanks then drawn through the engine pumps and
propagated to the small tanks (either from all engines or from each side or
from the own engine if I recall correctly).

Cheers,
Anders

valtsu41, 2021-01-22:
By the way, sorry if the feature requests are supposed to go to Discussions, I did not know that feature even existed on GitHub.

bcoconni, 2021-01-22:
> By the way, sorry if the feature requests are supposed to go to Discussions, I did not know that feature even existed on GitHub.

No worries. Bugs and feature requests should still be discussed in `Issues`. And the current issue clearly falls in that category.

We are trying to redirect support requests (_"How do I do this ?"_, _"I tried to do that and got an error message, why ?"_, ...)  to `Discussions` because experience shows that this kind of discussion tends to never reach a conclusion and therefore are never closed.

bcoconni, 2021-01-29:
I have given some thoughts to this issue and there are a number of topics I'd like to clarify.

@valtsu41 said:
> I know that that method exists, and the reason I opened this suggestion/issue in the first place is the fact that that method is so "hacky" in my opinion.
[...]
Basically, I just would like to control the fuel feed for individual engines while the simulation is running, without the need of adding those "buffers".
[...]
setting the feed tanks in the configuration file and then enabling or disabling them for the engine when needed could be a more "clean" solution (at least I think so).

For some reason you seem to frown upon the usage of buffer tanks as being an ugly hack that is not reflecting the reality of a fuel system (at least this is how I am interpreting the quotes that you put around the words _buffers_ and _clean_, not mentioning the word _hacky_ that speaks for itself).

That's quite a biased point of view to me: you have to keep in mind that there is only one fuel pipe that is connected to an engine fuel pump and that the tank selection valve is located at a distance upstream to that fuel pump. As a result the pipe between the selection valve and the fuel pump has a non zero volume capacity (it can be easily several litres/gallons on a airliner). So even if this is not a tank as the layman means it, it is still a volume that can store a given quantity of fuel i.e. it is a tank from a technical/engineering point of view. Let's not call it buffers if that is the name that bothers you.

@valtsu41 said:
> If you think that is the best way to implement it, I am fine with it.

This statement makes me uneasy about your exact intent. @dany93, @andgi and myself suggested that JSBSim is currently able to address the concern you raised. You dismissed these proposals on the ground that they are "hacky" (your own word). OK. Then @seanmcleod made a different proposal about how to address this issue without recurring to XML "programming" and your answer was "If you think that is the best way to implement it, I am fine with it." No, this is not an answer. The point is not about what **we** think but about what **you** think. **You** raised the issue, **you** tell us if our proposals address it or not. If you don't intend to use @seanmcleod proposal, just tell it, no offence will be taken. But we will not waste our time implementing a feature that will not be used.

valtsu41, 2021-02-08:
> No, this is not an answer. The point is not about what we think but about what you think. You raised the issue, you tell us if our proposals address it or not. If you don't intend to use @seanmcleod [...]
I actually meant that the idea is fine. It really does not bother me how exactly it works, as long as it works. The comments before you can ignore, I should have thought about this thing more thoroughly before sending them...