JSBSim-Team/jsbsim #383: glide mode

brats2017, 2021-01-27:
Hello all
I have a question and it will be appreciated if you help me with this. I wanted to ask you is implemented the boost-glide mode for an aircraft in JSBSim? If yes, in which aircraft code I can find it? I know the gliding mode is implemented for X-15. However, what about boost-glide mode?
Thanks in advance.

seanmcleod, 2021-01-27:
What do you mean by 'boost-glide mode'?

Do you mean like the X-15 which get's dropped from the wing of a B-52 carrier aircraft, then light's it's rocket engine for a couple of minutes, which is the boost? Then after the engine cuts out it glides back down?

brats2017, 2021-01-27:
Thank you for your response @seanmcleod. I understood what is the boost mode in X-15 with your explanation. Just one question: is there any possibility/idea to increase the boost mode time in X-15?

seanmcleod, 2021-01-27:
But you were asking about 'boost-glide mode', so let's first make sure we're talking about the same thing. Do we agree on what 'boost-glide mode' means?

If you want to increase the duration of the boost mode of the X-15 simply increase the amount of rocket fuel available.

brats2017, 2021-01-27:
The boost-glide mode that is in my mind is the same as discussed in this [paper](https://www.researchgate.net/publication/315833583_Trajectory_optimisation_for_a_rocket-assisted_hypersonic_boost-glide_vehicle).

> If you want to increase the duration of the boost mode of the X-15 simply increase the amount of rocket fuel available.
Thank you for your helpful idea @seanmcleod.

brats2017, 2021-01-28:
We have four phases in the [paper](https://www.researchgate.net/publication/315833583_Trajectory_optimisation_for_a_rocket-assisted_hypersonic_boost-glide_vehicle), and these phases are described below and depicted in Fig. 1.
1. The first 5 seconds of first stage boost phase, during which pitch manoeuvre does not take place. In this phase, the vehicle gains sufficient speed in the first 5 seconds, after which the stabilisers become effective and provide sufficient damping moment for a controlled pitch manoeuvre(17).
2. The time duration after t = 5s till the burn-out of the first stage rocket motor. It is during this phase that the launch vehicle pitches down using angle-of-attack control.
3. The second stage boost phase. In this phase, the flight path angle changes to meet the burn-out conditions.
4. The free flight and the re-entry stage. In this phase, the vehicle is steered to optimal down-range or cross-range distance with the help of angle-of-attack, bank angle and throttle control.
[image]

-I think the boost-glide hypersonic that is in the above paper, lunches from a rocket while the boost-glide of X-15 in JSBSim lunches from another aircraft. Do you think am I right?

-My other question is do you think it is reasonable that X-15 lunches from the rocket?
IF yes, is there any code for the boost-glide modes in JSBSim that I can use for the lunch of X-15 from a rocket? Do you think the J2461.xml can help me with this issue?

Your help will be greatly appreciated.
Thanks in advance

seanmcleod, 2021-01-28:
It's very difficult to provide useful responses without a detailed description of what you're trying to implement. So just asking about "boost-glide" in terms of whether it's been implemented (depending on exactly what it actually means, and without defining it clearly) in an existing JSBSim aircraft doesn't provide nearly enough information.

First of all reading through the paper it turns out that they test 2 different aircraft, a wave-rider aircraft and a wing-body aircraft, and 2 different configurations of the wing-body aircraft, one with no engine and one with a rocket engine. So 3 different configurations in all. 

These are all launched using a rocket. The wave-rider aircraft and the wing-body aircraft with no engine are pure gliders once they separate from the rocket that launched them. The wing-body aircraft uses a combination of gliding and can use it's rocket engine for short periods.

So first of all, what exactly are you trying to model. Do you want to model the complete system from launch on the ground and model all 3 configurations? 

Or do you only want to model say 1 of the configurations from the time it detaches from the rocket until it lands?

How closely do you want to model their results? If you want to be very close then you have to use their aerodynamic model etc., you can't just make use of the X-15 in it's place and expect to get results that are close to theirs.

brats2017, 2021-01-28:
Thank you so much for reading the paper and your explanation. 

> Or do you only want to model say 1 of the configurations from the time it detaches from the rocket until it lands?

I had sent the paper to clarify my think related to boost-glide mode. However, it is not important that I modeled the configuration exactly that is defined in the paper.
Consequently, right now, please do not consider the paper that I sent, and please let me know what you think about my following question:

I just wanted to change the configuration of the x153.xml file. The only thing that I want to change is related to the launch phase of the aircraft. How can I ask X-15 to be launched from a rocket instead of being launched from an aircraft (B-52)? 

Please consider that I am a beginner with aerospace and JSBSim.
Your help will be great;y appreciated.
Thanks in advance.

seanmcleod, 2021-01-28:
I think you're over complicating things in terms of thinking about a rocket versus a carrier/mothership like a B-52. All you really care about are the **initial conditions** at launch, you can set the initial conditions as you require them, doesn't matter what the mothership was.

So take a look at `scripts\x153.xml`, you'll notice the following:

```xml
  <use aircraft="X15" initialize="reset01"/>
``` 

The **initial conditions** are specified in the file **reset01** which lives in `aircraft\x15\reset01.xml`, here is a copy.

```xml
<?xml version="1.0"?>
<initialize name="reset01">
  <!--
    This file sets up the X15 to start off at altitude.
  -->
  <ubody unit="FT/SEC">  750.0  </ubody>
  <vbody unit="FT/SEC">    0.0  </vbody>
  <wbody unit="FT/SEC">  100.0  </wbody>
  <latitude unit="DEG">   33.0  </latitude>
  <longitude unit="DEG">-110.0  </longitude>
  <phi unit="DEG">         0.0  </phi>
  <theta unit="DEG">       0.0  </theta>
  <psi unit="DEG">       260.0  </psi>
  <altitude unit="FT"> 40000.0  </altitude>
</initialize>
```

So it starts the simulation off with the aircraft at 40,000ft, at a particular lat, lon, particular attitude and particular velocity.

So if you want to start the simulation off as if the X-15 had just been released from a rocket then create a new **initial conditions** file and you can for example set the starting altitude to 200,000ft, with a much faster starting velocity, and a different attitude, e.g. maybe it gets released from the rocket at a pitch attitude of 60 degrees etc.

brats2017, 2021-01-29:
Thank you so much for your response. I did as you mentioned and I got the results that I want.

Also, about the paper that I sent and regarding the guidance that you gave me, I used the aerodynamic model of waverider configuration for re-entry flight (phase 4 in the figure above). It behaves as a glider once it separates from the rocket that launched it. However, right now, I want to model the complete system in JSBSim (phase 1, phase 2,  phase 3, and phase 4).

I think I should define 2 different strategies for that:
To model phase1, phase2, and phase3: I should use the aerodynamic forces that exist in the main version of the X15.xml file. 
To model phase 4: I should use the aerodynamic forces that I defined for waverider and exist in the paper.
 
Can you please let me know do you agree with this? If yes, How can I do that?

Your help will be greatly appreciated.
Thanks in advance

seanmcleod, 2021-01-29:
If you want to model phases 1, 2 & 3 then you need to define a rocket model, it's aerodynamic model will be quite different to an X-15. Plus it gets much more complicated in that during after separation you then have to switch models completely in terms of the mass, moments of inertia, aerodynamic model etc.

Take a look at the Space Shuttle model available for FlightGear since they have to model these sorts of changes.

If the Wave-Rider is housed inside a fairing then it's aerodynamic model mustn't be active during the launch phases, but if it sits on the rocket without a fairing covering it then it's aerodynamic model in addition to the aerodynamics of the rocket itself have to be taken together, the Space Shuttle falls into this category.

closed 2021-01-29 by bcoconni