JSBSim-Team/jsbsim #384: External forces, fatal error

sthelia, 2021-01-28:
Hi! 
I'm new at using JSBSim and am currently going through some tutorials. I have tried to recreate the SGS 2-33 glider from the attached report. 
[Tesi_Pecora_JSBSim.pdf](https://github.com/JSBSim-Team/jsbsim/files/5886132/Tesi_Pecora_JSBSim.pdf)

I managed to do the first example without problem. I am now trying to introduce an external force as described in the report to reproduce a launching method. In the sgs233 configuration file, I add the following: 

```
<external_reactions>
<!-- "Declare" the force term that will be set in the script. -->
<property> fcs/external_force_bx_norm </property> 
<force name="APPLIED_EXTERNAL_FORCE" frame="BODY"> 
<function name="forces/external_force_bx_lbs">
<property> fcs/external_force_bx_norm </property>
 <value unit="LBS"> 50 </value>
</function>
<!--The location describes the point of application of the force,in this case it is CG -->
<location name="External_Force_at_CG" unit="IN">
 <x> 103.2 </x>
<y> 0 </y>
<z> -6.4 </z> 
</location>
<!-- The direction describes a unit vector. In this case, since the selected frame is the BODY frame, the "1" x component describes a direction exactly the same of the direction
into the body vector. That is, the direction specified below is the direction that the thrust force acts in. -->
   <direction>
        <x>1</x>
        <y>0</y>
        <z>0</z>
    </direction>
</force>
 </external_reactions>
```

I have tried to add this at different positions in my configuration file, but I still get the same error message. 
FATAL ERROR: JSBSim terminated with an unknown exception
[image]

Is it a problem in my configuration file or somewhere else? Where should I look to solve this? Thanks in advance! 

My script, configuration and initialization files: 
[sgs233.zip](https://github.com/JSBSim-Team/jsbsim/files/5886242/sgs233.zip)

seanmcleod, 2021-01-28:
I reformatted your `external_reaction` xml to make it easier to read. 

It looks like you have an invalid `<function>` definition, seem to be missing a `<product>` element?

```xml
      <function name="forces/external_force_bx_lbs">
         <product>
           <property> fcs/external_force_bx_norm </property>
           <value unit="LBS"> 50 </value>
         </product>
      </function>
```

sthelia, 2021-01-28:
Thank you! Yes, that was the problem.
I read a bit more about it in the [math documentation ](https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/user-manual-math/), if anyone else runs in to the same problem trying to reproduce this example.

closed 2021-01-28 by sthelia

seanmcleod, 2021-01-28:
There were also examples in the report that you attached 😉 

```xml
<external_reactions>
   <function>
      <product>
         <property> aero/qbar-psf </property>
         <property> fcs/parachute_reef_pos_norm </property>
         <value> 1.0 </value> <!-- Full drag coefficient -->
         <value> 20.0 </value> <!-- Full parachute area -->
      </product>
   </function>
</external_reactions>
```

seanmcleod, 2021-01-28:
Ah, but I see later on in the report on page 60 the report has the snippet you originally posted here with the `<product>` element missing.