JSBSim-Team/jsbsim #392: Scramjet engine design

Jacki43, 2021-02-03:
Hello all,

I want to model a scramjet engine with c++ software, and I am a beginner with JSBSIm. I wanted to ask you is it possible in JSBSim to model a scramjet engine? 
Can you please give me a brief guide on how can I do that?
Thanks in advance

seanmcleod, 2021-02-03:
I suggest you take a look at the source code for the existing engine models in `src\models\propulsion` as a first step to get familiar with how engine models are integrated in JSBSim. Which should give you an idea of how to go about adding a scramjet engine model.

Jacki43, 2021-02-04:
Thank you for your reply.
I think the engine model that exists in JSBSim which is similar to the scramjet engine, is FGTurboProp.cpp. Because both of them are airbreathing. Consequently, I should eliminate the compressor and turbine that are used in the turboprop model and add the nozzle and diffuser into that to implement a scramjet engine. Do you agree with that?
Is there any idea on how to consider diffuser and nozzle to implement scramjet?

Thanks in advance

seanmcleod, 2021-02-04:
I was suggesting more that you look at the engine models to get an idea of the classes involved and the methods you need to implement. I wouldn't try and start will all of the `FGTurboProp` or `FGTurbine` code and try and remove the compressor modeling.

What is the model (set of equations) you have for a scramjet, what are the inputs required including configuration options to then run through your equations to calculate the amount of thrust given the inputs?

bcoconni, 2021-02-05:
> I think the engine model that exists in JSBSim which is similar to the scramjet engine, is FGTurboProp.cpp. Because both of them are airbreathing. 

Please don't do that. I would recommend against using `FGTurboProp.cpp` or `FGTurbine.cpp` as a start for your scramjet model.
> Consequently, I think I should eliminate the compressor and turbine that are used in the turboprop model and add the nozzle and diffuser into that to implement a scramjet engine. Do you agree with that?

You can't. There are no compressor/turbine/nozzle in `FGTurboProp.cpp` and `FGTurbine.cpp`. They are not engineering models. You will not find any physics/thermodynamics model in them.

As @seanmcleod said, your best course of action is to build a new model from the ground up using the set of equations you have.

Basically, you need to:
* Write a new class `Scramjet` (or any name that you see fit) that inherits from [`FGEngine`](https://github.com/JSBSim-Team/jsbsim/blob/master/src/models/propulsion/FGEngine.h)
* Declare the new type of engine in `FGPropulsion.cpp`. For that, you need to add a new `if` condition with your model name `scramjet` and call the constructor of your class as in the code extract below.
https://github.com/JSBSim-Team/jsbsim/blob/c68ef9f26f55ed9920f66bd12a9b2fb2ed40ef38/src/models/FGPropulsion.cpp#L417-L425
* For a start you can skip fuel management and use stub methods for anything that is linked to fuel and tanks.
* You should rather use the `FGElectric` class for an example as it contains the bare minimum for an `FGEngine` class derivation.
  * In `FGElectric` you will see that the method `Calculate` is the place where the engine parameters are computed with respect to the throttle command, atmospheric properties, etc.

Jacki43, 2021-02-06:
Thank you so much @bcoconni, @seanmcleod.

I will consider your suggestion to design a scramjet engine. However, at first, I want to clarify something for myself. As you know, the `JSBSim-master\engine\PW125B.xml` is a turbojet. I wanted to know the trust equation of this [turbojet ](https://www.grc.nasa.gov/www/k-12/airplane/turbth.html)has been defined in which place of the code? Because, I want to apply the trust equation of [scramjet](https://www.grc.nasa.gov/www/k-12/airplane/ramth.html), but I do not know where can I define the parameters (for instance, mass flow rate, pressure, velocity, and area for stream and exit conditions).
Can you please let me know if there is an idea about that @bcoconni?

Thanks in advance

seanmcleod, 2021-02-06:
As @bcoconni mentioned use `FGElectric` as an example/template for your `FGScramjet` class. Also take a look at the `FGTurbine` class to see how it makes use of tables/functions in case you need that sort of functionality.

Looking at the NASA example you linked to:

[image]

At a minimum you would need to read in configuration parameters for the area of the exit nozzle Ae and area of the inlet nozzle etc. Here is an example showing how `FGTurbine` reads in the bypass ratio configuration option.

https://github.com/JSBSim-Team/jsbsim/blob/33d429e783589c27dccf2d70d231efe98c38f1df/src/models/propulsion/FGTurbine.cpp#L435-L436

Then in your `FGScramjet::Calculate` method you would make use of the configuration values plus make use of the input commands like the throttle command and atmospheric properties to calculate the thrust as per the equation shown in the NASA example.

The throttle command and atmospheric inputs are available via:

https://github.com/JSBSim-Team/jsbsim/blob/33d429e783589c27dccf2d70d231efe98c38f1df/src/models/propulsion/FGEngine.h#L117-L142

Then in `FGScramjet::Calculate` you'll have:

```c++
thrust = NASA equation;
Thruster->Calculate(thrust);
```

Jacki43, 2021-02-07:
Thank you so much for your detailed response. 

> Then in your `FGScramjet::Calculate` method you would make use of the configuration values plus make use of the input commands like the throttle command and atmospheric properties to calculate the thrust as per the equation shown in the NASA example.

As you mentioned, I applied the codes that were related to the bypass ratio configuration to the FGSceamjet. However, when I want to use the atmospheric properties in the thrust equation, I have the error that they are undefined. Is there any suggestion on how to read these properties? Should be added specific library to the FGScramjet?

Thanks in advance

bcoconni, 2021-02-07:
> However, when I want to use the atmospheric properties in the thrust equation, I have the error: the properties are undefined. Is there any suggestion on how to read these properties?

Could you please show us the lines of code that cause the compilation errors ?
Could you also show the error messages that you get ?

Jacki43, 2021-02-07:
Thank you for your reply. I only applied some changes in the FGElectric file that you mentioned to implement FGScramjet.
The line of code that I got error  is `  Thruster->Calculate(vt-vc);` . Please note that `vt-vc` is a sample thrust formula to test that whether the atmospheric properties can be read.
`void FGScramjet::Calculate(void)
{
  RunPreFunctions();

  if (Thruster->GetType() == FGThruster::ttPropeller) {
    ((FGPropeller*)Thruster)->SetAdvance(in.PropAdvance[EngineNumber]);
    ((FGPropeller*)Thruster)->SetFeather(in.PropFeather[EngineNumber]);
  } 

  RPM = Thruster->GetRPM() * Thruster->GetGearRatio();

  LoadThrusterInputs(); 
  Thruster->Calculate(vt-vc);

  RunPostFunctions();
}`

Also, the error is:
[image]

seanmcleod, 2021-02-08:
`Vt` and `Vc` are not member variables of `FGEngine`, take a look at the code snippet I shared earlier, they're part of a `struct`. If you take a look at the constructor for `FGElectric` or any of the other engine models you'll see a reference is passed to this struct.

```c++
FGElectric::FGElectric(FGFDMExec* exec, Element *el, int engine_number, struct FGEngine::Inputs& input)
  : FGEngine(engine_number, input)
{
```

Which is then passed to the constructor of `FGEngine` which is the base class. If you take a look at the `FGEngine` class you'll see it has the following member variable.

```c++
  struct Inputs& in;
```

You'll see this member variable `in` actually being used in the propeller code you copied.

Why have you copied the propeller related code into your scramjet model?

seanmcleod, 2021-02-08:
@Jacki43 I had mentioned in my first response - 

> I suggest you take a look at the source code for the existing engine models in src\models\propulsion as a first step to get familiar with how engine models are integrated in JSBSim. 

The idea being to understand how engine models are structured, what base classes to inherit from, where and how they get their configuration data, where and how they get their required input parameters from etc.

Jacki43, 2021-02-08:
Thank you so much for your response. I got the point.
I should use the `Vt`, `Vc` in this form: `Thruster->Calculate(in.Vt+in.Vc);`
I tested it and I do not have any error right now.

> Why have you copied the propeller related code into your scramjet model?

I just wanted to try to apply the thrust equation, so I did not check all of the things. As you mentioned, I should go through the sample engine codes and make sure I am designing the scramjet engine in the right way. 

It will be great if this issue stays open so that I can ask my questions and get your opinions that are very useful for me.
Thanks in advance

Jacki43, 2021-02-08:
Thank you so much for your help.

I am trying to implement the scramjet NASA thrust equation as I mentioned. However, all of the variables that I need, do not exist in `jsbsim/src/models/propulsion/FGEngine.h`. I need the pressure, mass flow rate, and velocity of both stream and exit conditions. 
Do you think for the exit condition (nozzle), should I use the variables that exist in `jsbsim/src/models/propulsion/FGNozzle.cpp`? 

Thanks in advance

seanmcleod, 2021-02-09:
Yes at some point in most cases you won't have every single variable available but will need to calculate and derive the variables you need from the existing variables.

So for example how would you derive the incoming mass flow rate?

Jacki43, 2021-02-09:
Thank you for your reply.
I rewrote the thrust equation in such a way that I use the existing variables. However, before applying I wanted to ask you please let me know what are the` Pressure ` and ` Density `. Are they related to the exit condition?
I found a [description](https://sourceforge.net/p/jsbsim/mailman/jsbsim-users/thread/CACdXh0C0_BQ9RifMmMskoF5mps5CkPRSZuyutk-LmYTVt0BQ0g@mail.gmail.com/) for `Vt` :

>  velocities/vt-fps is the total airspeed (in feet per second) : it is the
norm of the aircraft velocity with respect to the air measured in the body
frame.

Consequently, we have Ve= Vt-V0. Is it true?
Thanks in advance

seanmcleod, 2021-02-09:
> let me know what are the `Pressure` and `Density`. Are they related to the exit condition?

Did you search through the code to see where and how they're set?

https://github.com/JSBSim-Team/jsbsim/blob/193d28c20788bfd6d6daa27a9573c8a16d92acd8/src/FGFDMExec.cpp#L501-L525

So they're the free stream values, the 0 subscript values in the NASA equation.

Also, think about it, how would the generic base class `FGEngine` know anything about your specific engine to figure out how to calculate an exit value? What if you're modeling an electric engine?

Jacki43, 2021-02-09:
Thank you for your quick response. 

> Also, think about it, how would the generic base class FGEngine know anything about your specific engine to figure out how to calculate an exit value? What if you're modeling an electric engine?

I did not find `jsbsim/src/FGFDMExec.cpp`, so I was confused. 
Thank you so much.

closed 2021-02-12 by bcoconni