JSBSim-Team/jsbsim #397: Failing to build with MinGW64 on Windows

ElDesalmado, 2021-02-20:
Running CMake:
```
"C:\Program Files\CMake\bin\cmake.exe" -DCMAKE_BUILD_TYPE=Release -G Ninja C:\dev\repos\jsbsim
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/msys64/mingw64/bin/c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: C:/msys64/mingw64/bin/pkg-config.exe (found version "1.7.3") 
-- Found Doxygen: C:/Program Files/doxygen/bin/doxygen.exe (found version "1.8.18 (a1b07ad0e92e4526c9ba1711d39f06b58c2a7459)") found components: doxygen missing components: dot
-- Found PythonInterp: C:/msys64/mingw64/bin/python.exe (found version "3.8.7") 
-- Could NOT find Cython (missing: CYTHON_EXECUTABLE) 
CMake Warning at CMakeLists.txt:83 (message):
  JSBSim Python module and test suite will not be built

-- Could NOT find CxxTest (missing: CXXTEST_INCLUDE_DIR CXXTEST_PYTHON_TESTGEN_EXECUTABLE) 
-- Configuring done
-- Generating done
-- Build files have been written to: C:/dev/builds/jsbsim/Release-MinGW-w64

[Finished]
```

Building stops due to an error in `JSBSim.cpp`.
```
====================[ Build | all | Debug-MinGW-w64 ]===========================
"C:\Program Files\CMake\bin\cmake.exe" --build c:\dev\builds\jsbsim\Debug-MinGW-w64 --target all -- -j 8
[1/123] Building CXX object src/models/CMakeFiles/Models.dir/FGExternalReactions.cpp.obj
[2/123] Building CXX object src/models/CMakeFiles/Models.dir/FGAccelerations.cpp.obj
[3/123] Building CXX object src/models/CMakeFiles/Models.dir/FGBuoyantForces.cpp.obj
[4/123] Building CXX object src/models/CMakeFiles/Models.dir/FGExternalForce.cpp.obj
[5/123] Building CXX object src/models/CMakeFiles/Models.dir/FGGasCell.cpp.obj
[6/123] Building CXX object src/models/CMakeFiles/Models.dir/FGPropulsion.cpp.obj
[7/123] Building CXX object src/models/CMakeFiles/Models.dir/FGPropagate.cpp.obj
[8/123] Building CXX object src/models/CMakeFiles/Models.dir/FGInertial.cpp.obj
[9/123] Building CXX object src/models/CMakeFiles/Models.dir/FGInput.cpp.obj
[10/123] Building CXX object src/models/CMakeFiles/Models.dir/FGMassBalance.cpp.obj
[11/123] Building CXX object src/models/CMakeFiles/Models.dir/FGAuxiliary.cpp.obj
[12/123] Building CXX object src/models/CMakeFiles/Models.dir/FGLGear.cpp.obj
[13/123] Building CXX object src/models/CMakeFiles/Models.dir/FGGroundReactions.cpp.obj
[14/123] Building CXX object src/models/CMakeFiles/Models.dir/FGAtmosphere.cpp.obj
[15/123] Building CXX object src/models/CMakeFiles/Models.dir/FGSurface.cpp.obj
[16/123] Building CXX object src/CMakeFiles/libJSBSim.dir/FGJSBBase.cpp.obj
[17/123] Building CXX object src/models/CMakeFiles/Models.dir/FGModel.cpp.obj
[18/123] Building CXX object src/models/CMakeFiles/Models.dir/FGFCS.cpp.obj
[19/123] Building CXX object src/input_output/CMakeFiles/InputOutput.dir/FGGroundCallback.cpp.obj
[20/123] Building CXX object src/CMakeFiles/JSBSim.dir/JSBSim.cpp.obj
FAILED: src/CMakeFiles/JSBSim.dir/JSBSim.cpp.obj 
C:\msys64\mingw64\bin\c++.exe -DJSBSIM_VERSION=\"1.1.6\" -D_WIN32_WINNT=0x600 -IC:/dev/repos/jsbsim/src -g -std=gnu++11 -MD -MT src/CMakeFiles/JSBSim.dir/JSBSim.cpp.obj -MF src\CMakeFiles\JSBSim.dir\JSBSim.cpp.obj.d -o src/CMakeFiles/JSBSim.dir/JSBSim.cpp.obj -c C:/dev/repos/jsbsim/src/JSBSim.cpp
C:/dev/repos/jsbsim/src/JSBSim.cpp: In function 'int real_main(int, char**)':
C:/dev/repos/jsbsim/src/JSBSim.cpp:499:3: error: 'localtime_r' was not declared in this scope; did you mean 'localtime_s'?
  499 |   localtime_r(&tod, &local);
      |   ^~~~~~~~~~~
      |   localtime_s
[21/123] Building CXX object src/initialization/CMakeFiles/Init.dir/FGTrimAxis.cpp.obj
[22/123] Building CXX object src/models/CMakeFiles/Models.dir/FGOutput.cpp.obj
[23/123] Building CXX object src/input_output/CMakeFiles/InputOutput.dir/FGXMLParse.cpp.obj
[24/123] Building CXX object src/initialization/CMakeFiles/Init.dir/FGInitialCondition.cpp.obj
[25/123] Building CXX object src/CMakeFiles/libJSBSim.dir/FGFDMExec.cpp.obj
[26/123] Building CXX object src/initialization/CMakeFiles/Init.dir/FGTrim.cpp.obj
[27/123] Building CXX object src/input_output/CMakeFiles/InputOutput.dir/FGOutputSocket.cpp.obj
ninja: build stopped: subcommand failed.
```

ElDesalmado, 2021-02-20:
The problem can be solved by replacing `#ifdef _MSC_VER_` to `#if defined(_MSC_VER) || defined(__MINGW32__)`

bcoconni, 2021-02-20:
That's weird because [we are checking the compilation of JSBSim on MinGW32](https://github.com/JSBSim-Team/jsbsim/runs/1937712925?check_suite_focus=true) and it succeeds without an error. However after googling the issue, I came across the issue msys2/MINGW-packages#1342 and it seems that the flag `_POSIX_C_SOURCE` should be passed to the MinGW32 compiler for the build to succeed.

Could you try compiling with
```bash
> "C:\Program Files\CMake\bin\cmake.exe" -DCMAKE_CXX_FLAGS_RELEASE="-D_POSIX_C_SOURCE" --build c:\dev\builds\jsbsim\Debug-MinGW-w64 --target all -- -j 8 
```

> The problem can be solved by replacing `#ifdef _MSC_VER_` to `#if defined(_MSC_VER) || defined(__MINGW32__)`

Could you be more specific which line of code from which file are you talking about ?

ElDesalmado, 2021-02-21:
The thing is I've search through project files and found a place where the preprocessor checked not only for `_MSC_VER_` but also for `_MINGW32_`. So I just fixed all the lines where the preprocessor checks only for `_MSC_VER_` by adding `_MINGW32_`. 
That was not a lot of places to fix.
Here is the `git diff` 
```
PS C:\dev\repos\jsbsim> git diff master
diff --git a/src/JSBSim.cpp b/src/JSBSim.cpp
index 8f3828f4..b609bf0d 100644
--- a/src/JSBSim.cpp
+++ b/src/JSBSim.cpp
@@ -494,7 +494,7 @@ int real_main(int argc, char* argv[])
   time_t tod;
   time(&tod);
   struct tm local;
-#ifdef _MSC_VER
+#if defined(_MSC_VER) || defined(__MINGW32__)
   localtime_s(&local, &tod);
 #else
   localtime_r(&tod, &local);
@@ -565,7 +565,7 @@ int real_main(int argc, char* argv[])

   // PRINT ENDING CLOCK TIME
   time(&tod);
-#ifdef _MSC_VER
+#if defined(_MSC_VER) || defined(__MINGW32__)
   localtime_s(&local, &tod);
 #else
   localtime_r(&tod, &local);
diff --git a/src/models/flight_control/FGMagnetometer.cpp b/src/models/flight_control/FGMagnetometer.cpp
index bee2e7b6..14218bde 100644
--- a/src/models/flight_control/FGMagnetometer.cpp
+++ b/src/models/flight_control/FGMagnetometer.cpp
@@ -75,7 +75,7 @@ FGMagnetometer::FGMagnetometer(FGFCS* fcs, Element* element)
   time_t rawtime;
   time( &rawtime );
   struct tm ptm;
-  #ifdef _MSC_VER
...skipping...
@@ -75,7 +75,7 @@ FGMagnetometer::FGMagnetometer(FGFCS* fcs, Element* element)
   time_t rawtime;
   time( &rawtime );
   struct tm ptm;
-  #ifdef _MSC_VER
+  #if defined(_MSC_VER) || defined(__MINGW32__)
   gmtime_s(&ptm, &rawtime);
   #else
   gmtime_r(&rawtime, &ptm);
```

bcoconni, 2021-02-21:
Thanks.

Have you tried compiling with `-DCMAKE_CXX_FLAGS_RELEASE="-D_POSIX_C_SOURCE` and not making the source modifications above ?

bcoconni, 2021-03-21:
Just pushed the fix as per your proposal in commit edea29d (branch `master`) and fc940da (branch `release/v1.1`).

Could you please test and let us know if that fixes the issue that you have reported ?

ElDesalmado, 2021-03-21:
Everything compiles now

closed 2021-03-21 by ElDesalmado