JSBSim-Team/jsbsim #398: Was --suspend option deprecated?

MrMinos, 2021-03-05:
```
user@computer:~/jsbsim/build$ JSBSim --suspend
usage: JSBSim [-h] [--version] [--outputlogfile <filename>] [--logdirectivefile <filename>] [--root <path>] [--aircraft <filename>] [--script <filename>]
              [--initfile <filename>] [--end <time double>] [--realtime] [--nice]
              [input]
JSBSim: error: unrecognized arguments: --suspend
```
I've compiled the code with latest commit, on Ubuntu 20.04 running in WSL.

Note the parameter exists in the documentation [here](https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/quickstart-running-the-program/), but does not show up in the command helper doc.

I was hoping to use JSBSim FDM with Ardupilot SITL controller, and found this issue.

Jaeyoung-Lim, 2021-03-05:
@MrMinos Seems like you are on a different jsbsim version?

 If you are interested, we you can also run jsbsim with px4: https://github.com/Auterion/px4-jsbsim-bridge

seanmcleod, 2021-03-05:
Hmm, are you sure you're running the latest commit in master?

So, in the current repo:

https://github.com/JSBSim-Team/jsbsim/blob/518f2a74d1a5db65de3986816d3e6cd90521585e/src/JSBSim.cpp#L631-L633

https://github.com/JSBSim-Team/jsbsim/blob/518f2a74d1a5db65de3986816d3e6cd90521585e/src/JSBSim.cpp#L779-L781

And when I run JSBSim on my system. Ignore the version number etc., I'm building via Visual Studio project file.

```sh
(base) C:\source\jsbsim>.\Release\JSBSim.exe --suspend

     JSBSim Flight Dynamics Model v1.0.0 May  1 2020 18:32:05
            [JSBSim-ML v2.0]

JSBSim startup beginning ...

  No Aircraft, Script, or Reset information given
```

And

```sh
(base) C:\source\jsbsim>.\Release\JSBSim.exe --xx

  JSBSim version 1.0.0 May  1 2020 18:32:05

  Usage: jsbsim [script file name] [output file names] <options>

  options:
    --help  returns this message
    --version  returns the version number
    --outputlogfile=<filename>  sets (overrides) the name of a data output file
    --logdirectivefile=<filename>  specifies the name of a data logging directives file
                                   (can appear multiple times)
    --root=<path>  specifies the JSBSim root directory (where aircraft/, engine/, etc. reside)
    --aircraft=<filename>  specifies the name of the aircraft to be modeled
    --script=<filename>  specifies a script to run
    --realtime  specifies to run in actual real world time
    --nice  specifies to run at lower CPU usage
    --nohighlight  specifies that console output should be pure text only (no color)
    --suspend  specifies to suspend the simulation after initialization
    --initfile=<filename>  specifies an initilization file
    --catalog specifies that all properties for this aircraft model should be printed
              (catalog=aircraftname is an optional format)
    --property=<name=value> e.g. --property=simulation/integrator/rate/rotational=1
    --simulation-rate=<rate (double)> specifies the sim dT time or frequency
                      If rate specified is less than 1, it is interpreted as
                      a time step size, otherwise it is assumed to be a rate in Hertz.
    --end=<time (double)> specifies the sim end time

  NOTE: There can be no spaces around the = sign when
        an option is followed by a filename

The argument "--xx" cannot be interpreted as a file name or option.
(base) C:\source\jsbsim>
```

MrMinos, 2021-03-05:
> If you are interested, you can also run jsbsim with px4: https://github.com/Auterion/px4-jsbsim-bridge

Thanks @Jaeyoung-Lim I'm planning to support both flight controller stack! I do have a different issue with PX4 SITL with JSBSim -- which I would follow up in the appropriate forum (It relates to custom fixed wing model not moving within the simulation, when rascal model does).

@seanmcleod Thanks for the sanity check. It was a path issue. At some point `JSBSim` binary was placed at `~/.local/bin/JSBSim` and `make install` places the binary at `/usr/local/bin/JSBSim`. the `.local` folder was being referenced first!

Hindsight, should have ran the produced executable under `build/src/JSBSim` first. Thanks!

closed 2021-03-05 by MrMinos