JSBSim-Team/jsbsim #400: AeromatiC++ Segmentation fault

kwanzapili, 2021-03-19:
I downloaded jsbsim-1.1.5 and built on an opensuse linux platform so that I could run aeromatic. The consistent problem I am facing is a segmentation fault at the same step every time I run it. I go through the Q/A session and as soon as I enter "Systems", it fails. The last few of the session are as follows:

```
Htail` arm [ft] (0): 
Vtail area [ft2] (0): 150.76
Vtail arm [ft] (0): 

** Systems **

Segmentation fault (core dumped)

```
A brief look at the backtrace shows that it fails at:

```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x000055cfc4444441 in Aeromatic::Propulsion::param_reset() ()

```
Could someone please advise on what is going on?
Thanks.

kwanzapili, 2021-03-20:
I took a closer look at the code and found that the offending lines are the two commented out lines in the file _utils/aeromatic++/Systems/Propulsion.cpp_

```
void Propulsion::param_reset()
{
    _param = 0;
//    _propulsion[_ptype]->param_reset();
//    _propulsion[_ptype]->_thruster->param_reset();
}
```

With this modification, I can get the program to run without crashing. Is this the right thing to do, please?

bcoconni, 2021-03-21:
@kwanzapili thanks for reporting the issue.

Aeromatic++ is Erik Hoffman's area of expertise. @ermarch could you please review this issue and @kwanzapili's fix ?

ermarch, 2021-03-21:
Personally I think _propulsion[_ptype]->_thruster is NULL somehow and causes the segmentation fault.
It would be interesting to know which parameters you parsed to cause this.

ermarch, 2021-03-21:
Okay, I can only trigger this when I select an aircraft type number which is not listed.
The types range from 0..4 and only when I select 5 this bug is triggered.

kwanzapili, 2021-03-21:
For me the effect was very consistent regardless of the parameters I entered. For reference, by session was as follows:
```

** AeromatiC++ version 3.3.12
Aeromatiic is a JSBSim configuration file generation utility.
Please enter aircraft data when prompted.

You can always enter 'h' to get verbose help

** General Information **

Output directory [] (/home/kwanza): /home/kwanza/SR71
Create a subdirectory? [] (yes): 
Overwrite? [] (yes): 
Aircraft name [] (my_aircraft): SR71
Use dedicates System files? [] (yes): 
Select a system of measurement [] (no): 
 0: English (feet, pounds)
 1: Metric (meters, kilograms)
0
Type of aircraft [] (0): 
 0: Light General Aviation (Glider, Small Commuter)
 1: High Performance (WWII Fighter, Military Trainer, Aerobatic, Air Racer)
 2: Fighter Jet
 3: Jet Transport (Passenger Jet Airliner, Transonic Jet Transport)
 4: Propeller Transport (Propeller Airliner, Propeller Transport)
2

** Weight and Balance **

Stall speed VS1 (clean, no flaps) [kt] (0): 130
Maximum takeoff weight [lbs] (10000): 172000
Empty weight [lbs] (0): 67500
Inertia Ixx [slug/ft2] (0): 
Inertia Iyy [slug/ft2] (0): 
Inertia Izz [slug/ft2] (0): 

** Geometry **

Length [ft] (40): 107.41666667
Select a wing shape [] (0): 
 0: Straight
 1: Elliptical
 2: Delta
2
Wing span [ft] (40): 55.58333333
Wing area [ft2] (0): 1795
Wing aspect ratio [] (0): 1.939
Wing taper ratio [] (1): 
Wing root chord [ft] (0): 60.533
Wing incidence [] (2): 
Wing dihedral [] (0): 0
Wing sweep (quarter chord) [] (0): 52.629
Htail area [ft2] (0): 
Htail arm [ft] (0): 
Vtail area [ft2] (0): 150.76
Vtail arm [ft] (0): 

** Systems **

Then the crash. With the code commented out, I can proceed further.

Thanks.
```

ermarch, 2021-03-22:
I've got valgrind reporting an error when entering those numbers, so no doubt it will get triggered by other input as well.
I'll investigate, thanks for the report.

kwanzapili, 2021-03-23:
Incidentally, the program also produces some curious output for "Drag due to alpha", which leads to the simulator crashing/ I got the following:

```
        <function name="aero/force/Drag_alpha">
            <description>Drag due to alpha</description>
            <product>
                <property>aero/qbar-psf</property>
                <property>metrics/Sw-sqft</property>
                <table>
                    <independentVar lookup="row">aero/alpha-rad</independentVar>
                    <tableData>
                        -1.57    1.4082
                        -1.72    0.0246
                        0.00    0.0000
                        1.72    0.0246
                        1.57    1.4082
                    </tableData>
                </table>
            </product>
        </function>
```
Clearing the 1.72 and 1.57 are not is the proper sequence. Looking at some other configurations, I was tempted to infer that the -+1.72 should be closer to -+1.26, but I have no real estimate for this. Am I mistaken here?
Thanks.

ermarch, 2021-03-24:
Yes, that shouldn't have happened. So far I´ve not seen it with any of my test models but the SR-71 is rather extreme in every  way. Is there a chance you could run aeromatic again with the option "-l sr71.param" and post the param file?

This also comes handy when you want to rerun with a new version of aeromatic because you can then run "aeromatic -i sr71.param"

kwanzapili, 2021-03-24:
So I have run "aeromatic -l sr71.param" to get the following output. Please note that I am using AeromatiC++ version 3.3.12 without the commits you have made, but with the two lines I mentioned before commented out. So the aero file is this:

```
<?xml version="1.0"?>
<?xml-stylesheet type="text/xsl" href="http://jsbsim.sourceforge.net/JSBSim.xsl"?>

<fdm_config name="SR71" version="2.0" release="ALPHA"
   xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
   xsi:noNamespaceSchemaLocation="http://jsbsim.sourceforge.net/JSBSim.xsd">

 <fileheader>
  <author> Aeromatic v 3.3.12 </author>
  <filecreationdate> 24 Mar 2021 </filecreationdate>
  <version>$Revision: 1.80 $</version>
  <description> Models a SR71. </description>
 </fileheader>

<!--
  File:     SR71.xml
  Inputs:
    name:          SR71
    type:          Multi-engine Fighter Jet
    stall speed:   unspecified
    max weight:    172000.00 lb
    length:        107.42 ft
    wing: 
     span:         55.58 ft
     area:         1795.00 sq-ft
     mean chord:   60.53 ft
     aspect ratio: 1.94:1
     taper ratio:  1.00:1
     incidence:    2.00 degrees
     dihedral:     0.00 degrees
     sweep:        52.63 degrees

    no. engines:   2
    engine type:   Turbine Engine
    engine layout: wings

    control type:   Conventional with Yaw Damper

    gear type:     tricycle
    steering type: steering
    retractable?:  yes

  Outputs:
    wing loading:       95.82 lb/sq-ft
     - thickness ratio: 15.00%
    payload:            68896.00 lbs
    CL-alpha:           0.54 per radian
    CL-0:               0.08
    CL-max:             1.00
    CD-0:               0.01
    K:                  0.12
    Mcrit:              0.81

    min. turn radius    -nan ft
    max. turn rate:     -nan deg/s
-->

```

```
 <metrics>
   <wingarea  unit="FT2">  1795.00 </wingarea>
   <wingspan  unit="FT" >    55.58 </wingspan>
   <wing_incidence unit="DEG"> 2.00 </wing_incidence>
   <chord     unit="FT" >    60.53 </chord>
   <htailarea unit="FT2">   359.00 </htailarea>
   <htailarm  unit="FT" >    42.97 </htailarm>
[... 622 lines cut ...]
    <product>
     <property>aero/qbar-psf</property>
     <property>systems/chute/chute-reef-pos-norm</property>
     <property>systems/chute/chute-size-factor</property>
     <value> 0.9 </value>
     <value> 115 </value>
    </product>
   </function>
   <location unit="FT">
    <x> 97.7492 </x>
    <y> 0 </y>
    <z> 0 </z>
   </location>
   <direction>
    <x>-1 </x>
    <y> 0 </y>
    <z> 0 </z>
   </direction>
  </force>

 </external_reactions>

</fdm_config>

```

ermarch, 2021-03-25:
Oh I was hoping to get the created sr71.param file so I could reproduce your inputs locally.

kwanzapili, 2021-03-25:
I am sorry I misunderstood you completely. I didn't even realise that Aeromatic created this file until looked for it. Here it is.

```
SR71

SR71

0
2                             ; Type of aircraft
                              ; Stall speed VS1 (clean, no flaps)
172000                        ; Maximum takeoff weight
67500                         ; Empty weight
                              ; Inertia Ixx
                              ; Inertia Iyy
                              ; Inertia Izz
107.41666667                  ; Length
2                             ; Select a wing shape
55.58333333                   ; Wing span
1795                          ; Wing area
1.939                         ; Wing aspect ratio
                              ; Wing taper ratio
60.533                        ; Wing root chord
                              ; Wing incidence
0                             ; Wing dihedral
52.629                        ; Wing sweep (quarter chord)
                              ; Htail area
                              ; Htail arm
150.76                        ; Vtail area
                              ; Vtail arm

J58
2                             ; Number of engines
3                             ; Engine layout
2                             ; Engine type
25000                         ; Engine mil. thrust
0                             ; Bypass ratio
8.8                           ; Overall pressure ratio
yes

1                             ; Control system

0                             ; Nose or tail wheel type

no
no

yes

```

ermarch, 2021-03-26:
Great, thanks.
I'll dome some investigation.

ermarch, 2021-03-26:
The problem comes from calculating CLalpha for delta-wings. And I think the real problem is the fact that vortex lift is not taken into account.

bcoconni, 2021-03-26:
A new fix (commit 375f5be) has been pushed to `master`.
@kwanzapili could you please check ?

kwanzapili, 2021-03-26:
Thanks a lot. I revised the configuration and tested. My first observation was this key changes in the parameters:

```
* payload: 68896 lbs -> 104500 lbs
* CL-alpha: 0.54 -> 1.80 per radian

* htailarea 0 -> 359.00

* mass balance:
x: 136167.34 -> 118033.33
y: 699019.12 -> 740742.56
z: 585154.88 -> 556958.88
```

- Now, I am not clear what "htail-" is but I assume horizontal tail. I am not sure that the S-71 had a horizontal tail per se.
- For "aero/moment/Pitch_alphadot" should we not be using 'qbarUW-psf' instead of 'qbar-psf'? I had lots of problems with random wild oscillations when the plane was at standstill and the parking brakes on.
- The flying experience is very different now. The take-off is easier but slower. Previously, the transition to lift-off was very sudden taking the plane from 0 - 20 pitch instantly with elevators at about 40%. With the new configuration is is gentler but it also requires the plane to fly at over 10 degrees just for a very slow climb rate. This means I can never hit the target climb rate of over 9000 fpm.
- My take it that the lift it too low in the new configuration or something else is not quite right

bcoconni, 2021-03-26:
@kwanzapili Thanks for the bug report and testing !

Issues are dedicated to bug reports and feature requests: if you confirm that the bug you initially reported is now fixed then I suggest this issue should be closed and that the discussion should continue in [JSBSim Discussions](https://github.com/JSBSim-Team/jsbsim/discussions).

Thanks.

kwanzapili, 2021-03-27:
Sure, thanks a lot for fixing the problem.

closed 2021-03-27 by kwanzapili