JSBSim-Team/jsbsim #404: Libraries in Visual Studio

sthelia, 2021-03-25:
Hi again! 
For those who read my previous posts, my thesis is moving along nicely. I had to change my approach a bit and am now doing a Simulink interface using S-functions. It works nicely on my own Linux computer, and I am now trying to compile it on a Windows computer. (This is inspired by [this JSBSim-Matlab](https://github.com/podhrmic/JSBSim-Matlab) repo, I will post my solution once I have cleaned it up a bit) 

My problem now is on the Windows computer. Since I am using S-functions, I use the `mex` command in matlab. It looked like this for my Linuz computer: 
`mex -v -R2017b CXXFLAGS='$CXXFLAGS -std=c++14' ./matlab/JSBSim_SFunction.cpp ./matlab/JSBSimInterface.cpp -I./src -L./src/.libs -lJSBSim`

I am having trouble with the library files. I have tried to use Visual Studio 2019 and follow the JSBSim [user guide](https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/quickstart-building-using-visualstudio/), but it looks a bit different now than the screenshots in the user guide. I managed to build _something_, but I think it is a debug version because when I run my mex command I get these errors:
``` 
Error using mex
JSBSim.lib(sg_path.cxx.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in
JSBSim_SFunction.obj
JSBSim.lib(sg_path.cxx.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MDd_DynamicDebug' doesn't match value
'MD_DynamicRelease' in JSBSim_SFunction.obj
JSBSim.lib(FGFDMExec.cpp.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in
JSBSim_SFunction.obj
JSBSim.lib(FGFDMExec.cpp.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MDd_DynamicDebug' doesn't match value
'MD_DynamicRelease' in JSBSim_SFunction.obj
JSBSim.lib(FGPropertyManager.cpp.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in
JSBSim_SFunction.obj
JSBSim.lib(FGPropertyManager.cpp.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MDd_DynamicDebug' doesn't match value
'MD_DynamicRelease' in JSBSim_SFunction.obj
JSBSim.lib(FGMatrix33.cpp.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in
JSBSim_SFunction.obj

```
Google told me this has something to do with release vs debug configurations. Currently I'm locating them in the path `jsbsim\out\install\x64-Debug (default)\lib` since that is the only place I found lib-files, but I guess that is part of my problem since it's debug. 

 Does anyone have some advice on how I link to (or create?) the correct library file on Windows? 
Thanks in advance!

seanmcleod, 2021-03-25:
Yep, it looks like you've built the JSBSim_SFunction code as a release build and JSBSim has been built as a debug build.

So either build the SFunction code as a debug build or build JSBSim as a release build and link against the release version of the .lib.

sthelia, 2021-03-26:
Right, that is what I'm struggling with. I tried to figure out how to build it as a release, and a "Release"-directory did show up in the JSBSim directory. But I can't locate a .lib there. I've tried to play around a bit in Visual Studio to find how to correctly build a release build, but no luck. 
So my questions are: 
* How do I do a release build in Visual Studio 2019? 
* Where does the lib-file end up? 

Thanks for your reply!

seanmcleod, 2021-03-26:
Okay, so one potentially confusing issue is that we have 2 ways of building JSBSim using Visual Studio. 

The first option is to make use of the provided Visual Studio solution/project file to build JSBSim. Which is what the majority of Visual Studio users are most familiar with. However using this option the project file doesn't generate a .lib file, it simply generates an .exe. 

The 2nd option is to build using CMake, which is the default for non-Windows platforms, and the 'official' way to build JSBSim. In the CMake case a .lib file is first created and this is then later on linked in to also generate a .exe file.

The Visual Studio IDE from version 2017 onwards has included support in the IDE for building via CMake, see the following documentation which describes both Visual Studio options for building, i.e. via CMake and via the solution/project file.

https://jsbsim-team.github.io/jsbsim-reference-manual/mypages/quickstart-building-using-visualstudio/

So in your case you'll need to make use of the CMake option. You'll see in the CMake build documentation references to debug and release.

closed 2021-03-26 by bcoconni