JSBSim-Team/jsbsim #436: Aeromatic alpha value output

kwanzapili, 2021-06-06:
**I'm submitting a ...**
  - [X] bug report
  - [X] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
1. For the bug part, Aeromatic does not generate a logical "_aero/force/Lift_alpha_" function. The "_aero/alpha-rad_" values are in indeterminate order
2. Aeromatic does **not** offer generate a propulsion system or a drag parachute. The output then states that the aircraft has no engine

**What is the current behavior?**
I used the parameters used in the attached file "_vulcan.log_" and the output is shown in "_vulcan.txt_". See the first function in the "LIFT" axis.

The program outputs the following error when run;
```
*** ERROR: The alpha value for maximum lift is too high.
           This means the specified Stall Speed was too low.
           Make sure it is for a clean (no gear and no flaps) configurtion.

```
I am not sure what stall speed it expects. I have tried entering nothing and other reasonable values without any better luck.

**What is the expected behavior?**
In Aeromatic v1.1.5, we had the offer to generate propulsion and other system tiles. This seems to have be omitted in v1.1.6.
The alpha lift behaviour seems to be similar to issue #400 that I had reported before.

**What is the motivation / use case for changing the behavior?**
Simply revert to prior features.

**Please tell us about your environment:**
-  Linux
-  JSBSim 1.1.6

**Other information**
Refer to issue #400 for some prior problems with the output.
[vulcan.log](https://github.com/JSBSim-Team/jsbsim/files/6604709/vulcan.log)
[vulcan.txt](https://github.com/JSBSim-Team/jsbsim/files/6604710/vulcan.txt)

bcoconni, 2021-06-14:
@kwanzapili Thanks for the bug report.

Just to clarify things: when you are referring to `Aeromatic`, are you talking about the [Aeromatic web site](http://jsbsim.sourceforge.net/aeromatic2.html) or are you talking about the command line utility `aeromatic++` ?

kwanzapili, 2021-06-18:
Sorry for the lapse in responding. I mean the Aeromatic CLI that you get from compiling the JSBSim tar ball (`AeromatiC++ version 3.3.14`). I have tried two web-based versions: one never generates any output (it rigidly stays at the input form page) and the other works - (http://jsbsim.sourceforge.net/aeromatic2.html). The only reason I was trying the CLI version was because I wanted finer controls of the inputs.

bcoconni, 2021-06-18:
@ermarch This is an issue linked with aeromatic++. Could you please review it ?
Thanks.

ermarch, 2021-06-18:
A quick test makes me believe this is an issue with delta wings.
None of the other configurations suffer from it and if I set the stall speed to 200 it also works for the delta wing.

The error arises from the fact that the lift table will have a non ascending alpha scale.
What you could do for now is simplify the table like this:
```
<function name="aero/force/Lift_alpha">
      <description>Lift due to alpha</description>
      <product>
          <property>aero/qbar-psf</property>
          <property>metrics/Sw-sqft</property>
          <table>
            <independentVar lookup="row">aero/alpha-rad</independentVar>
            <tableData>
              -1.61 -0.7635
               0.00  0.2586
               1.21  1.0221
               1.22  0.7429
               1.57  0.0000
            </tableData>
          </table>
      </product>
    </function>
```

But I will need to take a closer look at why this happens, and how to resolve it.

bcoconni, 2021-06-23:
@kwanzapili the fix is now in the `master` branch. Unfortunately, the automatic build of executables is failing at the moment due to an issue detected by our tests. So you'll need to build `aeromatic++` yourself to get the latest version or wait for the problem reported by our test suite being fixed.

bcoconni, 2021-06-23:
Well, actually the automatic build has succeeded so you can download the executables for Windows from our rolling release [JSBSim-1.2.0.dev1-502-setup.exe](https://github.com/JSBSim-Team/jsbsim/releases/download/Linux/JSBSim-1.2.0.dev1-502-setup.exe).

Please test and let us know if that fixes the issue you have reported.

ermarch, 2021-06-24:
I noticed the failure, but also concluded that it would be unrelated. Thanks fro applying the patch.

kwanzapili, 2021-06-25:
Thanks a lot.
I've just downloaded and compiled "_jsbsim-master.zip_" and then re-run _aeromatic_. I appear to get similar results as before, i.e.`

```
*** ERROR: The alpha value for maximum lift is too high.
           This means the specified Stall Speed was too low.
           Make sure it is for a clean (no gear and no flaps) configurtion.

```
This is with a stall speed of 120 knots (200 knots would be well above VR for this aircraft). The alpha table looks misaligned as before:
```

    <function name="aero/force/Lift_alpha">
      <description>Lift due to alpha</description>
      <product>
          <property>aero/qbar-psf</property>
          <property>metrics/Sw-sqft</property>
          <table>
            <independentVar lookup="row">aero/alpha-rad</independentVar>
            <tableData>
              -1.57  0.0000
              -1.22 -0.5302
              -1.05 -0.7143
              -0.88 -0.8248
              -1.08 -0.3866
              -1.90 -0.9665
               0.00  0.3272
               1.42  1.2936
               0.60  0.7138
               0.88  1.1752
               1.05  1.0177
               1.22  0.7554
               1.57  0.0000
            </tableData>
          </table>
      </product>
    </function>

```
Unless, I am missing something, I think this is where we were before?

ermarch, 2021-06-26:
This sound like AeromatiC++ wasn't  built from the latest code.

This is how the table should look like if i use the vulcan configuration you provided earlier (with a stall speed of 130kts):
```
<function name="aero/force/Lift_alpha">
      <description>Lift due to alpha</description>
      <product>
          <property>aero/qbar-psf</property>
          <property>metrics/Sw-sqft</property>
          <table>
            <independentVar lookup="row">aero/alpha-rad</independentVar>
            <tableData>
              -1.57  0.0000
              -1.22 -0.5427
              -1.05 -0.7312
              -0.88 -0.8443
              -0.48 -0.2785
               0.00  0.1304
               0.55  1.0221
               0.62  0.2394
               0.88  1.1557
               1.05  1.0008
               1.22  0.7429
               1.57  0.0000
            </tableData>
          </table>
      </product>
    </function>
```

bcoconni, 2021-06-26:
> I've just downloaded and compiled `jsbsim-master.zip`

Did the archive contained the folder `/jsbsim-master/utils/aeromatic++/samples/` ? If not, then may be you should try [downloading the file again now](https://github.com/JSBSim-Team/jsbsim/archive/refs/heads/master.zip).

kwanzapili, 2021-06-26:
I have downloaded and recompiled the source, and this time I get no errors. The source I got yesterday had the samples directory you mention. Perhaps it was simply a timing issue. I've not actually examined the output in detail, but I can see that the alpha values are now properly aligned.

One question I had asked before was why it does not generate the engine file? It asks if the plane has a reverser, then does nothing more on engines.

Thanks a lot.

ermarch, 2021-06-27:
> One question I had asked before was why it does not generate the engine file? It asks if the plane has a reverser, then does nothing more on engines.

It should, but after testing it I see it doesn't. For none of the configurations I tested.
I ll have to look into that.

kwanzapili, 2021-06-29:
Great, the tool now generates the engine configuration.

There a couple of observations I should make. Firstly, the engine configuration has _inf_ for _TSFC_. I did not see any input for TSFC. So the partial output looks like:

```
<turbine_engine name="Olympus_301">
  <milthrust> 20000.0 </milthrust>
  <bypassratio>     1.000 </bypassratio>
  <tsfc>            inf </tsfc>
  <bleed>           0.03</bleed>
  <idlen1>         30.0 </idlen1>
  <idlen2>         60.0 </idlen2>
  <maxn1>         100.0 </maxn1>
  <maxn2>         100.0 </maxn2>
  <augmented>         0 </augmented>
  <injected>          0 </injected>
```

Second point is on the lift coefficient due to alpha. Whereas the row variable (alpha) is now aligned, I  am a bit confused by the column entries. Look at the change between alpha 0.55 and 0.59 in the output below. I would also be cautious about the first three rows.

```
    <function name="aero/force/Lift_alpha">
      <description>Lift due to alpha</description>
      <product>
          <property>aero/qbar-psf</property>
          <property>metrics/Sw-sqft</property>
          <table>
            <independentVar lookup="row">aero/alpha-rad</independentVar>
            <tableData>
              -1.57  0.0000
              -1.22 -0.5427
              -1.05 -0.7312
              -0.88 -0.8443
              -0.51 -0.3331
               0.00  0.0757
               0.55  1.0221
               0.59  0.2406
               0.88  1.1557
               1.05  1.0008
               1.22  0.7429
               1.57  0.0000
            </tableData>
          </table>
      </product>
    </function>
```

My wild guess is that curve would be something like:

```
	<function name="aero/force/Lift_alpha">
	    <description>Lift due to alpha</description>
	    <product>
		<property>aero/qbarUW-psf</property>
		<property>metrics/Sw-sqft</property>
		<table>
		    <independentVar lookup="row">aero/alpha-rad</independentVar>
		    <tableData>
                        -0.88	-0.844
                        -0.48	-0.279
                        0.00	0.130
                        0.17	0.500
                        0.35	1.000
                        0.44	1.200
                        0.52	1.100
                        0.55	1.022
                        0.70	0.800
                        0.87	0.700
                        1.05	0.600
                        1.22	0.400
                        1.40	0.200
                        1.57	0.000
		    </tableData>
		</table>
	    </product>
	</function>

I might be wrong since I simply followed the approach in [](https://www.icas.org/ICAS_ARCHIVE/ICAS2002/PAPERS/273.PDF) without much understanding.

Thanks a lot.

ermarch, 2021-07-01:
I'll have to look at the TFS problem.
Fir the alpha table: If you think you've got a better match from elsewhere then by all means replace the one generated by AeromatiC++.

AeromatiC++ is design to provide a first, working, setup from a small number of inputs. It tries to be as accurate as possible with those input but it will always be limited in it's accuracy.

ermarch, 2021-07-04:
The NaN's turn out to be generated by using an incomplete parameter file (it was created when not all parameters where processed). I've included a new configuration file for the Vulcan which solves this:
[Vulcan configuration](http://adalin.com/ehtw.info/vulcan.param)

kwanzapili, 2021-07-04:
Thanks a lot, that's great.

One final observation which might be completely useless, but might be helpful for someone who encounters the same issue. When the plane is on the ground at standstill, JSBsim sometimes randomly assigns an alpha degree of ~178 - 180. Otherwise it assigns an alpha close to 0. I have found that if the alpha lift coefficient is set to zero when alpha is ~180, then alpha will stay at that value whatever throttle is applied. The result is that the plane will never gain enough speed and never lift off. This happens for settings such as:
```
        <function name="aero/force/Lift_alpha">
            <description>Lift due to alpha</description>
            <product>
                <property>aero/qbar-psf</property>
                <property>metrics/Sw-sqft</property>
                <table>
                    <independentVar lookup="row">aero/alpha-rad</independentVar>
                    <tableData>
                        -1.57  0.0000
                        -1.22 -0.5427
                        -1.05 -0.7312
                        -0.88 -0.8443
                        -0.48 -0.2785
                        0.00  0.1304
                        0.55  1.0221
                        0.62  0.2406
                        0.88  1.1557
                        1.05  1.0008
                        1.22  0.7429
                        1.57  0.0000
                    </tableData>
                </table>
            </product>
        </function>

```
Now if we add a non-zero alpha coefficient above 90 degrees, JSBsim will magically reset the alpha to something nearer 0 if the plane begins to move. Then everything should work as expected. For example, if we add a line:

`3.14    0.100
`

Perhaps this is a situation peculiar to me. It took me a while to figure out the issue and the workaround I devised.

closed 2021-07-04 by kwanzapili

seanmcleod, 2021-07-04:
@kwanzapili in terms of the alpha value I was recently taking a look in that section of the code.

https://github.com/JSBSim-Team/jsbsim/blob/375f5be0fdfeb210f4cf66f938e3e6fd426c2966/src/models/FGAuxiliary.cpp#L151-L169

So on each simulation step alpha is initialized to 0 and then only if `Vt > 0.001` and `mUW >= 1e-6` is it then calculated based on the U and W velocities.

So the entries you have in a Cl vs alpha table really shouldn't make a direct difference. The only thing I can think of is that you have some drag table which results in massive drag with large alpha values which prevents the aircraft from accelerating so that the velocity UVW stays very small.

kwanzapili, 2021-07-05:
Thanks for the insight. The thing that really puzzles me is that on startup, _alpha-deg = 178.64_ and _alpha-rad = 3.1177_. These are not reset to 0 until I manage to change the velocity somewhat. if alpha is initialized to 0, then somewhere else it is set to 178 again. That is the mystery I cannot understand. An additional aspect is that alpha might revert to 178 on landing and when the velocity drops to 0. But these occurrences appear random.

You are correct about the drag. I had to eliminate much of it and I finally settled on the table below.

```
        <function name="aero/force/Drag_alpha">
            <description>Drag due to alpha</description>
            <product>
                <property>aero/qbar-psf</property>
                <property>metrics/Sw-sqft</property>
                <table>
                    <independentVar lookup="row">aero/alpha-rad</independentVar>
                    <tableData>
                        -1.57    1.4081
                        -0.15    0.0295
                        0.00    0.0000
                        0.15    0.0295
                        1.57    1.4081
                        1.58    -1.2000
                        2.98    -0.0295
                        3.14    0.0000
                    </tableData>
                </table>
            </product>
        </function>

```
As you can see, I have had to use negative drag for alpha above 90 (I know this is naughty, but that is the easiest way I could overcome the high alpha problem).

seanmcleod, 2021-07-05:
Plot/log the `UVW` velocities so that you can see whether the calculated alpha makes sense. Do you have some wind configured?

> The result is that the plane will never gain enough speed and never lift off.

Also plot/log the throttle command, thrust force and the aerodynamic forces, to see which one is blowing up in order to prevent the aircraft gaining enough speed. 

Take a look at https://github.com/JSBSim-Team/jsbsim/discussions/455 to see an example of the forces logged etc.

kwanzapili, 2021-07-05:
I'd not sure about what the UVW velocities, so I had a look at u-aero-fps etc and the alpha values are startup. The data shows the following pattern (full data set attached): You will notice, it rises sharply from -1.5 too 160 degrees, where it settles down.

```
Time	alpha-rad	alpha-deg	u-aero-fps	v-aero-fps	w-aero-fps
0	0	        0	        0	        0	        0
0.05	-0.0266         -1.5240	        4.8377	        1.7227	        -0.1287
1	-0.0272	        -1.5608	        4.8390	        1.7227	        -0.1319
5	-0.0257	        -1.4751	        3.6066	        -0.8555	        -0.0929
10	-0.0212	        -1.2123	        2.0922	        -4.0238	        -0.0443
15	0.0082	        0.4696	        0.5703	        -7.2081	        0.0047
20	2.8009	        160.4810	-0.0709	        -8.4761	        0.0251
30	2.8009	        160.4808	-0.0709	        -8.4761	        0.0251
```

This happens without me doing anything. So it is purely JSBsim assigning random number to alpha. It seems to like numbers around -1.5 to 0 degrees and 160 to 180 degrees. As I said before, it I can coax the plane into sufficient motion, then alpha will trend towards 0.Now one import of this is that I must **never** use aero/alphadot-rad_sec to calculate _Lift due to alpha rate_, for if I did, then the plane will get into horrendous oscillations on the ground as if it was in an earthquake. So my wild guess is something is **seriously wrong** with the way alpha is computed at startup. My workaround was merely masking the problem at high alpha.

One other major puzzle I have is why my pitch is set to -2 at startup even with the controls centered? It also happens in level flight. Could you please point me to the potential causes?

Thanks.
[data.csv](https://github.com/JSBSim-Team/jsbsim/files/6766010/data.csv)

seanmcleod, 2021-07-05:
> So it is purely JSBsim assigning random number to alpha.

JSBSim definitely doesn't assign **random** values to alpha.

```sh
>>> math.degrees(math.atan2(0.0251, -0.0709))
160.50503155537436
```

So the question is what is causing these velocities.

> This happens without me doing anything.

So this is the model simply sitting at rest on the ground when you start JSBSim? What aircraft model is this?

I asked earlier if there is any wind configured?

My guess based on your data and description so far and based on an issue from a while ago is that maybe it's similia to this issue - https://github.com/JSBSim-Team/jsbsim/issues/89

kwanzapili, 2021-07-05:
Sorry I should have mentioned that I am modelling a Vulcan which was the reason I used Aeromatic to get some initial parameters. I use the JSBSim Atmosphere model by setting the parameter `control-fdm-atmosphere` to `false`. I imagine that JSBSim decides on the wind conditions and thus the velocities. I do not interfere at all with the weather. What I do practically it start up the simulation and then fold my arms and watch the wild dance of alpha. To be fair to it, sometimes it settles at around zero on its own, in which case I can safely take off. My gripe is when it elects to hang around 180. if indeed JSBsim is changing the wind on a whim, I don't understand why the values always result in alpha at the ends of the range -1.5 to 180?

I was aware of the issue of sudden jumps when the plane is at standstill with the brakes on, having experienced it before with a different plane. For this plane I use QBarUW with formula involving alpha. This has largely resolved that issue except where aero/alphadot-rad_sec is involved.

seanmcleod, 2021-07-05:
Are you using JSBSim stand-alone or are you using it in conjunction with FlightGear?

JSBSim doesn't 'just decide` on wind conditions, any wind needs to be set, it defaults to no wind, hence my questions about wind being set.

kwanzapili, 2021-07-05:
I should have mentioned I always go through FlightGear with JSBsim as the FDM. On the weather conditions, I set nothing at all personally. All I have set is: time of day, season, real weather, basic weather. Beyond that I leave it to the FDM. When testing I use the same airport and runway, but I can see that the weather does change because the side winds can be very different on different runs (even within 5 minutes). I know this because I need different rudder settings to keep the plane in a straight line.

ermarch, 2021-07-06:
Could you post the aircraft configuration you are using?

When I run a modified version of the Vulcan using AeromatiC++ I get some strange output and it looks like it still doesn't handle delta wings really well.

seanmcleod, 2021-07-06:
> All I have set is: time of day, season, real weather, basic weather. Beyond that I leave it to the FDM. 

JSBSim as the FDM doesn't have a high-level option/facility to set things like weather. It only has lower-level options for wind to be specified and/or a turbulence model. By default there is no wind and no turbulence. So FlightGear must be setting the wind etc.

So just to confirm, if you set FlightGear to the equivalent of no weather and no turbulence do you still see these issues?

seanmcleod, 2021-07-06:
> I'd not sure about what the UVW velocities, so I had a look at u-aero-fps etc and the alpha values are startup.

In terms of the difference between the UVW velocities, i.e. 

`velocities/u-aero-fps, velocities/v-aero-fps, velocities/w-aero-fps`

versus

`velocities/u-fps, velocities/v-fps, velocities/w-fps`

The properties with `aero` in the property name include the wind.

```c++
  // Combine the wind speed with aircraft speed to obtain wind relative speed
  vAeroUVW = in.vUVW - in.Tl2b * in.TotalWindNED;
```

So if your aircraft is sitting still on the ground with the brakes on, and assuming no bouncing around, then the `velocities\uvw-fps` should be zero and `velocities\uvw-aero-fps` will give you the wind vector.

kwanzapili, 2021-07-06:
I have finally realised how to set the wind (silly me) and the experiment I ran appears to confirm that the alpha is highly sensitive to the wind direction but perhaps not the speed. What I did was:

- start the simulation,
- wait for it to do its alpha dance, 
- once settled, set the wind speed to 0
- allow it settle once again

The results are shown below:

```
Time	alpha-rad	alpha-deg	windspeed-kt	wind deg	u-aero-fps	v-aero-fps	w-aero-fps
0.00	0	        0	        0	        0	        0	        0       	0
0.05	-0.027	        -1.5	        3.04	        270.1   	4.84	        1.72	        -0.13
7.00	-0.036	        -2.1    	2.19	        325.1	        0.99    	3.57	        -0.04
8.00	-0.054  	-3.1    	2.30    	335.1	        0.37	        3.86	        -0.02
8.45	-0.147  	-8.4	        2.37    	339.3	        0.09	        4.00	        -0.01
8.50	-0.287  	-16.5	        2.38	        340.0	        0.04	        4.02	        -0.01
8.55	-0.443  	-25.4   	2.39	        340.2	        0.02	        4.03	        -0.01
8.60	-2.625  	-150.4	        2.40	        340.8	        -0.02	        4.05	        -0.01
9.00	-3.127  	-179.2	        2.48	        344.4	        -0.28	        4.17	        0.00
9.25	-3.141  	-180.0	        2.53    	346.4	        -0.43	        4.24	        0.00
9.30	3.141   	179.9	        2.54    	346.6	        -0.45	        4.25	        0.00
10.00	3.128   	179.2	        2.71    	352.1	        -0.92	        4.48	        0.01
20.00	3.123   	178.9	        4.02    	0.1	        -2.27	        6.39	        0.04
31.85	3.123   	178.9	        3.94	        0.1	        -2.22	        6.27	        0.04
31.90	3.123   	178.9	        3.93    	0.0	        -2.21	        6.25	        0.04
35.00	3.123   	178.9	        3.02    	0.0	        -1.69	        4.80	        0.03
40.00	3.123   	178.9   	1.53    	0.0	        -0.86	        2.44	        0.02
45.10	3.123   	178.9   	0.01    	0.0	        -0.01	        0.02	        0.00
45.15	3.124	        179.0   	0.00	        0.0	        0.00	        0.01	        0.00
45.20	0.000	        0.0	        0.00    	0.0	        0.00	        0.00	        0.00
```

I make the following observations:

- alpha immediately drops to -1.5 degrees (which is also the pitch for the plane at rest)
- the wind drifts from 270 degrees towards 350 while settling at about 2.4 knots
- the alpha value begins to change rapidly once the wind direction exceeds 339
- at a wind direction of ~346.5, alpha inverts its sign (hmmmmm ...)
- at around time 20, I open the wing dialog - the default wind speed was 4 knots, hence the change
- setting the wind speed to 0 takes a while to be reflected in the measurements; only by time 45 it is finally 0
- alpha has its own delay before it finally resets to 0 (hurrah!)
- if values for alpha above 90 degrees are legitimate, then I suppose my approach of using negative drag was not too wrong
- I still wish the rate of change of alpha was controlled somewhat rather the free-fall behaviour ii appears to displays around certain points.

I have attached the configuration I currently use. The coefficients were initially based on Aeromatic output, but I had to change a few drastically in order to get stable flights. For instance, I had to increase the coefficient for pitch damp 10-fold (based on a wild guess).
[vulcanb2.txt](https://github.com/JSBSim-Team/jsbsim/files/6770966/vulcanb2.txt)

seanmcleod, 2021-07-06:
> appears to confirm that the alpha is highly sensitive to the wind direction but perhaps not the speed. 

Remember I showed the code used by JSBSim to calculate alpha. So it simply depends on `u-aero-fps` and `w-aero-fps`.

```c++
alpha = atan2(vAeroUVW(eW), vAeroUVW(eU));
```

And this calculation is only performed if the following conditions are true, otherwise alpha is set to 0.

```c++
if ( Vt > 0.001 )

    if ( mUW >= 1E-6 )
```

> alpha has its own delay before it finally resets to 0 (hurrah!)

Note size of these conditions, in your results above you only display 2-decimal places. So even if your results show 0.00 for U and W that doesn't mean they aren't still big enough to be used to calculate alpha.

> setting the wind speed to 0 takes a while to be reflected in the measurements; only by time 45 it is finally 0

This must be FlightGear's weather system ramping the wind down.

> I still wish the rate of change of alpha was controlled somewhat rather the free-fall behaviour

Again, JSBSim isn't implementing some sort of 'free-fall behaviour', it's simply using the formula listed above based on the U and W velocities. Which in your case with the aircraft at rest on the ground then the values for alpha that you see and it's rate of change is simply based on the wind vector that FlightGear is feeding into JSBSim.

For example here is a photo I took of the alpha and beta vanes that were used on the first FTI system I worked on. While standing on the ground if there was a light breeze from behind the aircraft for example then the alpha vane would rotate to 180 degrees etc.

[image]

kwanzapili, 2021-07-06:
Thanks for the clarifications. If I have understood the issue a bit, then I might conclude:

- if the wind direction changes, then alpha will change
- alpha can vary from -180 to 180 degrees
- therefore it is wish to define coefficients that take into account alpha below -90 and above 90 degrees whenever alpha is used

If this is true, then it confirms my original suggestion of using negative drag at the start when alpha > 90. Alpha will revert to < 90 once the plane has enough motion. Why not just set alpha to zero until the plane begins to move?

seanmcleod, 2021-07-07:
Sure in general alpha can vary between -180 to +180 degrees. But that doesn't mean you have to have data for things like Cl vs alpha across the complete 360 degree range.

For example level-D airline flight simulators don't bother with Cl vs alpha across the full 360 degree range, typically they only have data for just past the stall.