JSBSim-Team/jsbsim #440: How should the product of inertia be computed?

aarondewindt, 2021-06-14:
On some sources such as "Aircraft Control and Simulation: Dynamics, Controls Design, and Autonomous Systems" by "Brian L. Stevens", the product moment of inertia are computed with.

[image]

However on some other sources the product of inertia is computed the equation shown below. So this would be the value in the inertia tensor.

[image]

After looking through the JSBSim source code we think we should fill in values calculated by the second equation. Is this the case?

seanmcleod, 2021-06-14:
See the following:

https://github.com/JSBSim-Team/jsbsim/blob/a35be5ce6f7d7ca0ef9a11a62563e54d769bc4f4/src/models/FGMassBalance.h#L68-L71

Which was updated as part of the following commit - https://github.com/JSBSim-Team/jsbsim/commit/03763374fcd688bdcf9faf6164106a68f8ba3c0b#diff-ef28318d289aee0dd2e968ada735838f3e7ad8daf7da1eaea2448ed27ec56b4a

With further info in this earlier issue - https://github.com/JSBSim-Team/jsbsim/issues/125

aarondewindt, 2021-06-14:
@seanmcleod Thanks!!

closed 2021-06-14 by bcoconni