JSBSim-Team/jsbsim #485: Linear Actuator behaving strangely

bcoconni, 2021-07-17:
### Discussed in https://github.com/JSBSim-Team/jsbsim/discussions/483

<div type='discussions-op-text'>

<sup>Originally posted by **Octal450** July 16, 2021</sup>
Hi,
I am modelling an analogue HSI instrument by use of a linear actuator, followed by an actuator + lag filter.

I notice some really strange behavior in the linear actuator when there are quick movements about the changeover point specified by the module tag (in my case 359 -> 0 or 0 -> 359

I have observed this affect with the heading bug implementation as well as with the heading card showing current heading, showing the issue is not related to integers vs floating point values.

Properties to take note of:
`heading-input` is magnetic heading in
`heading-linear-cmd` is the linear-actuator

In this image it is working correctly, instead of jumping to 359, it went negative.
[image]

I will now kick the rudder left and right profusely to move the aircraft through the 359 -> 0 or 0 -> 359 transition several times and it works most of the time correctly, then sometimes, it does this:

[image]

Huh? it just jumped up to 720 from roughly 0. So now since I have an actuator following the linear actuator, the HSI card rotated (in this case) 2 full revolutions due to the jump from near 0 all the way to 720 (meaning the bug occurred twice, usually its just once) which is the exact behavior I was trying to avoid. This is very strange.

I will also point out it doesn't matter if the linear is currently negative, positive, or with a large number from built up revolutions, or still small near 0 or near +/- 360, the bug still happens.

My code is as follows:
```xml
		<linear_actuator name="/instrumentation/hsi[0]/heading-linear-cmd">
			<input>/instrumentation/hsi[0]/heading-input</input>
			<module>360</module>
		</linear_actuator>
		
		<actuator name="/instrumentation/hsi[0]/heading-linear-actuator">
			<input>/instrumentation/hsi[0]/heading-linear-cmd</input>
			<rate_limit>90</rate_limit>
		</actuator>
```

Am I doing something wrong or did I uncover a bug :D 

Kind Regards,
Josh</div>

bcoconni, 2021-07-17:
This is a genuine bug: I've added a test [`tests/TestLinearActuator.py`](https://github.com/JSBSim-Team/jsbsim/blob/master/tests/TestLinearActuator.py) to trigger its occurence.
@abassign 
When the sequence 1, 2, 359 is supplied to the linear actuator the following output is obtained: 1, 2, 719. But it should be 1, 2, -1.

Octal450, 2021-07-17:
Thanks @bcoconni 

Kind Regards,
Josh

bcoconni, 2021-07-17:
A tentative fix is pushed in `master`.

If you plan to test it on FlightGear then please replace the content of the FlightGear source file `src/FDM/JSBSim/models/Flight_control/FGLinearActuator.cpp` by [this content](https://raw.githubusercontent.com/JSBSim-Team/jsbsim/master/src/models/flight_control/FGLinearActuator.cpp).

bcoconni, 2021-07-17:
Of course, let us know the outcome of your tests :smiley:

Octal450, 2021-07-17:
@bcoconni 
I will see if I can figure out how to build FG this weekend to test your fix... 

Kind Regards,
Josh

abassign, 2021-07-17:
@bcoconni @Octal450 
Hi, I have tried your modification and I have verified that it works great. Therefore it can be introduced in FGFS.
I did various tests, especially on the TACAN-ARN-21 navigation gauge which in the G91R1B corresponds to a magnetic compass repeater and ADF and TACAN / VOR direction indicator.
But I give an indication, the solution of ... is perhaps too simplistic. I think it is better to use a different code which makes the movement of the hand closer to reality, this is an example:
```
       <linear_actuator name="systems/gauges/radio/Collins-331-C4/indicated-bearing-deg-la">
            <input>systems/gauges/radio/ADF-Marconi-8283/indicated-bearing-deg</input>
            <module>360</module>
            <reset>systems/gauges/radio/TACAN-NAV-ARN-21/isNot-Active</reset>
        </linear_actuator>
        
        <actuator name="systems/gauges/radio/Collins-331-C4/indicated-bearing-deg">
            <input>systems/gauges/radio/Collins-331-C4/indicated-bearing-deg-la</input>
            <lag>1.0</lag>
            <rate_limit> 90.0 </rate_limit>
            <deadband_width> 0.5 </deadband_width>
            <hysteresis_width> 0.5 </hysteresis_width>
        </actuator>
```
Remember that the reset function allows the hand to position itself on zero when, for example, there is no power supply and / or signal.

Octal450, 2021-07-18:
@abassign Thanks for the info and thanks for checking the patch. I almost managed to build FG and will finish up tomorrow and verify the bug is resolved on my end -> then I will continue the work and I will check out other parameters. 

Thanks.

Kind Regards,
Josh

bcoconni, 2021-07-18:
> Hi, I have tried your modification and I have verified that it works great. Therefore it can be introduced in FGFS.

Thanks @abassign  :+1: 
>  I almost managed to build FG and will finish up tomorrow and verify the bug is resolved on my end -> then I will continue the work and I will check out other parameters.

OK we'll wait for your feedback @Octal450.

Octal450, 2021-07-18:
Didn't get it building yet... will keep trying. (Windows)

Kind Regards,
Josh

Octal450, 2021-07-19:
@bcoconni Got FlightGear to build and run. I was using the wrong OSG version. Derf.

OK, now inserted your file and rebuild the software. But I get a problem, the compilation is not successful.

```
E:\Downloads\FGFS\Build\flightgear\src\FDM\JSBSim\models\flight_control\FGLinearActuator.cpp(59,77): error C2661: 'JSBS
im::FGParameterValue::FGParameterValue': no overloaded function takes 3 arguments [E:\Downloads\FGFS\Build\flightgear-b
uild\src\FDM\JSBSim\JSBSim.vcxproj]
E:\Downloads\FGFS\Build\flightgear\src\FDM\JSBSim\models\flight_control\FGLinearActuator.cpp(68,79): error C2661: 'JSBS
im::FGParameterValue::FGParameterValue': no overloaded function takes 3 arguments [E:\Downloads\FGFS\Build\flightgear-b
uild\src\FDM\JSBSim\JSBSim.vcxproj]
E:\Downloads\FGFS\Build\flightgear\src\FDM\JSBSim\models\flight_control\FGLinearActuator.cpp(77,80): error C2661: 'JSBS
im::FGParameterValue::FGParameterValue': no overloaded function takes 3 arguments [E:\Downloads\FGFS\Build\flightgear-b
uild\src\FDM\JSBSim\JSBSim.vcxproj]
E:\Downloads\FGFS\Build\flightgear\src\FDM\JSBSim\models\flight_control\FGLinearActuator.cpp(86,78): error C2661: 'JSBS
im::FGParameterValue::FGParameterValue': no overloaded function takes 3 arguments [E:\Downloads\FGFS\Build\flightgear-b
uild\src\FDM\JSBSim\JSBSim.vcxproj]```

Kind Regards,
Josh

seanmcleod, 2021-07-19:
@Octal450 `FGParameterValue` was updated in this commit, Dec 4 2020 - https://github.com/JSBSim-Team/jsbsim/commit/d3e008a8e45ed19def750a2a23b2c4c9906b46f1#diff-36356a669a4b42d7e5ce30d0a51d93096b7f68755f5dd8385910022b804efd1c

The version in FlightGear is obviously older than this, so it isn't as trivial as copying in the 1 file @bcoconni linked to.

Octal450, 2021-07-19:
@seanmcleod I just cloned all the repos fresh yesterday -> should it not have the latest file?

Anyways, I just applied the diff manually to the file. Great work @bcoconni , works perfect now! Thank you very kindly! Can be closed.

Question, When that whole will planned to be added to FG next? (Reason, so that I know when I can enable linear actuator, as for now I must keep it not used)

Kind Regards,
Josh

seanmcleod, 2021-07-19:
> @seanmcleod I just cloned all the repos fresh yesterday -> should it not have the latest file?

Doesn't FlightGear include the JSBSim source within it's repo? I don't use FlightGear. My understanding is @bcoconni periodically pushes an update from the JSBSim repo into the FlightGear repo. So if he hadn't pushed from JSBSim to FlightGear after the 4 Dec 2020 then the JSBSim source within the FlightGear wouldn't have the changes based on that commit.

Unless you were also pulling the JSBSim repo directly and overwriting the JSBSim source included in the FlightGear repo?

Octal450, 2021-07-19:
I was using FlightGear source, not JSBsim. I used this to build: https://wiki.flightgear.org/Building_using_CMake_-_Windows

Thanks for the infos.

Kind Regards,
Josh

bcoconni, 2021-07-19:
> Anyways, I just applied the diff manually to the file. Great work @bcoconni , works perfect now! Thank you very kindly! Can be closed.

Great ! Thanks for testing 👍

> I just cloned all the repos fresh yesterday -> should it not have the latest file?

My mistake, I forgot that FlightGear is using [JSBSim v1.1.5](https://github.com/JSBSim-Team/jsbsim/releases/tag/v1.1.5). Unfortunately the source code that I have suggested is based on the development branch `master` which has a different API for some methods/functions.

> Question, When that whole will planned to be added to FG next? (Reason, so that I know when I can enable linear actuator, as for now I must keep it not used)

I was waiting for your feedback before proceeding (@abassign already confirmed that it's working for him as well). I'll take this opportunity to release a new version `v1.1.8` and push this new release to FlightGear. Hopefully, some time later this week.

Octal450, 2021-07-19:
@bcoconni Great, thank you! Your efficiency and diligence is to be highly commended.

Kind Regards,
Josh

abassign, 2021-07-20:
@bcoconni 
The test I did it simply by inserting the correction you proposed in the current code present in FGFS. I myself don't understand why I made such a logical mistake, but your fix works fine. However, the correction is done in this way if we want to test it on the current FGFS:
Edit the file:
**FGLinearActuator.cpp**
place in the path: _... /flightgear/src/FDM/JSBSim/models/flight_control/_
with the following code:
``` 
/* ***** Old code: ***** from line 185 *****
        if ((versus == 0) || (versus == direction)) {
          inputMem = Input;
          if (direction != 0) {
            if (abs(inputDelta) >= (module*rate)) {
              if (direction > 0) {
                countSpin++;
                direction = 0;
              } else if (direction < 0) {
                countSpin--;
                direction = 0;
              }
            }
          }
        } else if ((versus != 0) && (direction != 0) && (versus != direction)) {
          inputLast += inputDelta;
        }
***** end old code ***** */
/* ***** new code ***** */
        if ((versus == 0) || (versus == direction)) {
            inputMem = Input;
            if (abs(inputDelta) >= (module*rate)) {
                if (inputDelta < 0)
                    countSpin++;
                else
                    countSpin--;
            }
        } else if ((versus != 0) && (direction != 0) && (versus != direction)) {
            inputLast += inputDelta;
        }
```
I then recompiled everything and it worked perfectly.
Obviously I hope soon to see the fix propagated on the FGFS version.

bcoconni, 2021-07-24:
The fix has just been pushed to FlightGear `next` branch along with JSBSim release v1.1.8.
Issue closed.

closed 2021-07-24 by bcoconni

Octal450, 2021-08-01:
Thanks very much @bcoconni !

Kind Regards,
Josh