JSBSim-Team/jsbsim #491: Setting up the the JSBSim interface to Matlab on Windows

agodemar, 2021-08-04:
Hello JSBSimmer!

I have managed to download and install on Windows the [JSBSim release v1.1.8](https://github.com/JSBSim-Team/jsbsim/releases/tag/v1.1.8) that includes the interface to Matlab. 
**Yet, the interface doesn't work right out of the box unless you add some headers and do some extra work on Matlab side.**
Here's what you can do to make it work:

1. **Download the installer and had JSBSim set up in the following local directory tree:**
```
<JSBSim root>/
   aeromatic++/
   aircraft/
   data_output/
   engine/
   include/
   lib/
   matlab/
   scripts/
   systems/
   JSBSim.exe
   msvcp140.dll
```
The exact location `<JSBSim root>/` on your system depends on where you choose to install JSBSim.

2. **There are some header files missing in the current distribution**, namely all `*.hxx` files related to `simgear`. You have to copy all those missing header files, including their directory structure, [from the source tree](https://github.com/JSBSim-Team/jsbsim/tree/master/src/simgear) to the following local installation tree:
```
<JSBSim root>/
   ...
   include/
      simgear/
         io/
            iostreams/
         magvar/
         misc/
         props/
         structure/
         xml/
```
**@bcoconni This is an important step that should be fixed when packaging the new JSBSim setup executable on Windows.**

3. **Now, let's see the extra work you have to do on the Matlab side.** 

The following steps are tested for *Matlab v2020b*, which I have installed on my system.

To compile the file `<JSBSim root>/matlab/JSBSim_SFunction.cpp` and get the MEX-File `JSBSim_SFunction.mexw64` on Windows, you need a working C++ compiler. Among the available options, I suggest you to go for [Visual Studio 2019 Community Edition](https://visualstudio.microsoft.com/).

Once you have a working C++ development environment, from Matlab command line you need to tell the MEX utility what is the default C++ compiler. You do this with the following command:

`[Matlab prompt]>> mex -setup CPP`

MEX will detect your C++ compiler and will be configured to build your `*.mexw64` files.

4. **Adjusting the Matlab script `<JSBSim root>/matlab/JSBSimSimulinkCompile.m` to build your own MEX-file.**

First of all, copy the script `<JSBSim root>/matlab/JSBSimSimulinkCompile.m` to your own version `<JSBSim root>/myJSBSimSimulinkCompile.m` **in the root of my JSBSim installation directory**. This makes sense, because we want a Matlab EXecutable (MEX) that lives besides `<JSBSim root>/JSBSim.exe`, and must reach all input files in the directories `aircraft`, `engine`, `scripts`, and `systems`.

This is how the script **`<JSBSim root>/myJSBSimSimulinkCompile.m`** should look like:

```
disp('Compiling S-function from JSBSim...');
mex -v -R2017b COMPFLAGS='$COMPFLAGS' ./matlab/JSBSim_SFunction.cpp ./matlab/JSBSimInterface.cpp -I".\include" -L".\lib" -lJSBSim  wsock32.lib ws2_32.lib
disp('Finished.')
```

If you point the current *Matlab Working Directory* to `<JSBSim root>/` and run the above M-file from the Matlab editor or from the command line:

`[Matlab prompt]>> myJSBSimSimulinkCompile`

you'll get the MEX-file `<JSBSim root>/JSBSim_SFunction.mexw64`.

5. **Testing your MEX-file `<JSBSim root>/JSBSim_SFunction.mexw64`.**
[... 512 lines cut ...]
Creating input UDP socket on port 5139
Successfully bound to UDP input socket on port 5139

  Mass Properties Report (English units: lbf, in, slug-ft^2)
                                      Weight    CG-X    CG-Y    CG-Z         Ixx         Iyy         Izz         Ixy         Ixz         Iyz
    Base Vehicle                     83000.0   639.0     0.0   -40.0    562000.0   1473000.0   1894000.0        -0.0      8000.0        -0.0
0   Fuel                               10000     520     -80     -18           0           0           0
1   Fuel                               10000     520      80     -18           0           0           0
2   Fuel                                4000     480       0     -18           0           0           0
                                                                                                                                            
    Total:                          107000.0   610.8     0.0   -35.1    591572.3   1539552.7   1986235.4         0.0     19109.1         0.0

End of vehicle configuration loading.
-------------------------------------------------------------------------------

Simulation completed.
Remember to reset the program by typing clearSF in the matlab command window! 
JSBSim S-Function Reset
```

and in Simulink you'll have the following outputs:

[image]

So you just have to try it out and adapt this workflow to your needs.

huweius, 2021-08-04:
Dear Prof. De Marco, 

Thanks a lot for the detailed instruction. I followed all the steps and still arrived at some fatal errors: 

`JSBSimInterface.obj : error LNK2019: unresolved external symbol "public: __cdecl JSBSim::FGFDMExec::FGFDMExec(class JSBSim::FGPropertyManager *,class std::shared_ptr<unsigned int>)"
(??0FGFDMExec@JSBSim@@QEAA@PEAVFGPropertyManager@1@V?$shared_ptr@I@std@@@Z) referenced in function "public: __cdecl JSBSimInterface::JSBSimInterface(double)" (??0JSBSimInterface@@QEAA@N@Z)
JSBSimInterface.obj : error LNK2019: unresolved external symbol "public: class std::shared_ptr<class JSBSim::FGAccelerations> __cdecl JSBSim::FGFDMExec::GetAccelerations(void)const "
(?GetAccelerations@FGFDMExec@JSBSim@@QEBA?AV?$shared_ptr@VFGAccelerations@JSBSim@@@std@@XZ) referenced in function "public: __cdecl JSBSimInterface::JSBSimInterface(double)"
(??0JSBSimInterface@@QEAA@N@Z)
JSBSimInterface.obj : error LNK2019: unresolved external symbol "public: class std::shared_ptr<class JSBSim::FGFCS> __cdecl JSBSim::FGFDMExec::GetFCS(void)const "
(?GetFCS@FGFDMExec@JSBSim@@QEBA?AV?$shared_ptr@VFGFCS@JSBSim@@@std@@XZ) referenced in function "public: __cdecl JSBSimInterface::JSBSimInterface(double)" (??0JSBSimInterface@@QEAA@N@Z)
JSBSimInterface.obj : error LNK2019: unresolved external symbol "public: class std::shared_ptr<class JSBSim::FGPropulsion> __cdecl JSBSim::FGFDMExec::GetPropulsion(void)const "
(?GetPropulsion@FGFDMExec@JSBSim@@QEBA?AV?$shared_ptr@VFGPropulsion@JSBSim@@@std@@XZ) referenced in function "public: __cdecl JSBSimInterface::JSBSimInterface(double)"
(??0JSBSimInterface@@QEAA@N@Z)
JSBSimInterface.obj : error LNK2019: unresolved external symbol "public: class std::shared_ptr<class JSBSim::FGAerodynamics> __cdecl JSBSim::FGFDMExec::GetAerodynamics(void)const "
(?GetAerodynamics@FGFDMExec@JSBSim@@QEBA?AV?$shared_ptr@VFGAerodynamics@JSBSim@@@std@@XZ) referenced in function "public: __cdecl JSBSimInterface::JSBSimInterface(double)"
(??0JSBSimInterface@@QEAA@N@Z)
JSBSimInterface.obj : error LNK2019: unresolved external symbol "public: class std::shared_ptr<class JSBSim::FGAircraft> __cdecl JSBSim::FGFDMExec::GetAircraft(void)const "
(?GetAircraft@FGFDMExec@JSBSim@@QEBA?AV?$shared_ptr@VFGAircraft@JSBSim@@@std@@XZ) referenced in function "public: bool __cdecl JSBSimInterface::Open(class std::basic_string<char,struct
std::char_traits<char>,class std::allocator<char> > const &)" (?Open@JSBSimInterface@@QEAA_NAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)
JSBSimInterface.obj : error LNK2019: unresolved external symbol "public: class std::shared_ptr<class JSBSim::FGPropagate> __cdecl JSBSim::FGFDMExec::GetPropagate(void)const "
(?GetPropagate@FGFDMExec@JSBSim@@QEBA?AV?$shared_ptr@VFGPropagate@JSBSim@@@std@@XZ) referenced in function "public: __cdecl JSBSimInterface::JSBSimInterface(double)"
(??0JSBSimInterface@@QEAA@N@Z)
JSBSimInterface.obj : error LNK2019: unresolved external symbol "public: class std::shared_ptr<class JSBSim::FGAuxiliary> __cdecl JSBSim::FGFDMExec::GetAuxiliary(void)const "
(?GetAuxiliary@FGFDMExec@JSBSim@@QEBA?AV?$shared_ptr@VFGAuxiliary@JSBSim@@@std@@XZ) referenced in function "public: __cdecl JSBSimInterface::JSBSimInterface(double)"
(??0JSBSimInterface@@QEAA@N@Z)
JSBSim_SFunction.mexw64 : fatal error LNK1120: 8 unresolved externals`

The environment should be the same as you: JSBSim release 1.1.8 installer, MATLAB R2020b Update 6, Visual C++ 2019 compiler. I copied and replaced everything in the `\include` directory with everything from `\src` of the master branch. Is there anything else I've missed? 

Thanks a lot for the assistance.

agodemar, 2021-08-05:
@huweius it looks like the mex compiler doesn't see the class `JSBSim::FGFDMExec`, which is packaged in the library file
`<JSBSim root>/lib/JSBSim.lib` 

You should double check the arguments of your mex command. In my system I have this command within the script file `<JSBSim root>/myJSBSimSimulinkCompile.m`, which is placed in the `<JSBSim root>/` directory, one level above the other directories.
The command is the following:

```
mex -v -R2017b COMPFLAGS='$COMPFLAGS' ./matlab/JSBSim_SFunction.cpp ./matlab/JSBSimInterface.cpp -I".\include" -L".\lib" -lJSBSim  wsock32.lib ws2_32.lib
```

where 

- the option `-L".\lib"` tells the linker to search for `*.lib` files in `<JSBSim root>/lib`
- the option `-lJSBSim` tells the linker to link in the file `<JSBSim root>/lib/JSBSim.lib`

Dim1303, 2021-09-13:
Hello, 
I'm new here and I'm trying to install the JSBsim(v1.1.8) to use with matlab/simulink.
I followed all those steps described above, they really fixed some errors, but it still not working. Other errors are occurring during the compilation process in matlab. The compile are looking for the file "./simgear/props/propertyObject.hxx". I tryied to find it in the JSBSim repository but I didn't found too.
Has anyone had this error? 
Thanks in advance.

Below is the copy of matlab command screen, with the error:

Building with 'MinGW64 Compiler (C++)'.
C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\mingw_w64.instrset\bin\g++ -c -DMX_COMPAT_64  -DMATLAB_MEXCMD_RELEASE=R2017b  -DUSE_MEX_CMD   -m64 -DMATLAB_MEX_FILE  -I".\include"  -I"C:\Program Files\MATLAB\R2018a/extern/include" -I"C:\Program Files\MATLAB\R2018a/simulink/include" -fexceptions -fno-omit-frame-pointer -std=c++11 -O -DNDEBUG "C:\JSBSim\matlab\JSBSim_SFunction.cpp" -o C:\Users\dimit\AppData\Local\Temp\mex_428443625313531_13980\JSBSim_SFunction.obj
Error using mex
In file included from
.\include/input_output/FGPropertyReader.h:44:0,
                 from .\include/math/FGModelFunctions.h:38,
                 from .\include/models/FGModel.h:43,
                 from .\include/models/FGPropagate.h:41,
                 from .\include/FGFDMExec.h:47,
                 from C:\JSBSim\matlab\JSBSim_SFunction.cpp:133:
.\include/input_output/FGPropertyManager.h:45:44: fatal error:
simgear/props/propertyObject.hxx: No such file or directory
compilation terminated.
Error in JSBSimSimulinkCompile (line 6)
mex -v -R2017b COMPFLAGS='$COMPFLAGS'
./matlab/JSBSim_SFunction.cpp ./matlab/JSBSimInterface.cpp
-I".\include" -L".\lib" -lJSBSim  wsock32.lib ws2_32.lib

agodemar, 2021-09-13:
@Dim1303 I see that you are using Matlab v2018a. My tests are valid for Matlab v2020a and v2020b. In addition, you are using MinGW64 Compiler (C++); my suggestion is to use Visual Studio 2019 Community Edition.

Dim1303, 2021-09-13:
Professor De Marco, thank you so much for the support.

I have uninstalled and installed the JSBSim to do the modifications step by step (just to check), now I'm using the Matlab v2020a with the MEX configured to use 'Microsoft Visual C++ 2019 (C)' for C language compilation. The error is a little bit different now:

--
Building with 'Microsoft Visual C++ 2019'.
cl /c /Zp8 /GR /W3 /EHs /nologo /MD /O2 /Oy- /DNDEBUG /DMX_COMPAT_64  /DMATLAB_MEXCMD_RELEASE=R2017b  /DUSE_MEX_CMD   /D_CRT_SECURE_NO_DEPRECATE /D_SCL_SECURE_NO_DEPRECATE /D_SECURE_SCL=0   /DMATLAB_MEX_FILE -I".\include"  -I"C:\Program Files\Polyspace\R2020a\extern\include" -I"C:\Program Files\Polyspace\R2020a\simulink\include" "C:\JSBSim\matlab\JSBSim_SFunction.cpp" /FoC:\Users\dimit\AppData\Local\Temp\mex_483405025434860_14092\JSBSim_SFunction.obj
Error using mex
JSBSim_SFunction.cpp
C:\JSBSim\include\input_output/FGPropertyManager.h(45): fatal error C1083: NÆo ‚
poss¡vel abrir arquivo incluir: 'simgear/props/propertyObject.hxx': No such file or
directory

Error in myJSBSimSimulinkCompile (line 5)
mex -v -R2017b COMPFLAGS='$COMPFLAGS' ./matlab/JSBSim_SFunction.cpp
./matlab/JSBSimInterface.cpp -I".\include" -L".\lib" -lJSBSim  wsock32.lib ws2_32.lib

agodemar, 2021-09-13:
@Dim1303 you are right, the file `propertyObject.hxx` was missing in the link that I gave in my original post. I have edited the hyperlink that now [points to this zip archive, released as v1.1.8 source code](https://github.com/JSBSim-Team/jsbsim/archive/refs/tags/v1.1.8.zip). 
Then, follow Step 2 of my original post:

2. Copy all `*.hxx` files from the folder `simgear`, including the whole directory structure, from the path `<JSBSim root>/src/simgear` to `<JSBSim root>/include/simgear` and get the following tree:
```
<JSBSim root>/
   ...
   include/
      simgear/
         io/
            iostreams/
         magvar/
         misc/
         props/
         structure/
         xml/
``` 
At this point your compilation should work fine.

Dim1303, 2021-09-14:
Thank you so much @agodemar. It worked perfectly.

bcoconni, 2021-09-17:
> **There are some header files missing in the current distribution**, namely all `*.hxx` files related to `simgear`.

That's weird, the following line is supposed to include all files named `*.hxx` and process recursively all the folders under `src/` i.e. including `src/simgear`
https://github.com/JSBSim-Team/jsbsim/blob/c6856458ad5f160a65b1336d79ac0679f3540b26/JSBSim.iss.in#L34
> Download [from the sources](https://github.com/JSBSim-Team/jsbsim/tree/master/matlab) the following files: `TestJSBSim.m` and `ex737cruise.slx` and save them in `<JSBSim root>/`, besides `<JSBSim root>/JSBSim_SFunction.mexw64`.
>
> **@bcoconni These two files are missing from those packaged in the JSBSim setup executable on Windows. Probably this is fine, but we'll have to adjust our readme files to tell users how to start working with them on Matlab.**

OK we could provide them under a new installer category `Matlab test files` or similar so that they could be installed optionally.

huweius, 2021-09-18:
I'm afraid I'm still getting a different error for building this S-function despite copying all of the required headers and following Prof. de Marco's instructions. Compiler used was MEX with VS 2019. How could this type of type-casting error exist? 
```
Error using mex
JSBSimInterface.cpp
C:\JSBSim\matlab\JSBSimInterface.cpp(41): error C2440: '=': cannot convert from 'std::shared_ptr<JSBSim::FGPropagate>' to 'JSBSim::FGPropagate *'
C:\JSBSim\matlab\JSBSimInterface.cpp(41): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
C:\JSBSim\matlab\JSBSimInterface.cpp(42): error C2440: '=': cannot convert from 'std::shared_ptr<JSBSim::FGAccelerations>' to 'JSBSim::FGAccelerations *'
C:\JSBSim\matlab\JSBSimInterface.cpp(42): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
C:\JSBSim\matlab\JSBSimInterface.cpp(43): error C2440: '=': cannot convert from 'std::shared_ptr<JSBSim::FGAuxiliary>' to 'JSBSim::FGAuxiliary *'
C:\JSBSim\matlab\JSBSimInterface.cpp(43): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
C:\JSBSim\matlab\JSBSimInterface.cpp(44): error C2440: '=': cannot convert from 'std::shared_ptr<JSBSim::FGAerodynamics>' to 'JSBSim::FGAerodynamics *'
C:\JSBSim\matlab\JSBSimInterface.cpp(44): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
C:\JSBSim\matlab\JSBSimInterface.cpp(45): error C2440: '=': cannot convert from 'std::shared_ptr<JSBSim::FGPropulsion>' to 'JSBSim::FGPropulsion *'
C:\JSBSim\matlab\JSBSimInterface.cpp(45): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
C:\JSBSim\matlab\JSBSimInterface.cpp(46): error C2440: '=': cannot convert from 'std::shared_ptr<JSBSim::FGFCS>' to 'JSBSim::FGFCS *'
C:\JSBSim\matlab\JSBSimInterface.cpp(46): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
C:\JSBSim\matlab\JSBSimInterface.cpp(56): error C2440: '=': cannot convert from 'std::shared_ptr<JSBSim::FGPropagate>' to 'JSBSim::FGPropagate *'
C:\JSBSim\matlab\JSBSimInterface.cpp(56): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
C:\JSBSim\matlab\JSBSimInterface.cpp(57): error C2440: '=': cannot convert from 'std::shared_ptr<JSBSim::FGAccelerations>' to 'JSBSim::FGAccelerations *'
C:\JSBSim\matlab\JSBSimInterface.cpp(57): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
C:\JSBSim\matlab\JSBSimInterface.cpp(58): error C2440: '=': cannot convert from 'std::shared_ptr<JSBSim::FGAuxiliary>' to 'JSBSim::FGAuxiliary *'
C:\JSBSim\matlab\JSBSimInterface.cpp(58): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
C:\JSBSim\matlab\JSBSimInterface.cpp(59): error C2440: '=': cannot convert from 'std::shared_ptr<JSBSim::FGAerodynamics>' to 'JSBSim::FGAerodynamics *'
C:\JSBSim\matlab\JSBSimInterface.cpp(59): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
C:\JSBSim\matlab\JSBSimInterface.cpp(60): error C2440: '=': cannot convert from 'std::shared_ptr<JSBSim::FGPropulsion>' to 'JSBSim::FGPropulsion *'
C:\JSBSim\matlab\JSBSimInterface.cpp(60): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
C:\JSBSim\matlab\JSBSimInterface.cpp(61): error C2440: '=': cannot convert from 'std::shared_ptr<JSBSim::FGFCS>' to 'JSBSim::FGFCS *'
C:\JSBSim\matlab\JSBSimInterface.cpp(61): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called

Error in JSBSimSimulinkCompile (line 5)
mex -v -R2017b COMPFLAGS='$COMPFLAGS' ./matlab/JSBSim_SFunction.cpp ./matlab/JSBSimInterface.cpp -I".\include" -L".\lib" -lJSBSim  wsock32.lib ws2_32.lib
```

bcoconni, 2021-09-18:
@huweius you most likely downloaded the header files from than `master` branch and compiled with the library from `1.1.8`.
Is the problem fixed when you download headers from the `1.1.8` branch (https://github.com/JSBSim-Team/jsbsim/tree/v1.1.8) ?

huweius, 2021-09-20:
> @huweius you most likely downloaded the header files from the `master` branch and compiled them along with the library from the version `1.1.8` installer.
> Is the problem fixed when you download headers from the `1.1.8` branch (https://github.com/JSBSim-Team/jsbsim/tree/v1.1.8) ?

Thanks! That does solve it.

bcoconni, 2021-11-07:
>> That's weird, the following line is supposed to include all files named `*.hxx` and process recursively all the folders under `src/` i.e. including `src/simgear` [...]
>
> Well, I just realized this line has been modified *after* `1.1.8` was released... That explains it :disappointed:

This should now be fixed in JSBSim version `1.1.9` that has just been released :tada:

bcoconni, 2021-11-08:
> Download [from the sources](https://github.com/JSBSim-Team/jsbsim/tree/master/matlab) the following files: `TestJSBSim.m` and `ex737cruise.slx` and save them in `<JSBSim root>/`, besides `<JSBSim [...]

Oh man ! I forgot to include `TestJSBSim.m` and `ex737cruise.slx` in the Windows installer 🤦

bcoconni, 2021-11-08:
This has been fixed by the commit 131517fcd338a6b3f29b905e8bd901371d0d4798. I also took this opportunity to add the license files to the Windows installer by the way.

agodemar, 2021-11-09:
@bcoconni thanks
this JSBSim interface to Matlab/Simulink topic is getting hot on LinkedIn.
[This video](https://www.linkedin.com/posts/agostino-de-marco-08398a7_matlab-jsbsim-flightgear-activity-6861761772991332352-ep_F) that I've posted has got more than 18k visualizations in less than one week ;-)

bcoconni, 2021-11-10:
Yeah, I noticed a surge in how much JSBSim GitHub project has been starred recently. By the way, there has been another rise last July when you published a post in LinkedIn about JSBSim now having a Simulink interface (this was for release 1.1.8)

## Stargazers over time

[image]

bcoconni, 2021-11-12:
> Oh man ! I forgot to include `TestJSBSim.m` and `ex737cruise.slx` in the Windows installer 🤦

This is now fixed in release `1.1.10`. I checked on my side and could find the files in [`JSBSim-1.1.10-setup.exe`](https://github.com/JSBSim-Team/jsbsim/releases/download/v1.1.10/JSBSim-1.1.10-setup.exe). However I cannot build the S-Function as I do not own a copy of Matlab. Could one of you check that the S-Function can be built from the material included in `JSBSim-1.1.10-setup.exe` ?

Thanks.

scienceasdf, 2021-12-28:
> Hello, I'm new here and I'm trying to install the JSBsim(v1.1.8) to use with matlab/simulink. I followed all those steps described above, they really fixed some errors, but it still not working. [...]

See https://github.com/JSBSim-Team/jsbsim/pull/548. You can follow the instructions to build matlab mex file with MinGW-w64 toolset.

bcoconni, 2022-01-06:
FYI, [the Matlab guys just announced that Matlab actions for GitHub are now available for Windows and MacOS X](https://github.com/matlab-actions/setup-matlab/issues/18#issuecomment-1006990188). I'll therefore update our CI later this week to check that JSBSim S-Function also builds successfully on Windows and MacOS X (currently only checking on Linux).

seanmcleod, 2022-01-07:
Should  be useful in terms of minimizing the number of Matlab build issue requests we get 😉

bcoconni, 2022-01-08:
Well `matlab-actions/setup-matlab@v1.1.0` does not include the header file `mclcppclass.h` yet for MacOSX and Windows. I have submitted the issue to matlab-actions/setup-matlab#18. Hopefully that will be fixed.

bcoconni, 2022-01-08:
> Well `matlab-actions/setup-matlab@v1.1.0` does not include the header file `mclcppclass.h` yet for MacOSX and Windows. I have submitted the issue to [matlab-actions/setup-matlab#18](https://github.com/matlab-actions/setup-matlab/issues/18). Hopefully that will be fixed.

False alarm, the header file `mclcppclass.h` is actually not needed. The S-Function now compiles on the 3 platforms Linux, Windows, MacOSX. I still have a couple of details to fix. We are almost there ! :smile:

bcoconni, 2022-01-09:
MATLAB S-Functions are now built by our CI workflow for the 3 major platforms (Windows, Linux, MacOSX) since the commit db4d967b8e441ce0406b0c5baaad3201294386a9 that I pushed yesterday night.

CMake is now taking care of building the S-Function from the ground up as soon as you set the flag `BUILD_MATLAB_SFUNCTION` to `ON` (i.e. `cmake -DBUILD_MATLAB_SFUNCTION=ON ..` using the command line). May be someone could document how to set this flag to `ON` on Visual Studio and launch the build with CMake from within Visual Studio ? This procedure would be useful for users who build and compile JSBSim themselves using Visual Studio.

For the users who use the Windows installer to install JSBSim, I guess we should also update `matlab/JSBSimSimulinkCompile.m` to compile out of the box for a pristine installation of JSBSim using our Windows installer ? This would mainly involve updating the paths to the library and to the headers, wouldn't it ? From the release 1.1.10 and on, the Windows installer is supposed to provide all needed files so there should no longer have a need to download additional files and move them around prior to having a MEX file. Could someone please confirm this last statement ?

seanmcleod, 2022-01-09:
I'll take a look in terms of Visual Studio, although I won't be able to fully test since I don't have a copy of Matlab. Do you know off-hand if the matlab github action setup can be installed locally so that I could at least test building locally. I still wouldn't be able to actually test the s-function DLL in Matlab/Simulink.

@agodemar I'm pretty sure has a Matla license, so he's probably the best bet in terms of being able to fully test this on Windows with Visual Studio etc.

I am a bit confused about the combination of the CMake build support we now have and `JSBSimSimulinkCompile.m`. Aren't they basically doing the same thing? Both in invoking Matlab's MEX tool to generate a the S-Function .mex/.dll?

I guess similar to how we have support for building JSBSim using CMake, the official mechanism, but also provide a Visual Studio .sln/.vcxproj files?

seanmcleod, 2022-01-09:
> CMake is now taking care of building the S-Function from the ground up as soon as you set the flag `BUILD_MATLAB_SFUNCTION` to `ON` (i.e. `cmake -DBUILD_MATLAB_SFUNCTION=ON ..` using the command line). May be someone could document how to set this flag to `ON` on Visual Studio

To set `BUILD_MATLAB_SFUNCTION` select `Manage Configurations...`

[image]

Documentation on this can be found at - https://docs.microsoft.com/en-us/cpp/build/customize-cmake-settings?view=msvc-170

bcoconni, 2022-01-10:
> I am a bit confused about the combination of the CMake build support we now have and JSBSimSimulinkCompile.m. Aren't they basically doing the same thing? Both invoking Matlab's MEX tool to generate the S-Function .mex/.dll?

I agree they are doing the same thing but, in my opinion, there are not targeting the same audience.

### `CMake` is for C++ programmers
`CMake` takes care of setting automatically the include and library paths as well as the compile flags depending on the user configuration. Especially, if one change the MSVC configuration from `Release` to `Debug` then the compilation will take place in a different folder (`build/Debug` instead of `build/Release` for that particular example) so you would need to manually update `JSBSimSimulinkCompile.m` to reflect that change - something that's easy to overlook. So I'd say that a programmer needs `CMake` to take care of the gory compilation details and focus on the "important" topics.

### `JSBSimSimulinkCompile.m` is for MATLAB users.
In my view (but I may be wrong), a user that downloads our Windows installer to get the S-Function will not be interested in compiling the library and that sort of thing. Instead I guess they expect having a MATLAB script that automagically compiles the S-Function by a simple command line in MATLAB. That's where `JSBSimSimulinkCompile.m` shines especially since the relative paths are imposed by the installer so the library and include paths can be hardcoded in `JSBSimSimulinkCompile.m`.

seanmcleod, 2022-01-11:
My wild guess, could be way off 😉 is that 90+% of Matlab users simply want a pre-built binary of the S-Function when downloading the Windows installer so that they can incorporate it into their larger Simulink setup with no need or desire to compile the S-function code.

> ...to get the S-Function will not be interested in compiling the library and that sort of thing.

My guess is they don't really want to compile either the JSBSim lib or the S-Function. 

So I guess I was wondering whether it makes sense in terms of catering to the ~10% (wild guess) of Matlab users who want to make source code changes and then compile to offer both the CMake option as well as the `JSBSimSimulinkCompile.m` option.

bcoconni, 2022-01-11:
> 

> My wild guess, could be way off 😉 is that 90+% of Matlab users simply want a pre-built binary of the S-Function when downloading the Windows installer so that they can incorporate it into their [...]

I would tend to be in complement agreement with you but @agodemar told me otherwise in discussion #484.
Quoting @agodemar:
> I think that as we begin to support Matlab S-Functions in JSBSim, it's fine having the user to take care of mex file builds.

However if we reconsider this statetement, it's quite easy to release the MEX files that our GitHub actions build.

seanmcleod, 2022-01-11:
> However if we reconsider this statement, it's quite easy to release the MEX files that our GitHub actions build.

Ah, I assumed once you had added the CMake support for building the `mex` file that you were bundling the `mex` file with the installer.

But glancing through that discussion I noticed your comment:

> While I browsed the internet, I saw that distributing compiled `mex` files seems not to be the way to go since you need to get an S-function which has been compiled for the very version of Matlab that you are using, under the same platform and most likely with the same compiler.

Which I guess means that if we were to try and supply pre-built `mex` files then we'd need to ideally supply a whole bunch, e.g.

- jsbsim-matlab-2020a-compiler1.mex
- jsbsim-matlab-2020a-compiler2.mex
- jsbsim-matlab-2021a-compiler1.mex
 - jsbsim-matlab-2021a-compiler1.mex
 - etc.

Which does seem like a whole lot of work 😉 

So if we don't supply pre-built `mex` files then yes maybe the easiest/lowest friction option would be for users to use `JSBSimSimulinkCompile.m` to pick up the relevant version of Matlab and also their configured compiler.

And as you say if a user wants to make say source changes to JSBSim as well before building their `mex` file then they can go-ahead and use the CMake option.

So pretty much ignore everything else I said earlier 😉 

So let's leave both compile options, and maybe generate a Wiki page with build info/instuctions, e.g. how to configure CMake in Visual Studio for building the `mex` file etc. Describing both the CMake option and the `JSBSimSimulinkCompile` option etc.

agodemar, 2022-01-11:
> So let's leave both compile options, and maybe generate a Wiki page with build info/instuctions, e.g. how to configure CMake in Visual Studio for building the mex file etc. Describing both the CMake option and the JSBSimSimulinkCompile option etc.

Agreed

bcoconni, 2022-01-15:
> Ah, I assumed once you had added the CMake support for building the `mex` file that you were bundling the `mex` file with the installer.

Well, the installer is not managed by CMake but by our CI workflow:
https://github.com/JSBSim-Team/jsbsim/blob/d3536e95a9d4e68dcc9325696e6fc024558d574c/.github/workflows/cpp-python-build.yml#L385-L419

and the list of files that should be installed by the installer is managed in the file [`JSBSim.iss.in`](https://github.com/JSBSim-Team/jsbsim/blob/master/JSBSim.iss.in).
> Which I guess means that if we were to try and supply pre-built `mex` files then we'd need to ideally supply a whole bunch, e.g.
> 
> * jsbsim-matlab-2020a-compiler1.mex
> 
> * jsbsim-matlab-2020a-compiler2.mex
> 
> * jsbsim-matlab-2021a-compiler1.mex
> 
> * jsbsim-matlab-2021a-compiler1.mex
> 
> * etc.
> 
> 
> Which does seem like a whole lot of work :wink:

As long as the dull work is done by computers this is not much of an issue :smile:. In my opinion, the real problem would be that we cannot cover every possible combinations of MATLAB versions and compiler versions: at the moment, [GitHub supports `Visual Studio Enterprise 2019` and `Visual Studio Enterprise 2022` (`Visual Studio Enterprise 2017` is being deprecated)](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners#supported-software) and the [MATLAB setup action supports MATLAB R2021a+](https://github.com/matlab-actions/setup-matlab/issues/18#issuecomment-1006990188). So even if we would release a `mex` file for every combination that GitHub Actions allows, we would still need the `JSBSimSimulinkCompile.m` script for other compiler/MATLAB versions.

> So let's leave both compile options, and maybe generate a Wiki page with build info/instuctions, e.g. how to configure CMake in Visual Studio for building the `mex` file etc. Describing both the CMake option and the `JSBSimSimulinkCompile` option etc.

Yes, I think all the needed material is in the comments of the current issue. All we need is to collect it and make a tutorial out of it. I guess we need to release `v1.1.11` before that, but I'd like to fix #553 first.

bcoconni, 2022-01-18:
> Instead I guess they expect having a MATLAB script that automagically compiles the S-Function using a simple command line in MATLAB. That's where `JSBSimSimulinkCompile.m` shines especially since the relative paths are imposed by the installer so the library and include paths can be hardcoded in `JSBSimSimulinkCompile.m`.

I'd like to clarify the parameters that are used by the `mex` command.
* Should users modify the `-R2017b` to match the MATLAB version that is installed on their computer ?
* Is `COMPFLAGS` mandatory ?

https://github.com/JSBSim-Team/jsbsim/blob/29904e5175e7210c6679cf5ba1ab6ca74893d069/matlab/JSBSimSimulinkCompile.m#L5

bcoconni, 2022-01-23:
> I'd like to clarify the parameters that are used by the `mex` command.

Well, so I [RTFM for `mex`](https://fr.mathworks.com/help/matlab/ref/mex.html) :wink:

> * Should users modify the parameter `-R2017b` to match the MATLAB version that is installed on their computer ?

According to the [`mex` documentation](https://fr.mathworks.com/help/matlab/ref/mex.html#mw_5d9f5fd8-76a3-4406-817f-d4bba493eeb8), this parameter selects the API that is used for matrices. There are basically two options at the moment `-R2017b` (which is the default) and `-R2018a`. So we'll stick to the default for now i.e. `-R2017b`.

> * Is `COMPFLAGS` mandatory ?

According to the [`mex` documentation](https://fr.mathworks.com/help/matlab/ref/mex.html#mw_663100fe-c5a5-4a18-9171-7ab49fe7d509), no. So we'll pass on this argument for now. Note that the keyword used for compiler options depends on the platform :frowning_face:

bcoconni, 2022-01-24:
> So if we don't supply pre-built `mex` files then yes maybe the easiest/lowest friction option would be for users to use `JSBSimSimulinkCompile.m` to pick up the relevant version of Matlab and also their configured compiler.

This is now implemented by commit 2c12a7b7f3a1f4fddb7df65a7474d0098a1319f4.

bcoconni, 2022-02-13:
Following the release of JSBSim `v1.1.11`, all the bugs reported in this issue have been fixed. This issue is closed. Please open a new issue if some further issues are encountered.

closed 2022-02-13 by bcoconni

lemondada777, 2022-03-01:
> 您好 JSBSimmers！
> 
> 我已经设法在 Windows 上下载并安装了[JSBSim 版本 v1.1.8](https://github.com/JSBSim-Team/jsbsim/releases/tag/v1.1.8)，其中包括新发布的 Matlab 接口。 **然而，除非您添加一些标题并在 Matlab 端做一些额外的工作，否则该界面在 Windows 上无法直接使用。** 您可以采取以下措施使其发挥作用：
> 
> 1. **下载安装程序并在以下本地目录树中设置 JSBSim：**
> 
> ```
> <JSBSim root>/
>    aeromatic++/
>    aircraft/
>    data_output/
>    engine/
>    include/
>    lib/
>    matlab/
>    scripts/
>    systems/
>    JSBSim.exe
>    msvcp140.dll
> ```
> 
> 您系统上的确切位置`<JSBSim root>/`取决于您选择安装 JSBSim 的位置。
> 
> 2. **当前发行版中缺少一些头文件**，即所有`*.hxx`与`simgear`. 您必须将所有丢失的头文件，包括它们的目录结构，[从源代码树](https://github.com/JSBSim-Team/jsbsim/archive/refs/tags/v1.1.8.zip)复制到以下本地安装树：
> 
> ```
> <JSBSim root>/
>    ...
>    include/
>       simgear/
>          io/
>             iostreams/
>          magvar/
>          misc/
>          props/
>          structure/
>          xml/
> ```
> 
> **@bcoconni这是在 Windows 上打包 JSBSim 设置可执行文件时应在下一版本中修复的重要步骤。**
> 
> 3. **现在，让我们看看您在 Matlab 方面需要做的额外工作。**
> 
> 以下步骤针对我已安装在我的系统上的_Matlab v2020b进行了测试。_
> 
> 要在 Windows 上编译该文件`<JSBSim root>/matlab/JSBSim_SFunction.cpp`并获取 MEX 文件`JSBSim_SFunction.mexw64`，您需要一个可用的 C++ 编译器。在可用选项中，我建议您选择[Visual Studio 2019 社区版](https://visualstudio.microsoft.com/)。
> 
> 一旦你有一个工作的 C++ 开发环境，你需要从 Matlab 命令行告诉 MEX 实用程序什么是默认的 C++ 编译器。您可以使用以下命令执行此操作：
> 
> `[Matlab prompt]>> mex -setup CPP`
> 
> MEX 将检测您的 C++ 编译器并将配置为构建您的`*.mexw64`文件。
> 
> 4. **调整 Matlab 脚本`<JSBSim root>/matlab/JSBSimSimulinkCompile.m`以构建您自己的 MEX 文件。**
> 
> 首先，将脚本复制`<JSBSim root>/matlab/JSBSimSimulinkCompile.m`到您自己`<JSBSim root>/myJSBSimSimulinkCompile.m` **的 JSBSim 安装目录根目录下的版本中**。这是有道理的，因为我们想要一个`<JSBSim root>/JSBSim.exe`位于.`aircraft``engine``scripts``systems`
> 
> 这是脚本的**`<JSBSim root>/myJSBSimSimulinkCompile.m`**样子：
> 
> ```
> disp('Compiling S-function from JSBSim...');
> mex -v -R2017b COMPFLAGS='$COMPFLAGS' ./matlab/JSBSim_SFunction.cpp ./matlab/JSBSimInterface.cpp -I".\include" -L".\lib" -lJSBSim  wsock32.lib ws2_32.lib
> disp('Finished.')
> ```
> 
> 如果您将当前的_Matlab 工作目录_指向`<JSBSim root>/`并从 Matlab 编辑器或命令行运行上述 M 文件：
> 
> `[Matlab prompt]>> myJSBSimSimulinkCompile`
> 
> 你会得到 MEX 文件`<JSBSim root>/JSBSim_SFunction.mexw64`。
[... 522 lines cut ...]
> 
> 
>   Mass Properties Report (English units: lbf, in, slug-ft^2)
>                                       Weight    CG-X    CG-Y    CG-Z         Ixx         Iyy         Izz         Ixy         Ixz         Iyz
>     Base Vehicle                     83000.0   639.0     0.0   -40.0    562000.0   1473000.0   1894000.0        -0.0      8000.0        -0.0
> 0   Fuel                               10000     520     -80     -18           0           0           0
> 1   Fuel                               10000     520      80     -18           0           0           0
> 2   Fuel                                4000     480       0     -18           0           0           0
>                                                                                                                                             
>     Total:                          107000.0   610.8     0.0   -35.1    591572.3   1539552.7   1986235.4         0.0     19109.1         0.0
> 
> End of vehicle configuration loading.
> -------------------------------------------------------------------------------
> 
> Simulation completed.
> Remember to reset the program by typing clearSF in the matlab command window! 
> JSBSim S-Function Reset
> ```
> 
> 在 Simulink 中，您将获得以下输出：
> 
> [image]
> 
> 因此，您只需尝试一下并根据您的需要调整此工作流程。
Can I design its control law in matlab to directly input it for control?

bcoconni, 2022-05-07:
FYI I have copied the content of this issue in a README file located in `matlab/README.md`. That way users should be able to locate the instructions accumulated above in a single document.

Of course, comments are welcome about the content of `matlab/README.md`.

jevta23, 2023-06-24:
Dear Professor De Marco,

I have installed JSBSim (version 1.1.13) last week, but I have a problem with outputs from S-function in Simulink. I cannot figure out why model doesn't accept the throttle command from the joystick. I have checked functionality of the joystick, and it's working, but when I want to drive the plane (c172p) in the simulation environment, there is a problem with generating thrust. Furthermore, I have noticed that elevation and aileron commands work only when I set the initial velocity (w, vc, or vt) in the initialization xml file. Do you have any suggestions how to fix this problem, and how can I set up the files before running the Simulink model? Thanks in advance!

Kind regards,
Đorđe Jevtić
Ph.D. Student & Junior Research Assistant
University of Belgrade, Faculty of Mechanical Engineering

agodemar, 2023-06-24:
@jevta23 you should open a new issue and give all possible details about your problem.

jevta23, 2023-06-25:
Dear Professor,

Thanks for your quick reply! I have opened a new issue and tried to explain my problem.

Kind regards,
Đorđe