JSBSim-Team/jsbsim #494: FGTurbine don't start windmilling until powered up at least once

Octal450, 2021-08-13:
**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**

The engine stays at 0% N1/N2 and does not windmill unless it is started at least one time

**What is the current behavior?**

To reproduce start a simulation with FGTurbine but do not start one of the engines (example, in a twinjet, start only 1 engine), then takeoff and accelerate. Notice the engine stays at 0% N1 and N2. Then if you engage the starter and then shut down to get slightly rotation, then it seeks the windmill value properly

**What is the expected behavior?**

Engine should always windmill when off

**What is the motivation / use case for changing the behavior?**
So that he will work realistic

**Please tell us about your environment:**

seanmcleod, 2021-08-15:
Taking an initial look at the code in FGTurbine, it looks like windmilling is only implemented as part of the engine shutting down, i.e. 

https://github.com/JSBSim-Team/jsbsim/blob/08d2f857b407b5501fe8bba5c245c1051c606c60/src/models/propulsion/FGTurbine.cpp#L172-L183

That's the only reference to windmilling in FGTurbine excluding the lookup to see if windmilling is disabled.

https://github.com/JSBSim-Team/jsbsim/blob/08d2f857b407b5501fe8bba5c245c1051c606c60/src/models/propulsion/FGTurbine.cpp#L477-L478

bcoconni, 2021-09-17:
Yes, but the variable `phase` is initialized to `tpOff`:
https://github.com/JSBSim-Team/jsbsim/blob/08d2f857b407b5501fe8bba5c245c1051c606c60/src/models/propulsion/FGTurbine.cpp#L98
so `FGTurbine::Off` should be called even if the engine has not yet been started:
https://github.com/JSBSim-Team/jsbsim/blob/08d2f857b407b5501fe8bba5c245c1051c606c60/src/models/propulsion/FGTurbine.cpp#L154-L162
However `phase` is not initialized in the constructor so that might be the cause of the problem.

@Octal450, does the issue vanish if you initialize `phase = tpOff` in the `FGTurbine` constructor ?

Octal450, 2021-09-17:
I'll try my best to test this tonight or tomorrow. A little low on time. 

Thanks Bertrand!
Josh

Octal450, 2021-09-18:
Hi @bcoconni, no difference. 
Looks like the switch default is Off() anyways

Added phase = tpOff; to the constructor below disableWindmill in the latest on FlightGear next and recompiled and rebuilt.

Kind Regards,
Josh

seanmcleod, 2021-09-19:
For testing I modified `scripts\b737_runway_new.xml` and added the following so that only one engine is running, then set a breakpoint in `FGTurbine::Off()` to see what was happening as the aircraft accelerated.

```xml
    <event name="Set active engine">
      <description>Only make one of the engines active</description>
      <condition>simulation/sim-time-sec ge 19.0</condition>
      <set name="propulsion/active_engine" value="0"/>
      <notify/>
    </event>
```

So first off `FGTurbine::Off()` is being called, and the windmill section is being executed.

```c++
  if (disableWindmill == false) {
    N1 = Seek(&N1, in.qbar/10.0, N1/2.0, N1/N1_spindown);
    N2 = Seek(&N2, in.qbar/15.0, N2/2.0, N2/N2_spindown);
  }
``` 

However the issue is that the `accel` - `N1/2.0` and the `decel` - `N1/N1_spindown` parameters to `Seek()` are 0 since N1 starts off with a value of 0.

seanmcleod, 2021-09-19:
So do we add a check for N1/N2 being 0 and use something like `N1_spinup` for say `0 >= N1 < 15` and then switch to `N1/2.0` for the acceleration?

seanmcleod, 2021-09-27:
@Octal450 @bcoconni any suggestions in terms of a realistic spin up acceleration figure to use when N1 = 0 and the engine is off? I guess `N1_spinup` may be too large since I assume that's based on the APU starting up the engine?

Octal450, 2021-09-28:
Sorry I've been away from home for my medical reasons.

Can you be more specific, is the value for starting the engine? I've already found the values are pretty wrong by default for that
If you mean for windmilling the speed is already acceptable when after start & shutdown then moving to get it to windmill. So I really think the biggest problem right now is getting it to do that (the root of this issue)

I will play with the actual values regarding the other issue (windmill value) and let you know on that when I get the chance.

Kind Regards,
Josh

seanmcleod, 2021-09-28:
The particular issue at the moment is that when `N1 = 0`, applies to N2 as well, and the engine is in the off state then the third argument to `Seek` is the acceleration value to be used in order to seek from it's current value to the target value, in this case `in.qbar/10.0`. 

```c++
  if (disableWindmill == false) {
    N1 = Seek(&N1, in.qbar/10.0, N1/2.0, N1/N1_spindown);
    N2 = Seek(&N2, in.qbar/15.0, N2/2.0, N2/N2_spindown);
  }
```

So since N1 is 0, the acceleration value `N1/2.0` is also 0, so N1 never advances towards `in.qbar/10.0`.

bcoconni, 2021-10-11:
Alternatively we could add a shift to `N1/2.0` and `N2/2.0` such as `N1/2.0+0.1` so that the resulting value is never 0.0

Octal450, 2021-10-12:
Ah I understand, thanks.
Josh

bcoconni, 2021-10-12:
Yes and the constant `0.1` I mentioned is completely arbitrary. I guess this needs some testing before selecting a value.

seanmcleod, 2021-10-17:
I'd suggest that we add the 0.1 increment to the acceleration value, push a commit for that and close this issue, and leave https://github.com/JSBSim-Team/jsbsim/issues/195 open until we get some data to decide on what the target value should be for N1/N2 under windmilling conditions.

Octal450, 2021-10-17:
I agree.

I will test this and see if I can figure it out shortly on my build FGFS.

Kind Regards,
Josh

bcoconni, 2021-11-20:
FYI the fix is now available on the FlightGear `next` branch.

Octal450, 2021-11-20:
Thank you kindly!

Josh