JSBSim-Team/jsbsim #499: Bug report: Build fails on Intel Tigerlake CPU

auturgy, 2021-09-27:
**I'm submitting a ...**
  - [x ] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
cmake fail due to arch check:
cc1plus: error: bad value (‘tigerlake’) for ‘-march=’ switch
cc1plus: note: valid arguments to ‘-march=’ switch are: nocona core2 nehalem corei7 westmere sandybridge corei7-avx ivybridge core-avx-i haswell core-avx2 broadwell skylake skylake-avx512 cannonlake icelake-client icelake-server cascadelake bonnell atom silvermont slm goldmont goldmont-plus tremont knl knm x86-64 eden-x2 nano nano-1000 nano-2000 nano-3000 nano-x2 eden-x4 nano-x4 k8 k8-sse3 opteron opteron-sse3 athlon64 athlon64-sse3 athlon-fx amdfam10 barcelona bdver1 bdver2 bdver3 bdver4 znver1 znver2 btver1 btver2 native
cc1plus: error: bad value (‘tigerlake’) for ‘-mtune=’ switch
cc1plus: note: valid arguments to ‘-mtune=’ switch are: nocona core2 nehalem corei7 westmere sandybridge corei7-avx ivybridge core-avx-i haswell core-avx2 broadwell skylake skylake-avx512 cannonlake icelake-client icelake-server cascadelake bonnell atom silvermont slm goldmont goldmont-plus tremont knl knm intel x86-64 eden-x2 nano nano-1000 nano-2000 nano-3000 nano-x2 eden-x4 nano-x4 k8 k8-sse3 opteron opteron-sse3 athlon64 athlon64-sse3 athlon-fx amdfam10 barcelona bdver1 bdver2 bdver3 bdver4 znver1 znver2 btver1 btver2 generic native
make[2]: *** [src/input_output/CMakeFiles/InputOutput.dir/build.make:89: src/input_output/CMakeFiles/InputOutput.dir/FGScript.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:570: src/input_output/CMakeFiles/InputOutput.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

**What is the current behavior?**
make fails with above error

**What is the expected behavior?**
Successful compile on Tigerlake

**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**
 N/A

**Other information**
As above

seanmcleod, 2021-09-27:
Which compiler and version are you using?

If gcc it looks like they added support for Tigerlake in version 9.3.1 round about April 2021.

https://www.freshports.org/lang/gcc9-devel/

auturgy, 2021-10-11:
Apologies for the delay: I ended up installing the deb package in order to keep moving, and that works without issue.
@bcoconni I didn't pass any args to cmake at all, just:
```cmake ..
make ```

bcoconni, 2021-11-20:
Issuer @auturgy is not responding. Issue closed.

closed 2021-11-20 by bcoconni