JSBSim-Team/jsbsim #531: Latitude > 90 degrees causes fault

jonsberndt, 2021-12-10:
**I'm submitting a ...**
  - [X] bug report
  - [  ] feature request
  - [  ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
This error was observed when a waypoint_distance element was executed. This construct that caused the error was this:

```
<waypoint_distance name="system/distance-to-dugway" unit="FT">
	  <target_latitude unit="RAD">  location/site_long_rad </target_latitude>
	  <target_longitude unit="RAD"> location/site_lat_rad </target_longitude>
	  <source_latitude unit="RAD">  position/lat-geod-rad </source_latitude>
	  <source_longitude unit="RAD"> position/long-gc-rad </source_longitude>
</waypoint_distance>
```	

This code resulted in a floating point exception being executed within this call in the FGWaypoint::Run() function:

```
    double wp_distance = source.GetDistanceTo(target_longitude_rad,
                                              target_latitude_rad);
```
Looking more closely at the code and at the XML I discovered that I mistakenly switched the longitude and latitude values in the waypoint_distance element:
	  <**target_latitude** unit="RAD">  location/**site_long_rad** </target_latitude>
	  <**target_longitude** unit="RAD"> location/**site_lat_rad** </target_longitude>

While this was user error, it seems that it would be good to do some bounds checking to let the user know if they have entered a latitude > 90 degrees, and also if a latitude > 90 degrees is supplied to the GetDistanceTo() or GetHeadingTo() functions in FGWaypoint. There may be additional locations in the code from which a call to these GeographicLib functions take place, and it would be good to make sure that bad information is not being passed in through these functions. A check on the latitude value that is passed into the GetHeadingTo() and GetDistanceTo() functions might be good to have.

**Please tell us about your environment:**

 - Windows 10 / Cygwin
 - JSBSim version - Latest from the repo
 - C++

jonsberndt, 2021-12-10:
I can implement a fix after discussion if needed, but it may be a few days at least before I can get to it. Which is probably OK since this isn't a critical error, actually.

bcoconni, 2021-12-10:
> I can implement a fix after discussion if needed,

That would be great !
>but it may be a few days at least before I can get to it. Which is probably OK since this isn't a critical error, actually.

Sure. No rush.

seanmcleod, 2021-12-12:
Why only latitude in terms of bounds checking? Sure it may be more common if people mix up longitude with latitude like you did, but what about bounds checking longitude to prevent `abs(longitude) > 180` as well? There could be other bugs/issues in people's code which ends up with invalid longitude values.

jonsberndt, 2021-12-12:
> > I can implement a fix after discussion if needed,
> 
> That would be great !
> 
> > but it may be a few days at least before I can get to it. Which is probably OK since this isn't a critical error, actually.
> 
> Sure. No rush.

Bug fixed. Code change is to FGWaypoint.cpp only. It's in branch jsb-531.

What's the procedure that is generally used in addressing a bug?

jonsberndt, 2021-12-12:
> Why only latitude in terms of bounds checking? Sure it may be more common if people mix up longitude with latitude like you did, but what about bounds checking longitude to prevent `abs(longitude) > [...]

Bounds checking longitude might not be a bad idea. The difference is that latitude>90 results in a segfault. Longitude>180 is valid, though maybe not intended.

seanmcleod, 2021-12-12:
> What's the procedure that is generally used in addressing a bug?

To generate a pull request from your fork. 

Ah, I see `jsb-531` is a branch in the main JSBSim repo as opposed to a branch in your own fork.

jonsberndt, 2021-12-14:
> > What's the procedure that is generally used in addressing a bug?
> 
> To generate a pull request from your fork.
> 
> Ah, I see `jsb-531` is a branch in the main JSBSim repo as opposed to a branch in your own fork.

Yes, that should still work, no?

bcoconni, 2021-12-14:
> > > What's the procedure that is generally used in addressing a bug?
> > 
> > 
> > To generate a pull request from your fork.
> > Ah, I see `jsb-531` is a branch in the main JSBSim repo as opposed to a branch in your own fork.
> 
> Yes, that should still work, no?

Most likely, yes. To generate a pull request, follow [GitHub docs about creating a pull request](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request).

seanmcleod, 2021-12-15:
@jonsberndt fyi a pull request will trigger a bunch of github actions like generating a build of JSBSim for all the supported platforms, so that we can confirm that it builds successfully for all platforms and also running the test cases on each supported platform, e.g. take a look at - https://github.com/JSBSim-Team/jsbsim/runs/3920398341?check_suite_focus=true

Lastly it also sets up a github review allowing people to review the commit generally and on a per-line basis etc.

bcoconni, 2021-12-18:
FYI PR #536 is now merged.

closed 2022-01-07 by bcoconni