JSBSim-Team/jsbsim #583: Aeromatic: Turboprop engines are generated as turbine engines.

TheFGFSEagle, 2022-02-11:
Hello,
I have stumbled over a bug in Aeromatic:

**Description**
I was trying to generate a turboprop engine. After fighting with the engine (it wouldn't start in FlightGear) I decided to compare the engine of another turboprop aircraft model with the file I got, and only at that point noticed that I had got a <turbine_engine> instead of a <turboprop_engine> !! 

**What is the current behavior?**
* Open the Aeromatic website (jsbsim.sourceforge.net](http://jsbsim.sourceforge.net/aeromatic2.html)
* In the engine section, select turboprop as engine type
* Click "Generate"
* Aeromatic produces an XML file containing a <turbine_engine>.

**What is the expected behavior?**
Aeromatic should produce a <turboprop_engine> along with the appropriate tables.

dpculp, 2022-02-11:
Message ID: ***@***.***>
> The original AeroMatic was created before a turboprop model existed in 
> JSBSim, so a turbine model stood in for a turboprop.  Even after a 
> turboprop model was added I continued to use the turbine substitute 
> because the turboprop model was based on the Walter M601 engine and 
> has not been generalized for all types of turboprops.  Generalizing 
> the turboprop model will not be easy because of the wide variety of 
> turboprop configurations.  I once looked into the job of generalizaing 
> the model and decided it would be too messy.

It looks like aeromatic2 carries over the same scheme used in original 
AeroMatic.

dpculp, 2022-02-11:
Actually they are very different tools.  The original, web-based tool is based on a different philosophy than the AeromatiC++ tool.  I can have a look at the web-based version.  I'm surprised it's still working, actually.  It's that old.  

I'll see what I can do.

dpculp, 2022-02-13:
I've edited aeromatic2 (hosted at my server here:  [aeromatic2](http://www.daveshangar.org/aeromatic/aeromatic2.html)) to output a turboprop configuration file.  I'm trying to test it now, but I can't get it started.  Are you able to successfully start the turboprop engine?  Also, do you have a model at FlightGear that uses the turboprop that I can test with?

TheFGFSEagle, 2022-02-13:
Yes, you can use the Cessna 208B from FGAddon for example. It has an autostart function.

If the aircraft has no custom systems (which might add more things which have to be done) all you need to start a turboprop engine (in FlightGear) is
* set /controls/electric/engines/generator to true
* set /controls/engines/engine/cutoff to false
* set /controls/engine/engine/starter to true
The rest should be automatic.

TheFGFSEagle, 2022-02-13:
I didn't try it, but from a quick look the generated turboprop engine files should work.

dpculp, 2022-02-13:
OK, thanks!  Yes, it appears to work OK.

closed 2022-02-25 by TheFGFSEagle