JSBSim-Team/jsbsim #593: Can I design its control law in matlab to directly input it for control?

lemondada777, 2022-03-01:


BadboyIntheLoop, 2022-03-04:
Hello friend!!! Now that i'm discovering about JSBsim and embedding it into Matlab. I've already run jsbsim in matlab and there somethings are stucked like: 
- How actuator and FCS run in it? Where the file that Control Ouputs are computed from Control Commands? Because i see that they are different when i dont declare actuator. After solving that, i wanna take actuators into matlab for conveniently simulinking
- Setup control for elevon aircraft 
If we can make Control Outputs and Control Commands as the same, we can design autopilot in matlab. If find out something, i'll post cmt on this issue

lemondada777, 2022-03-04:
Thank you for noticing my problems.I have the same confusion as you. Now I find that some input commands of the s-function, such as thrust, don't seem to have any effect on the airplane model when not run as a script.

seanmcleod, 2022-03-04:
First off it all depends on what JSBSim aircraft model you're using. Neither of you have specified what you're using.

In general you can specify the FCS commands and actuators in JSBSim if you want, or you can put all that logic and actuator modelling in Matlab/Simulink.

For example take a look at say the A-4 JSBSim model. There is no actuator specified and the FCS is a trivial mapping of `fcs/elevator-cmd-norm` to `fcs/elevator-pos-rad`. So if you wanted to do all FCS logic and the actuator modelling in the Matlab/Simulink for this model you could since you won't have a 2nd FCS in JSBSim adding in logic and JSBSim won't also be modelling an actuator etc.

https://github.com/JSBSim-Team/jsbsim/blob/6461c864ca2570f8e0c63e6522bb911473d86e71/aircraft/A4/A4.xml#L198-L219

BadboyIntheLoop, 2022-03-04:
I dont seem like you, i'vs just test with aircraft A4, If i set DeCmd = 0.5, the control output DePos = 0.15. I dont know why, if you know how please show me where difference?

seanmcleod, 2022-03-04:
How are you reading `DePos`?

0.15 is exactly 0.5 of the max elevator position in radians as shown in the snippet I shared.

BadboyIntheLoop, 2022-03-04:
Yes, thank you for your support, I saw that, it have a scale

seanmcleod, 2022-03-04:
Correct, the A-4 FCS is simply mapping a control input in the range `[-1, +1]` to an elevator position in radians in the range `[-0.35, +0.3]`, i.e. no fancy FCS logic and no actuators modelled.

The elevator position `fcs/elevator-pos-rad` is then used by the aerodynamics section to calculate forces and moments, e.g.

https://github.com/JSBSim-Team/jsbsim/blob/6461c864ca2570f8e0c63e6522bb911473d86e71/aircraft/A4/A4.xml#L424-L432

If you want your Matlab/Simulink model to perform the FCS logic and actuator modelling and effectively just output a control surface position then you could modify the FCS in the A-4 FCS as shown above to not perform a scaling from `[-1, +1]` to `[-0.35, +0.3]`, rather just a 1:1 mapping/scaling.

lemondada777, 2022-03-06:
Sorry I still have some problems:

1. Why when I run the script, I can do a simple trim to make the plane fly at a constant height and constant speed, but when I run in standalone t mode with the same throttle and elevator inputs the plane keeps dropping high?

2. How to trim the aircraft when running in standalon mode?I want to build a PID controller in Simuink so that the plane can fly at a fixed altitude

https://github.com/JSBSim-Team/jsbsim/blob/6461c864ca2570f8e0c63e6522bb911473d86e71/aircraft/A4/A4.xml#L424-L432

seanmcleod, 2022-03-06:
> when I run the script

Which script are you running?

> but when I run in standalone mode with the same throttle and elevator inputs the plane keeps dropping high?

If you're using the control position results from an earlier trim solution are you sure you're duplicating the other state that the trim solution is using and it created. For example is the airspeed identical, is altitude identical, is AoA identical?

Matlab also has built-in tools to calculate trim solutions.

lemondada777, 2022-03-06:
It is 737_cruise.And I don't konw what other states need to be set and how other than the throttle and elevator and initial state setting files.

BadboyIntheLoop, 2022-03-07:
> It is 737_cruise. And I don't know what other states need to be set and how other than the throttle and elevator and initial state setting files.

I think you should read file JSBSimInterface.h/cpp if you run it on matlab! So you can get more variables or change something you want.