JSBSim-Team/jsbsim #601: Cannot Get FGWind* pointer in JSBSim 1.1.11

BadboyIntheLoop, 2022-03-14:
Hello the Pro, 
I wanna change the default codes that interface with matlab (JSBSimInterface and JSBSim_SFunction) in order to get/set more value! For get/set wind parameter, I declared like as following in JSBSimInterface.h:
`FGWinds *winds` 
And using it in source file JSBSimInterface.cpp:
`fdmExec = new FGFDMExec;`
`winds = fdmExec->GetWinds()`
I'm using MSVC C++ 2019 compiler but during compiling I get an error:
  > Error using mex
  > JSBSimInterface.cpp
  > C:\JSBSim\matlab\JSBSimInterface.cpp(763): error C2027: use of undefined type 'JSBSim::FGWinds'
  > .\include\FGFDMExec.h(63): note: see declaration of 'JSBSim::FGWinds'
I dont why. Clearly, I declared it fully though! Pls help me at this situation. So many thanks!!!

seanmcleod, 2022-03-14:
But have you included the header file `#include "models/atmosphere/FGWinds.h"` anywhere?

BadboyIntheLoop, 2022-03-14:
> But have you included the header file `#include "models/atmosphere/FGWinds.h"` anywhere?
Thank you so much, It 's fixed! 
<3