JSBSim-Team/jsbsim #616: FGFDMExec::FGFDMExec is not exported to static library

Scavanger, 2022-04-28:
**I'm submitting a ...**
 bug report

JSBSim::FGFDMExec::FGFDMExec  is not exported to static library, works with 1.1.11

Library compiles without error, but when I'm using it, i get an linker error: `unresolved external symbol ""public: __cdecl JSBSim::FGFDMExec::FGFDMExec(class JSBSim::FGPropertyManager *,unsigned int *)`

I checked the library with dumpbin:  No trace of the constructor

Works like a charm with 1.1.11

 - OS: Windows 11, Visual Studio 2019 (v142)
 - JSBSim version: Latest master
 - Language: C++,

bcoconni, 2022-05-07:
Let me rephrase your issue:
* When you are using the static library from the release 1.1.11 installer [JSBSim-1.1.11-setup.exe](https://github.com/JSBSim-Team/jsbsim/releases/download/v1.1.11/JSBSim-1.1.11-setup.exe), the linker successfully finds `JSBSim::FGFDMExec::FGFDMExec`.
* When you are compiling the static library yourself from the `master` branch, the linker later complains about not being able to find `JSBSim::FGFDMExec::FGFDMExec`.

Am I correct ? Also could you tell us how you are compiling JSBSim ? Are you using the default `*.sln` or `*.vcxproj` files or are you using CMake from within Visual Studio ?

Scavanger, 2022-05-10:
I'm using cmake.

It's compiles correctly whe  I'm using the code from latest release version but I get the error with code from current master.

bcoconni, 2022-05-27:
Might be linked to the parameters you are passing to CMake ?

After a compilation with CMake on my Windows PC, I'm getting some output from the following command (executed from the "Developer PowerShell for VS2019"):
```pwsh
dumpbin.exe /symbols .\build\Debug\src\Debug\JSBSim.lib | select-string -pattern "FGFDMExec::FGFDMExec"
```
Below is an extract of the output I'm getting. It clearly mentions `FGFDMExec::FGFDMExec`:
```text
210C 00000127 SECT5  notype ()    Static       |
?dtor$7@?0???0FGFDMExec@JSBSim@@QEAA@PEAVFGPropertyManager@1@V?$shared_ptr@I@std@@@Z@4HA (int `public: __cdecl
JSBSim::FGFDMExec::FGFDMExec(class FGFDMExec::FGPropertyManager *,class std::shared_ptr<unsigned int>)'::`1'::dtor$7)
2113 00000154 SECT5  notype ()    Static       |
?dtor$8@?0???0FGFDMExec@JSBSim@@QEAA@PEAVFGPropertyManager@1@V?$shared_ptr@I@std@@@Z@4HA (int `public: __cdecl
JSBSim::FGFDMExec::FGFDMExec(class FGFDMExec::FGPropertyManager *,class std::shared_ptr<unsigned int>)'::`1'::dtor$8)
211A 00000181 SECT5  notype ()    Static       |
?dtor$9@?0???0FGFDMExec@JSBSim@@QEAA@PEAVFGPropertyManager@1@V?$shared_ptr@I@std@@@Z@4HA (int `public: __cdecl
JSBSim::FGFDMExec::FGFDMExec(class FGFDMExec::FGPropertyManager *,class std::shared_ptr<unsigned int>)'::`1'::dtor$9)
```
I guess that in your case the symbols are stripped from the library (which is weird because what is the point of a library with no symbols ? 😄). Could you please share the parameters that you are passing to CMake ?

seanmcleod, 2022-05-27:
Just wondering if there is some issue/mismatch with the 2nd argument to the `FGFDMExec` ctor?

> `JSBSim::FGFDMExec::FGFDMExec(class JSBSim::FGPropertyManager *,unsigned int *)`

versus

`FGFDMExec::FGFDMExec(class FGFDMExec::FGPropertyManager *,class std::shared_ptr<unsigned int>)`

Although @bcoconni you did mention you'd only pasted an extract.

Having a quick glance on master, I see there is only a single ctor.

`FGFDMExec(FGPropertyManager* root = nullptr, std::shared_ptr<unsigned int> fdmctr = nullptr);`

bcoconni, 2022-05-27:
> Just wondering if there is some issue/mismatch with the 2nd argument to the `FGFDMExec` ctor?
> 
> > `JSBSim::FGFDMExec::FGFDMExec(class JSBSim::FGPropertyManager *,unsigned int *)`
> 
> versus
> 
> `FGFDMExec::FGFDMExec(class FGFDMExec::FGPropertyManager *,class std::shared_ptr<unsigned int>)`

Yes, the constructor signature is different between the `master` branch (which I used to test the occurrence of the `FGFDMExec::FGFDMExec` symbol) and releases `v1.1.x`:
* In the `master` branch, the constructor signature is `FGFDMExec::FGFDMExec(class FGFDMExec::FGPropertyManager *, class std::shared_ptr<unsigned int>)`
* In the `release/v1.1` branch, the constructor signature is `FGFDMExec::FGFDMExec(class FGFDMExec::FGPropertyManager *, unsigned int*)`.

bcoconni, 2022-06-19:
The problem reported in this issue could not be reproduced and the issuer is no longer responding. The issue is closed.

closed 2022-06-19 by bcoconni