JSBSim-Team/jsbsim #625: How to modify unreal engine aircraft ？

kalading2, 2022-05-16:
Jsbsimflightdynamicsmodel plug-in uses 787-8 model. How can I modify it to other types of aircraft models (such as c172p model) and truly simulate the flight attitude ?
Is it implemented by modifying XML ？

kalading2, 2022-05-16:
[image]

seanmcleod, 2022-05-16:
> and truly simulate the flight attitude

What do you mean by this? What isn't 'true' in terms of the example with the 787?

kalading2, 2022-05-16:
The case he gave is very good. I want to change to another flight model, such as C172

kalading2, 2022-05-16:
@seanmcleod,I want to know how to change it.

seanmcleod, 2022-05-16:
I understand you want to change the flight model. The name of the aircraft is passed in here which is then passed to JSBSim to load. Not 100% certain where the name comes from, presumably some UE property/setting. 

https://github.com/JSBSim-Team/jsbsim/blob/27bf560a3ffe10a4fb146803a5d931a9e1f3c492/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp#L81-L89

But I still don't understand what you meant by:

> and truly simulate the flight attitude

kalading2, 2022-05-16:
@seanmcleod The XML parameters of the flight dynamics model of each aircraft are different. Can this plug-in be as good as in the case .

seanmcleod, 2022-05-16:
Are you familiar with how JSBSim works in terms of how it processes the aircraft model's definition and in particular how it processes FCS inputs etc.? If not take a look at the following - https://jsbsim.sourceforge.net/JSBSimReferenceManual.pdf

kalading2, 2022-05-16:
No, thank you. I have only recently studied this aspect. Let me first look at the documents you sent.

seanmcleod, 2022-05-16:
Yep, once you have a general understanding of how JSBSim works then a lot of things will be clearer in terms of what you may or may not have to modify in the Unreal Plugin when switching aircraft models.

kalading2, 2022-05-16:
okok

AlbanBERGERET-Epic, 2022-05-18:
Yes, the Unreal Engine plugin is made in such a way that when you change the aircraft name property on the jsbsimcomponent, it will load the corresponding jsbsim XML file. 
If you need another aircraft, or edit an existing one, you have to edit the model the jsbsim way in the aircraft database.

kalading2, 2022-05-19:
[image]
@AlbanBERGERET-Epic Why can't I start after the refueling door, and the engine state is inconsistent with the command?

kalading2, 2022-05-19:
I use c172p .

kalading2, 2022-05-19:
[image]
I see it's been logged out here.

seanmcleod, 2022-05-19:
Yes, you'll see the basic demo code has only really been tested with a turbine engine model, so you will need to look for these sorts of blocks of code and ensure that the relevant code is enabled for piston engine types.

kalading2, 2022-05-20:
When will there be plugins for piston engine models ?

seanmcleod, 2022-05-20:
> When will there be plugins for piston engine models ?

When you or someone else submits a pull request that adds support :wink:

Take a look at the reply from @AlbanBERGERET-Epic of Epic - https://github.com/JSBSim-Team/jsbsim/discussions/618#discussioncomment-2725934 particularly in terms of the roadmap.

> It really depends on my work targets. If one day I have to demonstrate something to a helicopter company, I would for sure add this to the plugin. And remember the application was simple. I wrote it was a "rough sketch". No cockpit, no internal systems, no VR... All of that is possible with UE, but a solid and coordinated crew would be needed to bring that to life.

And a follow-up comment.

> No, nothing is planned for the immediate future. I have some travels ahead, so I don't plan to come back on it soon. You're good to go.

AlbanBERGERET-Epic, 2022-05-24:
Hi @kalading2

Unfortunately, I can't commit to that support yet.  But this is very straightforward to code. 
I put some TODOs at the appropriate locations. You just need to add a few lines of codes to convert JSBSim properties TO and FROM FEngineState and FEngineCommands (if needed). 

I won't have the time to write and test this part now, but I can quickly review your files/changes if needed.

AlbanBERGERET-Epic, 2022-06-09:
Hi @kalading2 

I had a look at what you noticed, and actually, even if there was a TODO in the code, it was the Engine State update part that was missing, not the command part. 

It's totally doable to start a c172p. The behavior that you noticed was because the default values of EngineCommands have mixture and magnetos set to 0, even when Start with engine running was set. 
It was not an issue for an airliner, but it is for a piston engine. 

But so far, it's still possible to start the p172 in manual mode : 
- Set both mixture to 1 and magnetos to 3 in the engine commands. 
- Activate the started for a second or so, until the engine state goes to running. 
- Then, you can increase the throttle and go! 
 
I'll soon publish a PR with better default values for commands when Engine is running at start, some explicit enum values for Magnetos state, and an input binding in the reference app to control magnetos...

bcoconni, 2022-06-11:
@kalading2, this bug should now be fixed in the `master` branch, following @AlbanBERGERET-Epic's PR #652. Could you please check and confirm if the plugin now works as expected ?

Thanks.

bcoconni, 2022-06-19:
The problem reported in this issue is now considered fixed and the issue is closed.

closed 2022-06-19 by bcoconni

kalading2, 2022-06-20:
@bcoconni ,I tested it according to your method. It can indeed fly. Your work is very meaningful.

bcoconni, 2022-06-20:
> @bcoconni ,I tested it according to your method. It can indeed fly. Your work is very meaningful.

All the credits should go to @AlbanBERGERET-Epic who did all the hard work !

nmfarras, 2024-02-07:
Hi, i have a question regarding modify unreal engine aircraft. I notice, if there is wrong format (my guess?), when i tried to load the custom aircraft on unreal it will crash. The case where i find it cause Unreal crash is when use following syntax:

```xml
<system file="NWS"/>
```

which load custom system from external file, and this one which i don't know what's wrong

```xml
<channel name = "Pitch-limiter">
	
	<!--
     This is a logics switch controls the state of serviceability of the FBW system
     and restrict engaging if Not ready to operate.
     [TODO]: this still needs to check for power availability !
	--> 
		<switch name = "FBW-Status Logical switch">
			<default value= "1"/>
			<test logic="OR" value="0">
				/su-27/SDU-10/serviceable NE 1
			</test>
			<output>/su-27/SDU-10/active</output>
		</switch>
	<!--
	 ***************************************************
	-->
	
		<summer name="/su-27/SDU-10/pitch-trim-sum">
			<input>/su-27/SDU-10/pitch-limiter-gain</input>
			<input>/su-27/SDU-10/pitch-stab-gain</input>
			<clipto>
				<min>-1</min>
				<max>1</max>
			</clipto>
		</summer>
		
		<switch name = "pitch-trim-switch">
			<default value= "/su-27/SDU-10/pitch-trim-sum"/>
			<test logic="OR" value="0">
				/su-27/SDU-10/pitch-trim-active EQ 0 
				/su-27/SDU-10/active EQ 0
				/su-27/instrumentation/ASI/airspeed-kmh LE 1
			</test>
			<output>/controls/flight/elevator-trim</output>
		</switch>
		
	<!--***********************************************-->
		<!--switch name = "pilot-stick-status">
			<default value= "0"/>
			<test logic="OR" value="0">
				/su-27/SDU-10/pitch-trim-active EQ 0 
				/su-27/SDU-10/active EQ 0
				/su-27/instrumentation/ASI/airspeed-kmh LE 1
				/controls/flight/elevator GE  0.15
				/controls/flight/elevator LE -0.15
			</test>
			<output>/su-27/SDU-10/pilot-input-state</output>
		</switch-->
		
		<!--scheduled_gain name="/su-27/SDU-10/pitch-limiter-gain-raw">
			<input>/orientation/alpha-deg</input>
			<gain>1</gain>
			<table>
		    <independentVar>/su-27/instrumentation/ASI/airspeed-kmh</independentVar>
		    <tableData>
		     300.00   0.01
		     800.00   0.07
		    1200.00   0.10
		    1600.00   0.12
		    </tableData>
			</table>
			<clipto>
				<min>-1</min>
				<max>1</max>
			</clipto>
		</scheduled_gain-->
	</channel>
```

The switch and schedule gain that i commented would cause crash if i uncomment it and call the aircraft model from unreal engine. Any advice or do you know what should i do with formatting and such that might cause crash? Is it due to unhandled error?

bcoconni, 2024-02-11:
@nmfarras You might want to start a new discussion about your particular issue since it is not good practice to resurrect an old thread with an entirely different topic. Especially when the discussion is tagged as closed.

We'd like to have one topic per discussion thread so that people might follow the discussion. Also new comments on an old thread generally remain unnoticed as they are no longer followed by anyone (but me :smile: ?) so the odds of it being answered are slim.

Thanks.