JSBSim-Team/jsbsim #626: JSBSIM plugin crashes with clean UE5.0.1 project

BreakerHUN, 2022-05-22:
- [ ] bug report

**Describe the issue**
The plugin immediately crashes to desktop in UE 5.0.1 when using in clean new project.

**What is the current behavior?**
I download and compile the project, then the referenceapp . It opens perfectly in UE, and can fly around.
Then I copy the whole "Plugin\JSBSimFlightDynamicsModel" to my clean project "MyProject\Plugins"

I can activate the plugin and restart UE.

Then creating a pawn, a static mesh and adding JSBSIMMovement component. The problem starts here.
When I touch the Structural  frame origin propertiy in the component editor, UE immediately crashes to desktop.

or

If I grab the pawn into the world and hit "Play" it immediately crashes.

**What is the expected behavior?**
To integrate into a clean project without crashing.

**What is the motivation / use case for changing the behavior?**
?

**Please tell us about your environment:**

AlbanBERGERET-Epic, 2022-06-09:
Hey guys, 

Sorry about it. Actually, I always edited the Structural Frame origin in the level viewport. There it doesn't crash... 

But indeed, I also reproduced the crash when working in the blueprint editor. This editor is tricky to work in because it doesn't go with the same initialization steps as with the engine viewport. But the good news is that when fixing the issue #637 it fixed this one as a side effect. 

A PR is coming soon with these fixes in one day or so.

bcoconni, 2022-06-11:
@BreakerHUN this bug should now be fixed in the `master` branch, following @AlbanBERGERET-Epic's PR #652. Could you please check and confirm if the plugin now works as expected ?

Thanks.

BreakerHUN, 2022-06-12:
@bcoconni Thanks for the effort! I recompiled it and can confirm that the problem is now gone. It is now working properly.

One thing to mention, that it still crashes if you do not add a "georeferencing" object in the leveleditor, and hit "play" button. But that can be fixed easily. Might worth to handle this case with an error message and avoid running instead of crashing to desktop :) But this issue can be closed.

bcoconni, 2022-06-12:
> @bcoconni Thanks for the effort! I recompiled it and can confirm that the problem is now gone. It is now working properly.

No, thanks @AlbanBERGERET-Epic who did all the hard work ! :+1: 

> One thing to mention, that it still crashes if you do not add a "georeferencing" object in the leveleditor, and hit "play" button. But that can be fixed easily. Might worth to handle this case with an error message and avoid running instead of crashing to desktop :) But this issue can be closed.

I'd suggest you open a new issue to report this problem.

Thanks for the bug report.

closed 2022-06-12 by bcoconni