JSBSim-Team/jsbsim #630: JSBSIM in Unreal Engine: Speedbrake  in Aircraft State not changing

tbuckup, 2022-05-27:
Hi,
I have written a sandbox UE project (UE5.1) with your library based on your example and everything seems to work fine except the speedbrake (did not test spoilers though, but all other aero surfaces are def working).
I have used the F-15 flight model, where a speed brake is defined. However, when I set the command Speed Brake, I can see in the debug that the flag is 1, BUT the value for the speedbrake in the aircraft state (Debug window) does not change.
Rest seems to be working.
I dont think it is a problem with the xml since I tested with others.

seanmcleod, 2022-05-27:
https://github.com/JSBSim-Team/jsbsim/blob/27bf560a3ffe10a4fb146803a5d931a9e1f3c492/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp#L518

https://github.com/JSBSim-Team/jsbsim/blob/27bf560a3ffe10a4fb146803a5d931a9e1f3c492/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp#L587

https://github.com/JSBSim-Team/jsbsim/blob/27bf560a3ffe10a4fb146803a5d931a9e1f3c492/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp#L1026-L1028

But there appear to be two speed brake properties.

https://github.com/JSBSim-Team/jsbsim/blob/2a5985c2e190ca05f5ecd55b480eed40216dedb6/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Public/FDMTypes.h#L310-L322

https://github.com/JSBSim-Team/jsbsim/blob/2a5985c2e190ca05f5ecd55b480eed40216dedb6/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Public/FDMTypes.h#L351

https://github.com/JSBSim-Team/jsbsim/blob/2a5985c2e190ca05f5ecd55b480eed40216dedb6/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Public/FDMTypes.h#L398

Are you seeing both speed brake properties being output?

tbuckup, 2022-05-27:
Problem in XML file of the F-15. F-16 has also a speedbrake and it works (-> degrees and effect are shown).

tbuckup, 2022-05-27:
Did several tests: Speedbrake only worked for F-16 and F-22 (that re-uses the fcs implementation of the F-16). 
Not working for the following so far:  F-15, T-38, T-37 and A4
Definition seems ok:

<channel name="Speedbrake">
            <kinematic name="Speedbrake Control">
                <input>fcs/speedbrake-cmd-norm</input>
                <traverse>
                    <setting>
                        <position>0</position>
                        <time>0</time>
                    </setting>
                    <setting>
                        <position>1</position>
                        <time>2</time>
                    </setting>
                </traverse>
                <output>fcs/speedbrake-pos-norm</output>
            </kinematic>

tbuckup, 2022-05-27:
So the bug is the following:
If the speedbrake is declared with fcs/speed-brake-pos-norm as output it does not work.
It is required to **also** add an output in degrees to the same channel:

`<aerosurface_scale name="Speedbrake normalization">
                <input>fcs/speedbrake-pos-norm</input>
                <domain>
                    <min>  0 </min>
                    <max> 1 </max>
                </domain>
                <range>
                    <min> 0 </min>
                    <max> 45 </max>
                </range>
                <output>fcs/speedbrake-pos-deg</output>
        </aerosurface_scale>`

seanmcleod, 2022-05-27:
Couple of issues with the F-15, haven't checked the T-38, T-37, A-4 etc. they may have similar issues.

First issue is that if you search through the F-15's xml, version in JSBSim repo that is, it isn't used by any of the aerodynamic sections to increase drag, change lift, change pitching moment etc.

Second issue is that this is the sum total of the references to speedbrake, and you'll notice there is no mapping to an angular range. Since the UE plug-in is calling `FCS->GetDsbPos(JSBSim::ofDeg)` as opposed to `::ofNorm` it's failing.

```xml
        <channel name="Speedbrake">
            <kinematic name="Speedbrake Control">
                <input>fcs/speedbrake-cmd-norm</input>
                <traverse>
                    <setting>
                        <position>0</position>
                        <time>0</time>
                    </setting>
                    <setting>
                        <position>1</position>
                        <time>2</time>
                    </setting>
                </traverse>
                <output>fcs/speedbrake-pos-norm</output>
            </kinematic>
        </channel>
```

tbuckup, 2022-05-27:
Yeap....."mapping to angular range" missing for all XMLs.
I have edited the drag and lift based on BMS models and some papers.
All good now and is working as advertised ;-).

seanmcleod, 2022-05-27:
BMS?

Are you using the F-15 model in the JSBSim repo, as opposed to say Richard Harrison's (@Zaretto) model which has a lot more detailed aero data etc.? http://zaretto.com/content/f-15-simulator-flightgear

tbuckup, 2022-05-27:
BMS is Falcon 4.36 BMS.
Thank you for the link.

AlbanBERGERET-Epic, 2022-06-09:
Ok, The UE plugin is not to blame here. 
@bcoconni, can you clear the tag?
@tbuckup, can you confirm this is fine and eventually close the issue?

tbuckup, 2022-06-10:
I can confirm it is working for me. Issue was the XML file..

closed 2022-06-10 by tbuckup