JSBSim-Team/jsbsim #674: F16 aileron and elevator oscilations at CAS 550kn+

Synergy91, 2022-06-30:
Here is my issue:
Using the f-16 model configuration - when going over 550 kn CAS, I experience aileron and elevator oscillations.

My environment:
Windows 10
JSBSim v1.1.12; f-16 config file revision 1.95
JSBSim built as a static lib and linked in an Unreal 5 project.

Not sure if this is a bug or a malpractice on my part, but I can't find any info about this, except for this thread on a JSBsim mailing list:
https://sourceforge.net/p/jsbsim/mailman/jsbsim-devel/thread/1393953.48.1476232800676.JavaMail.bclay%40Mariah/
The direction I'm exploring in is the definitions of the FCS, as per the discussion in the thread linked above, but just can't figure out the issue.

Omer S.

gallonmate, 2022-07-01:
From the sounds of the discussion, the current version 1.95 is the "fix" that Erik attempted, which likely improved things but didn't solve the oscillations you are seeing. 

First, to isolate any UE plugin issues, I would verify your game framerate is locked and smooth during flight. I added a small update to the plugin the other week, which locks jsb sim to 120hz, but you still have to manually lock/cap your game rate, it can be anything but just needs to hold steady. Otherwise this introduces another variable into the flight model. The other thing is you can take a look at one of the latest FlightGear F16 models, and try to extract the file/info needed to work with pure jsb sim. Since flightgear model files tend to have a lot of extra flightgear specific data added.

bcoconni, 2022-07-02:
@Synergy91 given the lack of elements to support the bug claim, this support request is moved to discussions.