JSBSim-Team/jsbsim #684: Prefer PROJECT_SOURCE_DIR over CMAKE_SOURCE_DIR

Aaron-SP, 2022-07-25:
When bundling JSBSIM as an external project using CMAKE FetchContent, JSBSIM is not configured because it uses CMAKE_SOURCE_DIR instead of PROJECT_SOURCE_DIR.

[CMAKE Issue Explained](https://stackoverflow.com/questions/32028667/are-cmake-source-dir-and-project-source-dir-the-same-in-cmake)

I've made a fork with this change at [repo](https://github.com/Aaron-SP/jsbsim).

Does it make sense to make this change to expand packaging capabilities of JSBSim with CMAKE?

bcoconni, 2022-07-26:
Thanks for reporting this issue.
Would you mind submitting a pull request from your repo ?

Aaron-SP, 2022-07-28:
I have opened a pull request [here](https://github.com/JSBSim-Team/jsbsim/pull/689). This fixes my issue but I did not check building the extra modules such as Python etc. We might want to double check that any files that code generate CMAKE are also updated, not sure.

bcoconni, 2022-07-29:
Your pull request has been merged.
Thanks for your contribution 👍

commit 3f939fa7 (2022-07-29): Replace `CMAKE_SOURCE_DIR ` with `PROJECT_SOURCE_DIR` instead (#689).

This allows embedding JSBSim into a higher level repository where JSBSim is not at the highest CMAKE level. This supports use cases such as CMAKE FetchContent.

Resolves #684
commit 186557f2 (2022-08-20): Replace `CMAKE_SOURCE_DIR ` with `PROJECT_SOURCE_DIR` instead (#689).

This allows embedding JSBSim into a higher level repository where JSBSim is not at the highest CMAKE level. This supports use cases such as CMAKE FetchContent.

Resolves #684
closed 2022-07-29 by bcoconni